openwrt/openwrt #19588: Qualcommax Linksys MX4300 - Snapshot - Unable to compile  - ERROR: package/libs/gettext-full [host] failed to build

kyogithub, 2025-07-29:
### Describe the bug

I got the error "ERROR: package/libs/gettext-full [host] failed to build" while compiling snapshot for the Linksys MX4300, attached git-action logs.

Git action: https://github.com/kyogithub/openwrt-mx4300/actions/runs/16582694569/job/46901957744

[logs_42509490308.zip](https://github.com/user-attachments/files/21480649/logs_42509490308.zip)

### OpenWrt version

Main

### OpenWrt release

snapshot

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX4300

### Image kind

Self-built image

### Steps to reproduce

https://github.com/kyogithub/openwrt-mx4300/actions/runs/16582694569/job/46901957744

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

paulcarroty, 2025-07-29:
The official Firmware Selector works well, SNAPSHOT (r30617-9c26d14489)

robimarko, 2025-08-01:
This was fixed yesterday

closed 2025-08-01 by robimarko