openwrt/openwrt #19592: USB3 since 24.10 not working on Asus rt-ac88u | bcm53xx

stevob99, 2025-07-29:
### Describe the bug

Since  24.10.xx the USB3 port of the router Asus rt-ac88u (bcm53xx) is not working anymore.
(22.03.xx and 23.05.xx has been working absolutely fine)

Nothing occurs at dmesg or logread when inserting the USB3-stick in USB3 port (in the front).
Looks like USB3 has been deactivated.

Just to make sure, I also tested 4 different usb-sticks 3.0 and also 3 different Asus Router rt-ac88u.
 I said, I tested the same device: with openwrt 22.xx this device has been working with the same USB-stick (SanDisk 3.2Gen1) and same usb-port.

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Asus RT-AC88U

### Image kind

Official downloaded image

### Steps to reproduce

_just insert any usb3 stick in the front usb slot_

### Actual behaviour

_nothing appears in dmesg / logread_

### Expected behaviour

```
248.730091] kmodloader: loading kernel modules from /etc/modules.d/*
[  248.742054] SCSI subsystem initialized
[  248.751359] usb-storage 4-1:1.0: USB Mass Storage device detected
[  248.758568] scsi host0: usb-storage 4-1:1.0
[  248.763241] usbcore: registered new interface driver usb-storage
[  248.770833] usbcore: registered new interface driver uas
[  248.776751] kmodloader: done loading kernel modules from /etc/modules.d/*
[  248.819225] kmodloader: loading kernel modules from /etc/modules.d/*
[  248.827797] kmodloader: done loading kernel modules from /etc/modules.d/*
[  248.869240] kmodloader: loading kernel modules from /etc/modules.d/*
[  248.877710] kmodloader: done loading kernel modules from /etc/modules.d/*
[  249.793568] scsi 0:0:0:0: Direct-Access      USB      SanDisk 3.2Gen1 1.00 PQ: 0 ANSI: 6
[  249.804186] sd 0:0:0:0: [sda] 60088320 512-byte logical blocks: (30.8 GB/28.7 GiB)
[  249.812063] sd 0:0:0:0: [sda] Write Protect is off
[  249.816897] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[  249.817451] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  249.833210]  sda: sda1 sda2
[  249.837639] sd 0:0:0:0: [sda] Attached SCSI removable disk

root@OpenWrt:/# lsusb -tv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003
    |__ Port 1: Dev 2, If 0, Class=, Driver=usb-storage, 5000M
        ID 0781:5591
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
    ID 1d6b:0002
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
    ID 1d6b:0001
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M
    ID 1d6b:0002
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

stevob99, 2025-07-29:
If I insert a usb3-device into the usb3-slot nothing appears!
dmesg or logread does not log anything at all! Also lsusb does not show up anything.

Here I inserted a usb2-stick into the usb3-port:
`
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.353847] kmodloader: loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:08 2025 kern.notice kernel: [  234.368117] SCSI subsystem initialized
Mon Jul 28 11:25:08 2025 kern.info kernel: [  234.379123] usbcore: registered new interface driver usb-storage
Mon Jul 28 11:25:08 2025 kern.info kernel: [  234.387270] usbcore: registered new interface driver uas
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.393571] kmodloader: done loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.441130] kmodloader: loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.450200] kmodloader: done loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.498665] kmodloader: loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:08 2025 user.info kernel: [  234.508021] kmodloader: done loading kernel modules from /etc/modules.d/*
Mon Jul 28 11:25:25 2025 kern.info kernel: [  251.259771] usb 1-1: new high-speed USB device number 4 using ehci-platform
Mon Jul 28 11:25:27 2025 kern.info kernel: [  253.761806] usb-storage 1-1:1.0: USB Mass Storage device detected
Mon Jul 28 11:25:27 2025 kern.info kernel: [  253.768780] scsi host0: usb-storage 1-1:1.0
Mon Jul 28 11:25:29 2025 kern.notice kernel: [  254.971118] scsi 0:0:0:0: Direct-Access     SMI      USB DISK         1100 PQ: 0 ANSI: 4
Mon Jul 28 11:25:29 2025 kern.notice kernel: [  254.982679] sd 0:0:0:0: [sda] 15974400 512-byte logical blocks: (8.18 GB/7.62 GiB)
Mon Jul 28 11:25:29 2025 kern.notice kernel: [  254.991050] sd 0:0:0:0: [sda] Write Protect is off
Mon Jul 28 11:25:29 2025 kern.debug kernel: [  254.995861] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
Mon Jul 28 11:25:29 2025 kern.warn kernel: [  254.996551] sd 0:0:0:0: [sda] No Caching mode page found
Mon Jul 28 11:25:29 2025 kern.warn kernel: [  255.001907] sd 0:0:0:0: [sda] Assuming drive cache: write through
Mon Jul 28 11:25:29 2025 kern.info kernel: [  255.013061]  sda: sda1
Mon Jul 28 11:25:29 2025 kern.notice kernel: [  255.015697] sd 0:0:0:0: [sda] Attached SCSI removable disk

root@OpenWrt:~# lsusb -tv
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/2p, 480M
    ID 1d6b:0002  
    |__ Port 001: Dev 004, If 0, Class=[unknown], Driver=usb-storage, 480M
        ID 090c:1000  
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/2p, 12M
    ID 1d6b:0001  
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003  
`
vice versa, the usb3-stick into the usb2-port (in the back) also works fine.

I can test a usb mouse, etc later, but I'm pretty sure, because of USB1.x or USB2 it will work also.

brada4, 2025-07-29:
Please do not post (edit away) everything to do with working USB2 port. It is hard to tell apaart where working device is connected. Nobody doubts your skill operating USB plugs.

Anyway @arinc9 may know better what was broken since he added device support.

namiltd, 2025-07-29:
lsmod

stevob99, 2025-07-29:
lsmod:

`
bcma_hcd               12288  0 
brcmfmac              200704  1 brcmfmac_bca
brcmfmac_bca           12288  0 
brcmutil               12288  1 brcmfmac
cdc_ether              12288  1 rndis_host
cdns_usb_common        16384  0 
cfg80211              294912  1 brcmfmac
chacha_neon            20480  1 libchacha20poly1305
compat                 12288  3 brcmfmac_bca,brcmfmac,cfg80211
crc_ccitt              12288  1 ppp_async
crc32_generic          12288  0 
crc32c_generic         12288  1 
ehci_fsl               12288  0 
ehci_hcd               45056  3 ehci_pci,ehci_platform,ehci_fsl
ehci_pci               12288  0 
ehci_platform          12288  0 
ext4                  450560  0 
f2fs                  393216  0 
fsl_mph_dr_of          12288  0 
gpio_button_hotplug    12288  0 
ip6_udp_tunnel         12288  1 wireguard
jbd2                   69632  1 ext4
kpp                    12288  0 
leds_gpio              12288  0 
ledtrig_usbport        12288  0 
libchacha20poly1305    12288  1 wireguard
libcrc32c              12288  1 nf_tables
libcurve25519_generic   16384  1 wireguard
mbcache                12288  1 ext4
mii                    12288  1 usbnet
nf_conntrack           86016  8 nft_redir,nft_nat,nft_masq,nft_flow_offload,nft_ct,nf_nat,nf_flow_table,nf_conntrack_netlink
nf_conntrack_netlink   36864  0 
nf_defrag_ipv4         12288  1 nf_conntrack
nf_defrag_ipv6         16384  1 nf_conntrack
nf_flow_table          28672  2 nf_flow_table_inet,nft_flow_offload
nf_flow_table_inet     12288  0 
nf_log_syslog          16384  0 
nf_nat                 32768  4 nft_redir,nft_nat,nft_masq,nft_chain_nat
nf_reject_ipv4         12288  2 nft_reject_ipv4,nft_reject_inet
nf_reject_ipv6         12288  2 nft_reject_ipv6,nft_reject_inet
nf_tables             192512292 nft_fib_inet,nf_flow_table_inet,nft_reject_ipv6,nft_reject_ipv4,nft_reject_inet,nft_reject,nft_redir,nft_quota,nft_numgen,nft_nat,nft_masq,nft_log,nft_limit,nft_hash,nft_flow_offload,nft_fib_ipv6,nft_fib_ipv4,nft_fib,nft_ct,nft_chain_nat
nfnetlink              16384  3 nf_tables,nf_conntrack_netlink
nft_chain_nat          12288  2 
nft_ct                 20480  4 
nft_fib                12288  3 nft_fib_inet,nft_fib_ipv6,nft_fib_ipv4
nft_fib_inet           12288  0 
nft_fib_ipv4           12288  1 nft_fib_inet
nft_fib_ipv6           12288  1 nft_fib_inet
nft_flow_offload       12288  0 
nft_hash               12288  0 
nft_limit              12288  1 
nft_log                12288  0 
nft_masq               12288  1 
nft_nat                12288  0 
nft_numgen             12288  0 
nft_quota              12288  0 
nft_redir              12288  0 
nft_reject             12288  3 nft_reject_ipv6,nft_reject_ipv4,nft_reject_inet
nft_reject_inet        12288  2 
nft_reject_ipv4        12288  0 
nft_reject_ipv6        12288  0 
nls_base               12288  2 f2fs,usbcore
ohci_hcd               32768  2 ohci_pci,ohci_platform
ohci_pci               12288  0 
ohci_platform          12288  0 
phy_bcm_ns_usb2        12288  1 
phy_bcm_ns_usb3        12288  1 
poly1305_arm           16384  1 libchacha20poly1305
ppp_async              16384  0 
ppp_generic            32768  3 pppoe,ppp_async,pppox
pppoe                  16384  0 
pppox                  12288  1 pppoe
rndis_host             12288  0 
roles                  12288  1 cdns_usb_common
scsi_common            12288  4 uas,usb_storage,sd_mod,scsi_mod
scsi_mod              102400  3 uas,usb_storage,sd_mod
sd_mod                 40960  0 
slhc                   12288  1 ppp_generic
uas                    20480  0 
udp_tunnel             16384  1 wireguard
usb_common             12288  7 cdns_usb_common,xhci_plat_hcd,xhci_hcd,ohci_platform,ohci_hcd,ehci_platform,usbcore
usb_storage            49152  1 uas
usbcore               151552 18 brcmfmac,rndis_host,cdc_ether,usbnet,ledtrig_usbport,uas,usb_storage,ohci_pci,cdns_usb_common,ehci_pci,xhci_plat_hcd,xhci_pci,xhci_hcd,ohci_platform,ohci_hcd,ehci_platform,ehci_fsl,ehci_hcd
usbnet                 24576  2 rndis_host,cdc_ether
wireguard              61440  0 
xhci_hcd              114688  2 xhci_plat_hcd,xhci_pci
xhci_pci               12288  0 
xhci_plat_hcd          12288  0 
`

kernel log is attached.

[dmesg_asus_rt-ac88u.txt](https://github.com/user-attachments/files/21494124/dmesg_asus_rt-ac88u.txt)

generic-internet-user, 2025-10-26:
So I flashed snapshot (`OpenWrt SNAPSHOT r31594-93f86627c4`) on the device instead, because why not, and it does not work there either...

Saw [this](https://github.com/openwrt/openwrt/issues/20153) where there was a patch for this but that didn't do anything apparently; the commit for that is [here](https://github.com/GoetzGoerisch/openwrt/commit/9e55056017b19977a608409274ba7e5df4ce5507).

brada4, 2025-11-10:
LPM is later negotiated feature. You dont get the initial device ID and query, less even to power management.

Ansuel, 2025-11-10:
@raenye from my experience (i had similar problems with ipq806x) unless someone invest lots of time with bisecting it's hard to find the problem... One starting point would be to check the broken kernel between 5.15 and 6.1

Stephen2301, 2026-01-04:
When I built my own image, the USB3 port worked. After switching to the builder snapshot version, the USB3 port stopped working. I made many tweaks and changed several files; I used GCC 15 and binutils 2.45. I suspect the issue is that I removed the fake-doorbell patch (180-usb-xhci-add-support-for-performing-fake-doorbell.patch) when building the image, that may be why USB3 worked before.