openwrt/openwrt #19608: php8-fpm broken since update to libxml2 library

brackenhill-mob, 2025-07-31:
### Describe the bug

libxml2 update to 2.14.5-r1 breaks php8-fpm which is looking for shared library libxml2.so.2

This breaks a **_lot_** of stuff (including all websites using WordPress)

Fix urgently needed ;)

[temporary fix is to manually copy libxml2.so.16 to libxml2.so.2]

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

x86/64

### Device

Dell Inc. OptiPlex 3020

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mhei, 2025-08-01:
Just recompiling should be sufficient, but a new version bump is also on the way.

mic-user, 2025-08-01:
It affect to entire libxml2 usage packages include lighttpd webdav mod.
What are the amature junior doing,
openwrt is not a child grade fun source update activity.
Do not touch/comit  the source code if you 200% sure. Damn amature fuckers.

workaround : cd /usr/lib/ && cp libxml2.so.16 libxml2.so.2

hauke, 2025-10-19:
Fixes with ABI versioning and rebuild, see bc9f1cab83d09f91e28c7ca830de778b08bd2842

closed 2025-10-19 by hauke