openwrt/openwrt #19630: dnsmasq no listen directive at boot

glaive-tordu, 2025-08-02:
### Describe the bug

Hi,

Dnsmasq do not generate listen directives in config file at boot.
But if reload it, it's good.

Best regards,

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-08-02:
That seems to be by design.
https://github.com/openwrt/openwrt/commit/2336b942b37f265c59547d738ca558b61102833d

brada4, 2025-08-02:
How is that no reload occur?
Default is 
> If no --interface or --listen-address options are given dnsmasq listens on all available interfaces

glaive-tordu, 2025-08-03:
I'm referring to @dave14305's comment. Looking at the script, listen directives are not generated at boot time. So I think that as soon as the network is configured, it triggers the reload of dnsmasq. Is that how it should work?

If I not reload manually dnsmasq after the boot, there is no directive listen, it can make security issue.