openwrt/openwrt #19637: netifd: (null) is not iterable when enable MLO

CoelacanthusHex, 2025-08-02:
### Describe the bug

When I try to enable MLO, I get this error, and the SSID isn't up.
```
Sun Aug  3 03:25:32 2025 daemon.warn netifd: (null) is not iterable In config_init(), file /lib/netifd/wireless.uc, line 113, byte 36:   called from function config_init (/lib/netifd/main.uc:31:27)   called from anonymous function (/lib/netifd/main.uc:57:21)   `            dev_names = [ wdev.mlo_name, ...dev_names ];`   Near here ----------------------------------^
```

### OpenWrt version

r30655+1-287f1ea3dd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Banana Pi BPI-R4 (2x SFP+)

### Image kind

Self-built image

### Steps to reproduce

Setup SSID with config below according to https://github.com/openwrt/openwrt/commit/217c31542330ad47907fb153d3232a15833698b1:
```
config wifi-iface 'wifinet0'
    list radio 'radio0'
    list radio 'radio1'
    option mlo '1'
    option ssid 'OpenWrt'
    option mode 'ap'
    option network 'lan'
    option ssid 'Coelacanthus Test Wi-Fi'
```

### Actual behaviour

You will see error above in logread and SSID isn't up.

### Expected behaviour

No error and SSID works.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_44=y
CONFIG_BINUTILS_VERSION="2.44"
CONFIG_BINUTILS_VERSION_2_44=y
# CONFIG_BPF_TOOLCHAIN_BUILD_LLVM is not set
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
CONFIG_CCACHE=y
# CONFIG_DROPBEAR_DBCLIENT is not set
CONFIG_DROPBEAR_ZLIB=y
CONFIG_GCC_DEFAULT_PIE=y
CONFIG_GCC_USE_GRAPHITE=y
CONFIG_GNUTLS_ALPN=y
[... 457 lines cut ...]
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PACKAGE_zoneinfo-europe=y
CONFIG_PACKAGE_zoneinfo-indian=y
CONFIG_PACKAGE_zoneinfo-pacific=y
CONFIG_PACKAGE_zoneinfo-poles=y
CONFIG_PACKAGE_zstd=y
CONFIG_PARTED_READLINE=y
CONFIG_PCAP_HAS_NETFILTER=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_QRENCODE_PNG=y
CONFIG_RSYNC_acl=y
CONFIG_RSYNC_xattr=y
CONFIG_SOCAT_SSL=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LLVM_HOST=y
CONFIG_USE_LTO=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
CONFIG_ZSTD_OPTIMIZE_O3=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

CoelacanthusHex, 2025-08-02:
Sorry, I applied https://github.com/openwrt/openwrt/pull/19071 and https://github.com/openwrt/openwrt/pull/19503. I will try a clean build, but it should be unrelated.

CoelacanthusHex, 2025-08-02:
Ok, I can confirm local changes are unrelated, clean r30655-287f1ea3dd has the same issue. I have updated the brief. Is there a way to retrigger the bot?

nbd168, 2025-08-02:
This only works with CONFIG_WIFI_SCRIPTS_UCODE=y

CoelacanthusHex, 2025-08-02:
The issue still seems to be here after I rebuild the image with CONFIG_WIFI_SCRIPTS_UCODE=y

nbd168, 2025-08-02:
Ah, now I see it. You used `list radio` instead of `list device` in your config

CoelacanthusHex, 2025-08-02:
> Ah, now I see it. You used `list radio` instead of `list device` in your config

Oh. That's because you also typo it in commit message of https://github.com/openwrt/openwrt/commit/217c31542330ad47907fb153d3232a15833698b1

> And by the way, upstream mt76 does not have MLO support enabled yet. It's still being worked on

Thanks for your reminder.

closed 2025-08-02 by CoelacanthusHex