openwrt/openwrt #19640: 2,5Gb-Ports on Zyxel XGS1210 not available

sh-dvl, 2025-08-03:
### Describe the bug

the two 2,5Gbit-Ports (ports 9+10) are not available for configuration.

### OpenWrt version

r30647-9dddc0bed0

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

realtek/rtl930x

### Device

Zyxel XGS1210-12 Switch

### Image kind

Official downloaded image

### Steps to reproduce

device: Zyxel XGS1210-12 rev. B1, running before with Firmware XGS1210-12_V2.00(ABTY.1)C0

now installed SNAPSHOT r30647-9dddc0bed0 by following description at: 
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=6af9476b8a3c75ce65f17cfa89da7d35360a4756 

### Actual behaviour

ports 9 +10 unavailable:

```
>ls -1 /sys/class/net/
eth0
lan1
lan11
lan12
lan2
lan3
lan4
lan5
lan6
lan7
lan8
lo
switch
switch.1
```

### Expected behaviour

all ports should be available

```
>ls -1 /sys/class/net/
eth0
lan1
lan10
lan11
lan12
lan2
lan3
lan4
lan5
lan6
lan7
lan8
lan9
lo
switch
switch.1
```

### Additional info

bootlog shows:
```
[    2.322914] rtl83xx-switch switch@1b000000 lan9 (uninitialized): no phy at 24
[    2.330958] rtl83xx-switch switch@1b000000 lan9 (uninitialized): failed to connect to PHY: -ENODEV
[    2.340979] rtl83xx-switch switch@1b000000 lan9 (uninitialized): error -19 setting up PHY for tree 0, switch 0, port 24
[    2.353185] rtl83xx-switch switch@1b000000 lan10 (uninitialized): no phy at 25
[    2.361309] rtl83xx-switch switch@1b000000 lan10 (uninitialized): failed to connect to PHY: -ENODEV
[    2.371427] rtl83xx-switch switch@1b000000 lan10 (uninitialized): error -19 setting up PHY for tree 0, switch 0, port 25
```
with the original firmware from zyxel there wasn't any problem with these ports.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

plappermaul, 2025-08-03:
This is v2 version and needs a separate DTS plus a new build recipe. See https://forum.openwrt.org/t/support-for-rtl838x-based-managed-switches/57875/3472

@janh might remember the location of a modified DTS.

sh-dvl, 2025-08-03:
thanks for this info. The linked thread let me assume, that the "rev B1" is equivalent to v2? would be nice, to have this documented on the wiki-page of this switch (I've got no write-access to it and maybe it's not a  good idea, to publish things, which can't be verified by myself).
Currently I've to use this switch as a quick-replacement for a larger one, which has to be reconfigured, but afterwards, there's actually no productive use planned for the zyxel before it completly works as expected, so I'd be able to do some tests during my spare-time. Even I've some experience with hardware-debugging and -programming, unfortunately I've not much experience with openwrt-development (for example I don't know the meaning of "DTS"), so I need a bit more info, how to go on. The given thread seems to be already a bit special and multiple different devices are mentioned, which makes a bit difficult, to identify the relevant information, so do you perhaps have other links or hints?

Djfe, 2025-08-28:
DTS (device tree source) is one of the files in the commit you linked
see: https://github.com/openwrt/openwrt/blob/664424aaeb4a7b74123cc94c536e42fa925cf8c3/target/linux/realtek/dts/rtl9302_zyxel_xgs1210-12.dts

https://en.wikipedia.org/wiki/Devicetree

> In [computing](https://en.wikipedia.org/wiki/Computing), a devicetree (also written device tree) is a [data structure](https://en.wikipedia.org/wiki/Data_structure) describing the hardware components of a particular computer so that the [operating system](https://en.wikipedia.org/wiki/Operating_system)'s [kernel](https://en.wikipedia.org/wiki/Kernel_(operating_system)) can use and manage those components, including the [CPU](https://en.wikipedia.org/wiki/Central_processing_unit) or CPUs, the [memory](https://en.wikipedia.org/wiki/Computer_memory), the [buses](https://en.wikipedia.org/wiki/Bus_(computing)) and the [integrated peripherals](https://en.wikipedia.org/wiki/Integrated_peripheral).

The advantage is that most devices can share the same (cpu) template: dtsi file
https://github.com/openwrt/openwrt/blob/664424aaeb4a7b74123cc94c536e42fa925cf8c3/target/linux/realtek/dts/rtl930x.dtsi

In this case someone with access to the hardware can create a common dtsi file from the dts file I linked at the top of my comment and version a and b will then get different dts files with slight adjustments/differences

sh-dvl, 2025-08-28:
thanks. now I have the file to play with.

meanwhile I've had a look at this, but it's still a bit unclear.

with setting phy24 smi from original 1/8 to 1/0 the first msg in log has has gone ('mdio_bus mdio-bus: MDIO device at address 8 is missing' ) and the second has changed from ' failed to connect to PHY: -ENODEV' to 'failed to connect to PHY: -EIO'.

setting then phy25 smi from original 2/9 to 1/1 leaves ports 1-8 unusable (?)

setting phy25 smi from original 2/9 to 2/0 gives same result as for phy24 and ports 1-8 are still usable

this let me assume, that the smi-values 1/0 and 2/0 for v2 matches, but there's still a proböem with initialization.

the sds-values doesn't seem to be relevant, because all ports ar set from 0 to 9 without any discontinuity.

the "compatible"-entry is not clear, because I didn't find any declaration or definition of the used values.

because the v1 has same features as v2 I assume the value should be equal, and MDIO-Init seems to be done already. nevertheless it would be nice to know, where the declaration/definitions can be found...

within "ports" there's a "phy-mode" set as "hsgmii". a test with setting it to "2500base-x" makes no difference, but I expected same as for compatible, so no change should be needed.

maybe the missing mdio-failure is mis-leading and smi still doesn't match? or has something to be done within uboot-startup to get the ports work?

I'm currently a bit unsteady how to gon on...

plappermaul, 2025-08-29:
Give https://github.com/openwrt/openwrt/pull/19843 a try for another test run.

bendavid, 2025-08-30:
Building from current main  (https://github.com/openwrt/openwrt/commit/886382ba256cc795c1bf2bf5d990ced9f23071e8)  on XGS1210-12 rev B1 and changing the smi addresses in the DTS as below I see lan9 and lan10 appearing, and showing the correct link behaviour at 2.5gbps, but no traffic flowing.  I guess because it still needs https://github.com/openwrt/openwrt/pull/19518 to be resolved.

```
diff --git a/target/linux/realtek/dts/rtl9302_zyxel_xgs1210-12.dts b/target/linux/realtek/dts/rtl9302_zyxel_xgs1210-12.dts
index 9ff919573a..4057ae5392 100644
--- a/target/linux/realtek/dts/rtl9302_zyxel_xgs1210-12.dts
+++ b/target/linux/realtek/dts/rtl9302_zyxel_xgs1210-12.dts
@@ -205,7 +205,7 @@
                phy24: ethernet-phy@24 {
                        reg = <24>;
                        compatible = "ethernet-phy-ieee802.3-c45";
-                       rtl9300,smi-address = <1 8>;
+                       rtl9300,smi-address = <1 1>;
                        sds = < 6 >;
                        // Disabled because we do not know how to bring up again
                        // reset-gpios = <&gpio0 21 GPIO_ACTIVE_LOW>;
@@ -214,7 +214,7 @@
                phy25: ethernet-phy@25 {
                        reg = <25>;
                        compatible = "ethernet-phy-ieee802.3-c45";
-                       rtl9300,smi-address = <2 9>;
+                       rtl9300,smi-address = <2 2>;
                        sds = < 7 >;
                        // Disabled because we do not know how to bring up again
                        // reset-gpios = <&gpio0 21 GPIO_ACTIVE_LOW>;

```

sh-dvl, 2025-08-30:
have done a quick test:

with local test-tree of 20250827, where I tested various smi-values above patch doesn't change anything.

then make a fresh clone, applied above patch, I can confirm that they 2.5G-phys seems to be recognized, but same problem as before with the other patches: nothing is accessible.

last msgs on serial port are:
```
[   12.172212] procd: - early -
[   12.175971] procd: - watchdog -
[   12.807626] procd: - watchdog -
[   12.811687] procd: - ubus -
```

then nothing happens any more.

because of the fresh clone and not applying the patches of plappermaul, the reason doesn't seem to be there or in patch of bendavid.

it's late here, maybe I'll have a look at other possible reasons tomorrow.

sh-dvl, 2025-08-31:
another short test today shows, that the unresponsiveness is due to ubus, which seems to exit with failure and procd endless tries to restart it, so it has nothing to do with the patches here.

Maybe I find later today some time to test both patches together and find a method, that ubus does not exit...

sh-dvl, 2025-08-31:
it's a bit strange. with tree of 20250829, where plappermauls patches should be established and now the smis are set as bendavid suggests I still get -EIO. I'm now going to integrate plappermauls patches into yesterdays clone...

the other problem doesn't seem to be ubus, but procd itself, which shows a zombie-subprocess

sh-dvl, 2025-08-31:
ok, now applied pappelmaul patches manually to latest local clone (20250930) (to make sure, they are applied)

but unfortunately still no traffic occurs. make an ifconfig to one of those ports with 1Gb-connected cable gives:

```
[  287.033401] rtl83xx_fib_event_work_do: FIB4 failed
[  287.039016] rtl83xx-switch switch@1b000000 lan9: configuring for phy/2500base-x link mode
[  287.048227] rtl93xx_phylink_mac_config SDS is 6
[  287.053288] rtl9300_sds_rst 31
[  287.150000] rtl9300_phy_enable_10g_1g 1gbit phy: 00001140
[  287.157124] rtl9300_phy_enable_10g_1g 1gbit phy enabled: 00001140
[  287.164922] rtl9300_phy_enable_10g_1g 10gbit phy: 00002040
[  287.172086] rtl9300_phy_enable_10g_1g 10gbit phy after: 00002040
[  287.179861] rtl9300_phy_enable_10g_1g set medium: 00000000
[  287.187040] rtl9300_phy_enable_10g_1g set medium after: 00000002
[  287.213853] rtl9300_serdes_setup: Configuring RTL9300 SERDES 6
[  287.222410] rtl9300_serdes_mac_link_config: registers before 00000000 00001403
[  287.234503] rtl9300_serdes_mac_link_config: registers after 00000000 00001403
[  287.259630] rtsds_930x_config_pll: SDS 6 using LC PLL for 2500base-x
[  287.335324] start_1.1.1 initial value for sds 6
[  287.397561] end_1.1.1 --
[  287.400400] start_1.1.2 Load DFE init. value
[  287.407204] end_1.1.2
[  287.409748] start_1.1.3 disable LEQ training,enable DFE clock
[  287.428190] end_1.1.3 --
[  287.431023] start_1.1.4 offset cali setting
[  287.437735] end_1.1.4
[  287.440272] start_1.1.5 LEQ and DFE setting
[  287.444915] rtl9300_do_rx_calibration_1 not PHY-based or SerDes, implement DAC!
[  287.465095] end_1.1.5
[  287.476704] start_1.2.1 ForegroundOffsetCal_Manual
[  287.486138] end_1.2.1
[  287.497235] start_1.2.3 Foreground Calibration
[  287.515787] rtl9300_do_rx_calibration_2_3: fgcal_gray: 31, fgcal_binary 31
[  287.524453] rtl9300_do_rx_calibration_2_3: end_1.2.3
[  287.530020] start_1.4.1
[  287.754408] end_1.4.1
[  287.757214] start_1.4.2
[  287.769788] vth_set_bin = 7
[  287.772533] vth_set_bin = 3
[  287.776672] Vth Maunal = 0
[  287.896125] Tap0 Sign : +
[  287.899162] tap0_coef_bin = 31
[  287.903061] tap0 manual = 0
[  287.912481] end_1.4.2
[  287.986809] rtl9300_sds_sym_err_reset unsupported phy mode
[  287.995485] rtl9300_sds_sym_err_reset unsupported phy mode
[  288.001596] rtl9300_sds_sym_err_get unsupported PHY-mode
[  288.008544] rtl9300_sds_sym_err_get unsupported PHY-mode
[  288.014465] start_1.1.1 initial value for sds 6
[  288.076718] end_1.1.1 --
[  288.079558] start_1.1.2 Load DFE init. value
[  288.086377] end_1.1.2
[  288.088923] start_1.1.3 disable LEQ training,enable DFE clock
[  288.107399] end_1.1.3 --
[  288.110234] start_1.1.4 offset cali setting
[  288.116947] end_1.1.4
[  288.119494] start_1.1.5 LEQ and DFE setting
[  288.124133] rtl9300_do_rx_calibration_1 not PHY-based or SerDes, implement DAC!
[  288.144321] end_1.1.5
[  288.155965] start_1.2.1 ForegroundOffsetCal_Manual
[  288.165373] end_1.2.1
[  288.176424] start_1.2.3 Foreground Calibration
[  288.194927] rtl9300_do_rx_calibration_2_3: fgcal_gray: 31, fgcal_binary 31
[  288.203656] rtl9300_do_rx_calibration_2_3: end_1.2.3
[  288.209247] start_1.4.1
[  288.433676] end_1.4.1
[  288.436458] start_1.4.2
[  288.449028] vth_set_bin = 7
[  288.451769] vth_set_bin = 3
[... 70 lines cut ...]
[  290.198670] end_1.4.1
[  290.201415] start_1.4.2
[  290.213980] vth_set_bin = 7
[  290.216771] vth_set_bin = 3
[  290.220886] Vth Maunal = 0
[  290.340329] Tap0 Sign : +
[  290.343368] tap0_coef_bin = 31
[  290.347309] tap0 manual = 0
[  290.356761] end_1.4.2
[  290.461407] rtl83xx-switch switch@1b000000 lan9: Link is Up - 1Gbps/Full - flow control rx/tx
```
doing the same on one of the 1Gb-Port give this:
```
[  694.790725] rtl83xx_fib_event_work_do: FIB4 failed
[  694.796281] rtl83xx-switch switch@1b000000 lan8: configuring for phy/usxgmii link mode
[  694.805122] rtl93xx_phylink_mac_config SDS is -1
[  694.811144] 8021q: adding VLAN 0 to HW filter on device lan8
[  694.817832] rtl83xx_fib_event_work_do: FIB4 failed
[  694.823509] rtl83xx_fib_event_work_do: FIB4 failed
```
and ping works.

well, this problem seems to be really deep in HW/Code.

If you've further suggestions, which I could test: the switch currently is not productive and when I've got some time, I'll make further tests, but currently I don't know, what I could do on my own for solving this without having more informations about the chips themselves

plappermaul, 2025-08-31:
Run "sys dump hwp all" as per https://svanheule.net/switches/zyxel_xgs1210_series

sh-dvl, 2025-09-01:
I could get access to original firmware serial console. there's a command "diag", where above command is available.

output is:

```
unitMapStruct[0].pUnitInfo:83cf8014,parsed_info=83cf8000,swDescp=c014ab7c 
unitMapStruct[1].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[2].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[3].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[4].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[5].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[6].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[7].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[8].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[9].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[10].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[11].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[12].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[13].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[14].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
unitMapStruct[15].pUnitInfo:c0155e2c,parsed_info=0,swDescp=0 
(&parsedInfoEmpty.unitInfo[EMPTY]=c0155e2c)
info->hwp=XGS1210_12_V2(c014aacc)                   
info->hwp_localUnitCnt=1          
info->hwp_embeddedCpuUnitId=0     
info->hwp_cascade_mode=0          
info->hwp_cascadeSlaveUnitID=0    
info->hwp->swDescp[0]=c014ab7c
info->hwp->swDescp[1]=0
info->hwp->swDescp[2]=0
info->hwp->swDescp[3]=0
info->hwp->swDescp[4]=0
info->hwp->swDescp[5]=0
info->unitInfo[0].hwp_chip_id=93021000
info->unitInfo[0].hwp_chip_revision=1
info->unitInfo[0].hwp_chip_family_id=93000000
info->unitInfo[0].hwp_maxMacId=28
info->unitInfo[0].hwp_minMacId=0
info->unitInfo[0].hwp_macID2PortDescp:
Ether  : count=12, max=27, min=0
Uplink : count=4, max=27, min=24
Cascade: count=0, max=0, min=64
Fe     : count=0, max=0, min=64
Ge     : count=8, max=7, min=0
Xge    : count=2, max=27, min=26
SXge   : count=0, max=0, min=64
Copper : count=10, max=25, min=0
Fiber  : count=0, max=0, min=64
Combo  : count=0, max=0, min=64
Serdes : count=2, max=27, min=26
SDS  6 PHY:20
SDS  7 PHY:20
SDS  8 PHY:36
SDS  9 PHY:36
attr[0]=0
attr[1]=f000000
attr[1]=0
attr[2]=0
attr[2]=0
attr[3]=10000000
attr[3]=0
attr[4]=0
attr[4]=0
eth[0]=0
eth[1]=ff
eth[1]=0
eth[2]=3000000
eth[2]=0
eth[3]=0
eth[3]=0
eth[4]=c000000
eth[4]=0
eth[5]=0
eth[5]=0
med[0]=0
med[1]=0
med[1]=0
med[2]=0
med[2]=0
med[3]=c000000
med[3]=0

info->unitInfo[0].hwp_allPortMask=1f0000ff

info->unitInfo[0].hwp_allPortMask=0

info->unitInfo[0].hwp_multiSdsBaseSds:

info->unitInfo[0].hwp_macID2SCDescp:

info->unitInfo[0].hwp_scBaseMacID:

info->unitInfo[0].swDescp=c014ab7c
info->unitInfo[0].parsed_info=83cf8000
```

other available commands are:
```
exit             - exit diag shell
unit             - unit configure
terminal         - terminal configuration
sdk              - sdk configuration
nic              - NIC configuration
stack            - stack configuration
oam              - OAM configuration
l2-ntfy          - L2-notification configuration
acl              - ACL configuration
debug            - debug configuration
diag             - diag configuration
serdes           - serdes interface
eee              - EEE configuration
flowctrl         - flowctrl configuration
l2-table         - l2 table configuration
l3               - l3 configuration
led              - LED configuration
mib              - mib configuration
mirror           - mirror configuration
rspan            - RSPAN configuration
sflow            - sFlow configuration
port             - port configuration
qos              - QoS configuration
bandwidth        - bandwidth configuration
storm-control    - storm-control configuration
register         - register configuration
vlan             - VLAN configuration
security         - security configuration
stp              - stp configuration
switch           - switch configuration
sys              - system
time             - time configuration
trap             - trap configuration
trunk            - trunk configuration
ipmcast          - ipmcast configuration
mcast            - mcast configuration
pie              - PIE configuration
field-selector   - field selector configuration
range-check      - range check configuration
gpio             - generalCtrl GPIO configuration
spi              - SPI configuration
i2c              - I2C configuration
```

anything else, which might be interesting?

plappermaul, 2025-09-01:
That PR contains a bugfix that might help. No other knowledge around bringing up RTL8221B on RTL930x exists right now.

sh-dvl, 2025-09-02:
ok, cloned yesterday evening again, verified by random sample, that all here mentioned patches are applied, but still no bit is coming throgh the hardware.
(btw. the procd/ubus-problem has gone)

one further thing is maybe worth mentioning: there's a pause of ~20s on startup between these two msgs:
`[   20.606625] rtl83xx_fib_event: FIB_RULE ADD/DEL for IPv6 not supported`
`[   41.109958] rtl838x-eth 1b00a300.ethernet eth0: Link is Down`
the serial terminal is usable during this pause and after it the system continues normally with initialization...
(I don't know if this is currently relevant for anything)

bevanweiss, 2025-09-02:
> [@bevanweiss](https://github.com/bevanweiss) what is the current RTL8221B state on your hasivo switch?

The PHY is detected in c45 mode, and reports link status etc well.
Still no traffic passing however, but that seems more SERDES related I suspect.

I do note that if I have the cable plugged in at boot it doesn’t seem to update LED status, and I need to unplug and replug.
I’m away from home (and the switch) for the next month however.

bendavid, 2025-09-02:
> That PR contains a bugfix that might help. No other knowledge around bringing up RTL8221B on RTL930x exists right now.

Right but if I understood the discussion correctly, https://github.com/openwrt/openwrt/pull/19518 in its current form will not solve the problem yet because it only applies to the SFP ports so far.

bevanweiss, 2025-09-02:
> It is the same code I use locally, which works for the RTL8226 in my XGS1010-12 (it is still necessary to initialize network in the bootloader using `rtk network on` though).

It doesn’t work out of the box on at least the Hasivo s1100wp-8gt (8x2.5gbps), but the ‘rtk network on’ doesn’t get uboot network up either…

So YMMV…

sh-dvl, 2025-09-02:
'current draft' is still the patch seen here: https://github.com/openwrt/openwrt/pull/19518/commits/98b45815c4591b632832a89d11a66de8c5adcb91 ?

and if there's any doubt: of course 'rtk network on' is always set.

if there's anything, which may be tested/viewed with original firmware, please tell me; I can switch via tftfboot to a version with simple shell-xs via serial port (have deactivated /bin/cli),

bendavid, 2025-09-02:
Ok yes this is working for me now on XGS1210-12 B1 with current main + https://github.com/openwrt/openwrt/pull/19518 + https://github.com/openwrt/openwrt/pull/19908 + appropriate DTS for rev B1

Linking and network traffic working properly at 2.5gbps

For concreteness: https://github.com/bendavid/openwrt/tree/XGS1210-12_B1

sh-dvl, 2025-09-03:
nothing changed.
any port but the 2,5Gb-ports works well.

wtf?

made a clone of https://github.com/bendavid/openwrt/, the checkout XGS1210-12_B1, selected the B1-target, call make and transfered the new image via tftpboot ...

sh-dvl, 2025-09-03:
mmmh, ok, no, only tested with 1GBit-RJ45 and 10Gbit-SFP so far, but I'll try it with 2,5Gb-peer soon

bevanweiss, 2025-09-06:
I personally think getting it into the OpenWRT mainline would be good sooner vs later, so making it non-draft would be good.
There might need to be some rework around MDIO stuff, but I think there needs to be some changes on MDIO also, to more closely align with how other upstream MDIO drivers represent multiple buses with common control registers (each bus appears as it’s own mii_bus instance).

closed 2026-06-19 by jonasjelonek