openwrt/openwrt #19646: WDS  broken after recent update of hostapd

grosjo, 2025-08-03:
### Describe the bug

I have several openwrt boxes connected by WDS

Upgrading to latest git the WDS client does not create probkems.
However, the WDS master does not transmit packets anymore to the attached interface ('lan')

### OpenWrt version

r30660-a2c361c520

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Yuncore CPE830

### Image kind

Official downloaded image

### Steps to reproduce

Configure master and clients in WDS mode. 
Connection gets established but packets are not forwarded to the related bridged interface

### Actual behaviour

Just upgrade to latest git

### Expected behaviour

WDS boxes exchanging packets as expected

### Additional info

Master config is :

```
# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/ahb/18100000.wmac'
	option htmode 'HT20'
	option cell_density '1'
	option channel '11'
	option country 'US'
	option txpower '30'
	option band '2g'
	list ht_capab 'SMPS-STATIC'

config wifi-iface 'wifinet7'
	option device 'radio0'
	option mode 'ap'
	option ssid 'PARKING_WDS'
	option encryption 'none'
	option wds '1'
	option network 'lan'
	list maclist 'xxx'
	list maclist 'yyy'
```

Client config is :
```
# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option band '2g'
	option htmode 'HT20'
	option path 'platform/ahb/18100000.wmac'
	option country 'US'
	option channel '11'
	option cell_density '1'
	option txpower '30'
	list ht_capab 'SMPS-STATIC'

config wifi-iface 'wifinet2'
	option device 'radio0'
	option mode 'sta'
	option wds '1'
	option ssid 'PARKING_WDS'
	option network 'lan'
	option encryption 'none'
```

### Diffconfig

```text
# ./scripts/diffconfig.sh
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_yuncore_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AES128 is not set
# CONFIG_DROPBEAR_AES256 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT is not set
CONFIG_DROPBEAR_DEBUG_TRACE=3
# CONFIG_DROPBEAR_ED25519 is not set
CONFIG_DROPBEAR_LASTLOG=y
CONFIG_DROPBEAR_LASTLOG_PATH="/var/log/lastlog"
# CONFIG_DROPBEAR_LEGACY_COMPAT is not set
CONFIG_DROPBEAR_LOG_COMMANDS=y
# CONFIG_DROPBEAR_REEXEC is not set
# CONFIG_DROPBEAR_SCP is not set
# CONFIG_DROPBEAR_SFTPSERVER is not set
# CONFIG_DROPBEAR_SMALL_CODE is not set
# CONFIG_DROPBEAR_SNTRUP761 is not set
# CONFIG_DROPBEAR_SVR_AGENTFWD is not set
# CONFIG_DROPBEAR_SVR_LOCALSTREAMFWD is not set
# CONFIG_DROPBEAR_SVR_LOCALTCPFWD is not set
# CONFIG_DROPBEAR_SVR_PUBKEY_OPTIONS is not set
# CONFIG_DROPBEAR_SVR_REMOTETCPFWD is not set
CONFIG_DROPBEAR_ZLIB=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CC_OPTIMIZE_FOR_PERFORMANCE is not set
CONFIG_KERNEL_CC_OPTIMIZE_FOR_SIZE=y
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EROFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EROFS_FS_SECURITY is not set
# CONFIG_KERNEL_EROFS_FS_XATTR is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_PAGE_MONITOR=y
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LINUX_6_12=y
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ATH_DFS is not set
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_apk-mbedtls is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_kmod-ag71xx=y
# CONFIG_PACKAGE_kmod-ag71xx-legacy is not set
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-net-selftests=y
CONFIG_PACKAGE_kmod-nft-bridge=y
CONFIG_PACKAGE_kmod-phylink=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmbedtls=m
CONFIG_PACKAGE_libubus-lua=y
# CONFIG_PACKAGE_libuclient is not set
CONFIG_PACKAGE_libustream-mbedtls=m
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
# CONFIG_PACKAGE_uclient-fetch is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-log=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_wpad-basic=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TESTING_KERNEL=y
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_libuclient is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_uclient-fetch is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

grosjo, 2025-08-03:
In case, the network config of the master

```
# cat /etc/config/network 

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config device
	option name 'br-lan'
	option type 'bridge'
	option ipv6 '0'
	option bridge_empty '1'
	list ports 'eth0'
	list ports 'eth1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.61'
	option netmask '255.255.0.0'
	list dns_search 'maduao.net'
	list dns '192.168.1.11'
	option gateway '192.168.1.11'

config device
	option name 'eth0'
	option ipv6 '0'
	option multicast '1'

config device
	option name 'eth1'
	option ipv6 '0'
	option multicast '1'

config device
	option name 'phy0-ap0'
	option ipv6 '0'
	option multicast '1'

config device
	option name 'phy0-ap1'
	option ipv6 '0'
	option multicast '1'

config device
	option name 'phy0-ap2'
	option ipv6 '0'

```

diizzyy, 2025-08-03:
I've seen similar issues on mediatek (Totolink A8000ru, mt7622), using https://github.com/openwrt/openwrt/commit/a2c361c5204883a154fc2fbc46505b37ce3925d6 (compiled locally).

WDS breaks and also communication between wireless clients, wired <--> wireless clients works however as intended.

A snapshot from June 5 works as intended

As a side note, configuration syntax for at least /etc/config/wireless seems to have changed over the past months. I did redo the configuration as I belived potential mismatches could've cause these issues.

diizzyy, 2025-08-03:
That commit isn't the issue, I just referrenced what I built and tested.
It appears like only WDS AP mode is affected, WDS STA (client) mode seems to operate fine.

nbd168, 2025-08-03:
Please try the latest version

diizzyy, 2025-08-03:
@nbd168 
Works on my end, both WDS AP mode and wireless clients can now reach each other.
Thanks for the quick fix!

grosjo, 2025-08-04:
Would that fix https://github.com/openwrt/openwrt/issues/17578 as well ?

grosjo, 2025-08-04:
@nbd168 ?

closed 2025-08-09 by grosjo