openwrt/openwrt #19651: realtek: TP-Link TL-ST1008F v2.0 lzma recipe

plappermaul, 2025-08-03:
The TP-Link build recipe reads.

```
define Device/tplink_tl-st1008f_v2
  SOC := rtl9303
  UIMAGE_MAGIC := 0x93030000
  DEVICE_VENDOR := TP-Link
  DEVICE_MODEL := TL-ST1008F
  DEVICE_VARIANT := v2.0
  DEVICE_PACKAGES := kmod-gpio-pca953x
  IMAGE_SIZE := 31808k
endef
TARGET_DEVICES += tplink_tl-st1008f_v2
```
This builds initramfs/sysupgrade with gzip. It should be possible to reuse the Vimin VM S100 recipe to build both images with lzma. Smaller images more flash left.

```
define Device/vimin_vm-s100-0800ms
  SOC := rtl9303
  UIMAGE_MAGIC := 0x93000000
  DEVICE_VENDOR := Vimin
  DEVICE_MODEL := VM-S100-0800MS
  IMAGE_SIZE := 13312k
  $(Device/kernel-lzma) <-- SIMPLY ADD THIS
endef
TARGET_DEVICES += vimin_vm-s100-0800ms
```

jonasjelonek, 2025-08-04:
Did as you suggested, boots properly after sysupgrade. Haven't actually checked the output during load (before Linux) nor how much space this saves but nothing to complain since it boots.

Can do a PR for this.

plappermaul, 2025-08-04:
Would be great. Not critical just a drive-by finding. Thanks in advance.