openwrt/openwrt #19674: LuCI fails to apply static route for 192.168.10.0/24 despite showing it in UI

alex8585, 2025-08-05:
### Describe the bug

When adding a static route via the LuCI web interface, a specific route — `192.168.10.0/24` — fails to apply properly. After clicking "Save & Apply", the route **appears in the LuCI interface**, but it is **not present in the system routing table** (`ip r` output).

Even after a system reboot, the LuCI UI continues to display the route as if it were successfully applied, but the system does not actually install it.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

QEMU Ubuntu 24.04 PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

1. Navigate to **Network > Static Routes** in LuCI.
2. Add a new route for `192.168.10.0/24` via the web interface.
3. Click **Save & Apply**.
4. Check the routing table using `ip r` or `ip route`.

[image]

### Actual behaviour

- The route **shows up in LuCI**, even after reboot.
- The route **is not present in the system routing table**.
- No validation errors were displayed during input or after submission.
- There were no warning messages shown.

### Expected behaviour

The route should be properly applied to the system and visible in the routing table.

- Improve validation to catch invalid or malformed input (e.g., trailing spaces).
- Ensure a rejected route is not falsely shown as active in LuCI if it failed to apply at the system level.

### Additional info

- Creating other routes **after** this one works fine — they are saved and applied as expected.
- After multiple attempts, I was able to workaround the issue:
    1. Changed the route to `192.168.10.0/16` — it saved and applied successfully.
    2. Then changed it back to `192.168.10.0/24` — this time it worked and appeared in the system routing table.

This suggests that something went wrong during the first attempt, possibly an **invisible character (like a trailing space)** in the form field. However:
- There was **no validation error** or indication of this in the UI.
- The route was **accepted into LuCI config** but **not committed to the system**.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

alex8585, 2025-08-14:
> Change route table from automatic to main ?
> 
> ```
> ip ru sh
> ip ro sh table local
> ip ro sh table main
> ip ro sh table default
> ```

root@OpenWrt102:~# ip ru sh
0:      from all lookup local
32766:  from all lookup main
32767:  from all lookup default
root@OpenWrt102:~# ip ro sh table local
local 10.0.0.2 dev wg0 proto kernel scope host src 10.0.0.2 
broadcast 10.0.0.255 dev wg0 proto kernel scope link src 10.0.0.2 
local 127.0.0.0/8 dev lo proto kernel scope host src 127.0.0.1 
local 127.0.0.1 dev lo proto kernel scope host src 127.0.0.1 
broadcast 127.255.255.255 dev lo proto kernel scope link src 127.0.0.1 
local 172.22.0.102 dev eth1 proto kernel scope host src 172.22.0.102 
broadcast 172.22.0.255 dev eth1 proto kernel scope link src 172.22.0.102 
local 192.168.16.102 dev eth0 proto kernel scope host src 192.168.16.102 
broadcast 192.168.16.255 dev eth0 proto kernel scope link src 192.168.16.102 
local 192.168.20.102 dev eth2 proto kernel scope host src 192.168.20.102 
broadcast 192.168.20.255 dev eth2 proto kernel scope link src 192.168.20.102 
root@OpenWrt102:~# ip ro sh table main
default via 172.22.0.1 dev eth1 proto static 
10.0.0.0/24 dev wg0 proto kernel scope link src 10.0.0.2 
172.22.0.0/24 dev eth1 proto kernel scope link src 172.22.0.102 
192.168.10.0/24 via 10.0.0.1 dev wg0 proto static 
192.168.15.100 via 192.168.16.101 dev eth0 proto static 
192.168.16.0/24 dev eth0 proto kernel scope link src 192.168.16.102 
192.168.20.0/24 dev eth2 proto kernel scope link src 192.168.20.102 
root@OpenWrt102:~# ip ro sh table default
Error: ipv4: FIB table does not exist.
Dump terminated
root@OpenWrt102:~# 

https://snipboard.io/TbSA6V.jpg
https://snipboard.io/sh61Iu.jpg

brada4, 2025-08-14:
Plese re-format to markdown code blocks `<>`
Route is totally there as much as i see.

alex8585, 2025-08-15:
> Plese re-format to markdown code blocks `<>` Route is totally there as much as i see.
<>
root@OpenWrt102:# ip ru sh
0: from all lookup local
32766: from all lookup main
32767: from all lookup default
root@OpenWrt102:# ip ro sh table local
local 10.0.0.2 dev wg0 proto kernel scope host src 10.0.0.2
broadcast 10.0.0.255 dev wg0 proto kernel scope link src 10.0.0.2
local 127.0.0.0/8 dev lo proto kernel scope host src 127.0.0.1
local 127.0.0.1 dev lo proto kernel scope host src 127.0.0.1
broadcast 127.255.255.255 dev lo proto kernel scope link src 127.0.0.1
local 172.22.0.102 dev eth1 proto kernel scope host src 172.22.0.102
broadcast 172.22.0.255 dev eth1 proto kernel scope link src 172.22.0.102
local 192.168.16.102 dev eth0 proto kernel scope host src 192.168.16.102
broadcast 192.168.16.255 dev eth0 proto kernel scope link src 192.168.16.102
local 192.168.20.102 dev eth2 proto kernel scope host src 192.168.20.102
broadcast 192.168.20.255 dev eth2 proto kernel scope link src 192.168.20.102
root@OpenWrt102:# ip ro sh table main
default via 172.22.0.1 dev eth1 proto static
10.0.0.0/24 dev wg0 proto kernel scope link src 10.0.0.2
172.22.0.0/24 dev eth1 proto kernel scope link src 172.22.0.102
192.168.10.0/24 via 10.0.0.1 dev wg0 proto static
192.168.15.100 via 192.168.16.101 dev eth0 proto static
192.168.16.0/24 dev eth0 proto kernel scope link src 192.168.16.102
192.168.20.0/24 dev eth2 proto kernel scope link src 192.168.20.102
root@OpenWrt102:# ip ro sh table default
Error: ipv4: FIB table does not exist.
Dump terminated
root@OpenWrt102:~#<>

brada4, 2025-08-16:
Can you yourself read it?