openwrt/openwrt #19680: Zyxel NWA50AX boot fail

Sonusss, 2025-08-05:
### Describe the bug

I tried to install OpenWrt onto Zyxel NWA50AX using openwrt-24.10.2-ramips-mt7621-zyxel_nwa50ax-squashfs-factory.bin

The flash process was OK but the unit don't boot properly

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zyxel NWA50AX

### Image kind

Official downloaded image

### Actual behaviour

I can connect using the serial console and see the booting process being interrupted with this error message:
```
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2025-08-06:
https://openwrt.org/toh/zyxel/nwa50ax_pro#important_note_regarding_installation

Sonusss, 2025-08-07:
Finally I solved the situation by making a fresh install from serial console as described here:
https://openwrt.org/inbox/toh/zyxel/nwa50ax?s[]=nwa50ax

The unit is fully functional however I keep having these errors during boot process in the serial console:
```
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
```

neheb, 2025-08-13:
I have no idea why there's an fw_printenv or fw_saveenv.

Djfe, 2025-10-21:
isn't that error displayed during the uboot phase of the boot process (before booting Linux) I feel like it's harmless.

please post a full log or close the issue if your issue has been solved :)

closed 2025-10-21 by Sonusss