openwrt/openwrt #19693: mac80211-regular error

fuqiang03, 2025-08-06:
### Describe the bug

make[3]: Entering directory '/media/ubuntu/OpenWrt/package/kernel/mac80211'
rm -f /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/.built
touch /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/.built_check
. /media/ubuntu/OpenWrt/include/shell.sh; var2file "V_mac80211_config" /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/.config
make -C "/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11" KCFLAGS="-fmacro-prefix-map=/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi=target-arm_cortex-a9+vfpv3-d16_musl_eabi -fno-caller-saves " HOSTCFLAGS="-O2 -I/media/ubuntu/OpenWrt/staging_dir/host/include -I/media/ubuntu/OpenWrt/staging_dir/hostpkg/include -I/media/ubuntu/OpenWrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Aug  6 10:29:03 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/media/ubuntu/OpenWrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/fast-classifier.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/gpio-button-hotplug.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/oaf.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/shortcut-fe.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/xtables-addons.symvers" CC="arm-openwrt-linux-muslgnueabi-gcc" KERNELRELEASE=6.12.40 EXTRA_CFLAGS="-I/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/include -fmacro-prefix-map=/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11=backports-6.14.11 " KLIB_BUILD="/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.12.40" MODPROBE=true KLIB=/lib/modules/6.12.40 KERNEL_SUBLEVEL=12 KBUILD_LDFLAGS_MODULE_PREREQ= allnoconfig
make[5]: 'Kconfig.versions' is up to date.
make[6]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
rm -rf /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/modules
make  -C "/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11" KCFLAGS="-fmacro-prefix-map=/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi=target-arm_cortex-a9+vfpv3-d16_musl_eabi -fno-caller-saves " HOSTCFLAGS="-O2 -I/media/ubuntu/OpenWrt/staging_dir/host/include -I/media/ubuntu/OpenWrt/staging_dir/hostpkg/include -I/media/ubuntu/OpenWrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Aug  6 10:29:03 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/media/ubuntu/OpenWrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/fast-classifier.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/gpio-button-hotplug.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/oaf.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/shortcut-fe.symvers /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symvers/xtables-addons.symvers" CC="arm-openwrt-linux-muslgnueabi-gcc" KERNELRELEASE=6.12.40 EXTRA_CFLAGS="-I/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/include -fmacro-prefix-map=/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11=backports-6.14.11 " KLIB_BUILD="/media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.12.40" MODPROBE=true KLIB=/lib/modules/6.12.40 KERNEL_SUBLEVEL=12 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/net/mac80211/cfg.o
In file included from ./include/linux/string.h:390,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/string.h:3,
                 from ./include/linux/bitmap.h:13,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/bitmap.h:3,
                 from ./include/linux/cpumask.h:12,
                 from ./include/linux/smp.h:13,
                 from ./include/linux/lockdep.h:14,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/lockdep.h:3,
                 from ./include/linux/spinlock.h:63,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/spinlock.h:3,
                 from ./include/linux/wait.h:9,
                 from ./include/linux/wait_bit.h:8,
                 from ./include/linux/fs.h:6,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/fs.h:3,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/skbuff.h:17,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/skbuff.h:3,
                 from ./include/linux/if_ether.h:19,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/backport-include/linux/if_ether.h:3,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/include/linux/ieee80211.h:19,
                 from /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:11:
In function 'fortify_memcpy_chk',
    inlined from 'copy_mesh_setup' at /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2534:2,
    inlined from 'ieee80211_join_mesh' at /media/ubuntu/OpenWrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mac80211-regular/backports-6.14.11/net/mac80211/cfg.c:2687:8:
./include/linux/fortify-string.h:571:25: error: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Werror=attr [... cut ...]
make[1]: *** [package/Makefile:177: /media/ubuntu/OpenWrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/ubuntu/OpenWrt'
make: *** [/media/ubuntu/OpenWrt/include/toplevel.mk:233：world] 错误 2

### OpenWrt version
[... 10 lines cut ...]

### Device

mwlwifi

### Image kind

Official downloaded image

### Expected behaviour

linksys 1900acs v2 
 KERNEL 6.12.40
Latest main branch
Am I the only one who has encountered this error????

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

fuqiang03, 2025-08-06:
[image]

[image]

openwrt supported devices
Use the official source code. This error only appeared in kernel version 6.12.

tk154, 2025-08-06:
If you're building with GCC 15, this could be a duplicate of #19469.

fuqiang03, 2025-08-07:
Ubuntu Environment

[image]

@tk154   It's indeed the same issue/problem.

closed 2025-08-07 by fuqiang03

brada4, 2025-08-07:
Please attach `diffconfig.sh` output (as text, and it is imperative you post text as text not as pictures)

This is NOT the gcc complicit with error.