openwrt/openwrt #19702: qualcommax - ipq60xx: Kernel memory leak on axt1800

flipreverse, 2025-08-07:
### Describe the bug

The Linux kernel 6.12 suffers from a memory leak. It is most likely located in the ath11k driver.
A quick check with kmemleak do not reveal any memory leaks. Given that, the memory seems to be referenced somewhere.

### OpenWrt version

r0-d802e63

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

GL.iNet GL-AXT1800

### Image kind

Self-built image

### Steps to reproduce

- Load the `ath11kk-ahb` kernel module
- Start at least one hostapd for either 2,4 GHz or 5 GHz
- Just wait
- Memory usage starts to increase

### Actual behaviour

- Kernel eats the memory. Presumably the ath11k driver.
- Behavior does not occur when the module is not loaded or no

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

flipreverse, 2025-08-07:
> How much it grows compared to v24.10.2?

I cannot relate it to the growth in v24.10.2.  I, however, can show use the memory usage over the past six hours.

[image]

flipreverse, 2025-08-12:
> how does swap look in meminfo?

Swap shall be fine since I reduces swapiness to 20.

[mem.txt](https://github.com/user-attachments/files/21737912/mem.txt)

Djfe, 2025-08-12:
I'm on mobile right now. could you create a similar graph for me please with all 3 swap entries? :)
That would help in seeing how they correspond to the ones above and if there is a memory leak still but it is written to disk instead now

flipreverse, 2025-08-13:
> I'm on mobile right now. could you create a similar graph for me please with all 3 swap entries? :) That would help in seeing how they correspond to the ones above and if there is a memory leak still but it is written to disk instead now

`SwapTotal`, `SwapFree` and `SwapCached`? What do you expect to see? Since I reduced swapiness to 20, I'd expect the kernel to use the swap space quite late.

brada4, 2025-08-28:
Please add "memtotal" as it shrinks when kernel allocates more slabs (yeah it changes)

flipreverse, 2025-08-28:
Unfortunately, `/proc/slabinfo` was neither enabled nor recorded. Shall I wait for a few more days to record `/proc/slabinfo`, or shall I plot the information from `/proc/meminfo`?

flipreverse, 2025-08-28:
Interesting. `MemTotal` in `/proc/meminfo` does not change. It stays at 418856 kB.

flipreverse, 2025-09-05:
@brada4 I'm currently recording `/proc/slabinfo`. Which info do you want me to plot?

Edmondena, 2026-02-02:
I see a similar behavior even today with 25.12.0-rc4 when running e.g. 2APs and with or without relay mode with one STA on 2G4 wifi interfaces like described and confirmed in more details here: [25.12.0-rc1+2+3+4 #21512](https://github.com/openwrt/openwrt/issues/21512)