openwrt/openwrt #19774: DHCP LAN IP Pool Overlaps with WAN Subnet When WAN Port Is Disconnected

Barry-baymax, 2025-08-14:
### Describe the bug

We noticed that when the WAN port is connected, the DHCP service on the LAN side correctly prevents assigning IP addresses from the WAN subnet. However, when the WAN port is disconnected, this restriction is lifted. This means the LAN DHCP pool can be configured to include IP addresses in the WAN subnet without any warning or restriction.
For example:
1.	Disconnect the WAN port.
2.	Set the LAN DHCP range to 10.0.0.1 ~ 10.255.255.254.
3.	Assign 10.8.6.102 as a reserved IP for a LAN device.
4.	Reconnect the WAN port.
At this point, the WAN interface still obtains 10.8.6.102 from the upstream network, resulting in an IP conflict with the LAN device.
Could you please advise if this is an intended behavior? If not, is there a known fix or configuration setting that can prevent LAN DHCP from assigning IPs from the WAN subnet, even when the WAN port is disconnected?
Thank you for your time and support.

### OpenWrt version

r0-bbce708

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq53xx/generic

### Device

Qualcomm Technologies, Inc. IPQ5332/RDP468/AP-MI01.6

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2025-08-14:
> At this point, the WAN interface still obtains 10.8.6.102 from the upstream network, resulting in an IP conflict with the LAN device.
Could you please advise if this is an intended behavior? If not, is there a known fix or configuration setting that can prevent LAN DHCP from assigning IPs from the WAN subnet, even when the WAN port is disconnected?

If the WAN port is down/disconnected, it does not have any configuration, so the LAN DHCP has no information about what IP addresses WAN may use.

The only way I can think of to avoid an invalid configuration here would be to reject a WAN network configuration that overlaps with LAN (and thus fail WAN configuration).

Barry-baymax, 2025-08-15:
> If the WAN port is down/disconnected, it does not have any configuration, so the LAN DHCP has no information about what IP addresses WAN may use.
>
> The only way I can think of to avoid an invalid [...]

The solution I’m considering is to add a check for the DHCP address pool range in the hotplug script, so that when the WAN interface comes up, it can verify whether the WAN IP address is valid. Does OpenWrt provide any built-in functionality for this?

brada4, 2025-08-15:
Ask your SDK sales place, OpenWrt does not offer such functionality.