openwrt/openwrt #19788: GL-MT6000 + irqbalance, logs path error.

K1m0n, 2025-08-15:
### Describe the bug

GL-MT6000 with irqbalance installed/enabled, daemon logs invalid paths on syslog.

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Using opkg install luci-app-irqbalance (ver# 25.222.74622~4d26a4d)
wich in turn also installs irqbalance (ver# 1.9.4-r1).

### Actual behaviour

Service enabled/running,
but on reboot/enable I get the following error on system log:

`Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ arch_timer(11) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ keys(24) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mdio-bus:01(61) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mdio-bus:07(62) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7530(81) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ ttyS0(116) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 15100000.ethernet(120) guessed as class 5
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 15100000.ethernet(121) guessed as class 5
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ wdt_bark(122) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/11230000.mmc/: No such file or directory
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 11230000.mmc(123) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7530-0:00(124) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7530-0:01(125) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7530-0:02(126) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7530-0:03(127) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ xhci-hcd:usb1(128) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/10320000.crypto/: No such file or directory
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 10320000.crypto(129) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/10320000.crypto/: No such file or directory
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 10320000.crypto(130) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/10320000.crypto/: No such file or directory
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 10320000.crypto(131) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/10320000.crypto/: No such file or directory
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ 10320000.crypto(132) guessed as class 0
Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: IRQ mt7915e(133) guessed as class 0`

To my (limited) understanding it seems it complains about not finding the 10320000.crypto on the specific path
wich I checked there is none under /sys/devices/platform/

I see the crypto kmod being succesfully loaded on bootlog as:

`Wed Aug 13 17:26:27 2025 kern.info kernel: [    8.595543] crypto-safexcel 10320000.crypto: EIP97:230(0,1,4,4)-HIA:270(0,5,5),PE:150/433(alg:7fcdfc00)/0/0/0`

and it does exist on the following paths:

`/sys/bus/platform/devices/10320000.crypto`
`/sys/bus/platform/drivers/crypto-safexcel/10320000.crypto`

So, I suppose the irqbalance searches for the module on the wrong path?
It seems a benign error, otherwise I see no ill-effects.

Maybe one of the devs would like to have a look.

### Expected behaviour

irqbalance starting without complaining for missing files on an invalid path.

`Wed Aug 13 17:26:30 2025 daemon.debug /usr/sbin/irqbalance: No directory /sys/devices/platform/10320000.crypto/: No such file`

### Additional info

I don't really know if this should be clasified as a bug,
could be a cosmetic issue.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-08-16:
irqbalance guesses device paths, from those device types starting from /proc/interrupts
Then sets permitted cores for each interrupt in /proc/irq/...
Can you set crypto device irq affinity to cpu0? To other CPU? If so the misdetection prevents intended resource balancing. Otherwise jyst cosmetic problem.

brada4, 2025-08-16:
you can add afalg engine and run openssl speed or instal/ (heavy) cryptsetup without any usb present and benchmark that.
lets start with trying to bind crypto accelerators to cpu0 and then to other cpu.
if you get like invalid value setting to last cpu you know that log barf is just noise.
save config and download prospective sysupgrade from luci-app-attendedsysupgrade to revert testing bloat.

brada4, 2025-08-16:
smp_affinity_list is "human" form like 0-1,3 where other os hex mask

K1m0n, 2025-08-16:
I gave it a try (and I learned a couple of things in the way),
command: `openssl speed -engine devcrypto aes-256-cbc`

no irqbalance:

`type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256-cbc      62617.20k   132294.40k   821273.60k  1390545.45k 14234910.72k 48677683.20k`

---

with irqbalance set all crypto interupts to cpu 0:

`type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256-cbc      41504.53k   152718.63k   685243.73k  2194344.23k 35548774.40k 48676864.00k`

^all interupts (129-133) indeed on cpu 0 (so cpu affinity works)

---

with smp_affinity set to cpus 1 to 4 for irqs 129 to 132 (129 to cpu 1, 130 to cpu 2, etc):

`type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256-cbc      41927.47k   154012.34k  2095104.00k  3876224.00k 23849369.60k 19550044.16k`

Curiously, most interupts still hit cpu 0, some cpu 2, rest empty.

---

with irqbalance @default settings (affinity f):

`type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-256-cbc      83449.07k   532505.60k   587296.91k  5182054.40k 23837081.60k 32573576.53k`

Looks better, but still most interupts hit cpu0.

---

Generally speaking, results are all over the place...
I might get 50% better/worst on a second run, but irqbalance seems to work.

I'm over my head with this :)
Omw to install a clean image, I'm pretty sure I broke things with my ignorance.

brada4, 2025-08-16:
The parsing bug actually protects particular device, but leave this open, if path issue gets fixed this device must stay unmoved on CPU0 anyway,
Dont close the issue, regression is behind the corner if cosmetic issue is fixed too blindly.

K1m0n, 2025-08-16:
Will do.
Thanks for looking into it.