openwrt/openwrt #19795: initramfs images too large for old devices

luizluca, 2025-08-17:
### Describe the bug

Some bootloaders cannot handle recent initramfs. For example TP-Link Archer C7-v2 can boot up to:

https://downloads.openwrt.org/releases/23.05.0-rc3/targets/ath79/generic/openwrt-23.05.0-rc3-ath79-generic-tplink_archer-c7-v2-initramfs-kernel.bin

```
done
Bytes transferred = 5557717 (54cdd5 hex)
## Booting image at 81000000 ...
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.127 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23389-5deed175a5) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Sat Aug 19 14:01:06 2023
```

But fails starting with next release (https://downloads.openwrt.org/releases/23.05.0-rc4/targets/ath79/generic/openwrt-23.05.0-rc4-ath79-generic-tplink_archer-c7-v2-initramfs-kernel.bin):

```
Bytes transferred = 5608924 (5595dc hex)
## Booting image at 81000000 ...
   Uncompressing Kernel Image ... 
( a reboot happened)
U-Boot 1.1.4 (Dec 17 2013 - 16:37:27)
```

My guess is that the bootloader simply cannot handle the ever-increasing images and broke between 5557717 and 5608924 bytes.

Is there any alternative to load larger initramfs? Something like chainload a better u-boot or somehow make kexec works? If we do not have a good alternative, I guess we could drop the initramfs for those devices once it surpass the bootloader limit.

### OpenWrt version

r30677

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

1. Download a recent initramfs image for Archer C7-v2 (it happens with many other devices)
2. Use the serial interface to stop the bootloader and load the initramfs image.
3. Try to boot

### Actual behaviour

4. The device reboots

### Expected behaviour

4. It should boot

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-08-17:
"we" do not ship that uboot.

plappermaul, 2025-08-17:
Speaking from realtek target point of view (MIPS32) but I guess you can use lzma-loader to circumvent that.

As lzma-loader is a little bit clumsy we forked it and made a more flexible one with simpler recipes. See https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=ccbff8bbdd1aaca3ad76fc129b8d53effb161183

This detects the SoC and the memory size on its own.

- https://github.com/openwrt/openwrt/blob/3491a4da6b9dd28fffbf2cd211337ead048e6682/target/linux/realtek/image/rt-loader/src/board.c#L44
- https://github.com/openwrt/openwrt/blob/3491a4da6b9dd28fffbf2cd211337ead048e6682/target/linux/realtek/image/rt-loader/src/board.c#L65

During initialization it copies itself and the kernel image to the highest possible memory address and restarts itself. From there it extracts the kernel back to the initial load adress. 

That in place it is as easy as this: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3e09991c079604b00640cded9f74e999d9b52d43

marcin-fm, 2025-08-17:
There is a related issue with EAP615-Wall (still persists, despite being closed): https://github.com/openwrt/openwrt/issues/19403