openwrt/openwrt #19811: Critical cache poisoning in Dnsmasq

Sunspark-007, 2025-08-19:
### Describe the bug

Hi,

https://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2025q3/018288.html

To address it they recommend:

> Detection mechanisms when upstream resolvers remain silent.
> Rate limiting and spoof-detection techniques, similar to those in PowerDNS.

### OpenWrt version

All

### OpenWrt release

All

### OpenWrt target/subtarget

All

### Device

All

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Sunspark-007, 2025-08-26:
Surprised this issue isn't getting any attention.

Guess the bot put the kibosh on it.

Djfe, 2025-08-30:
from IRC 

[image]