openwrt/openwrt #19814: wan DROP policy works for IPv4 but not for IPv6

alcoolfire, 2025-08-20:
### Describe the bug

Changing the `wan` zone's input and forward policies from the default REJECT to DROP correctly causes incoming IPv4 packets to be silently dropped when targeting the `wan` interface public IPv4 address (input). However, when incoming IPv6 packets target a `lan` host's public IPv6 address (forward) they are rejected by the firewall, *not dropped*.

My ISP does not allocate a public IPv6 address to the `wan` interface, so I could test IPv6 forward but not input.

My admittedly very novice reading of the `nftables` rule chains looks like at least the `wan` forward drop policy change isn't getting implemented correctly. I've been unable to find anything I might've done that would cause this, but I stand willing to be corrected.

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Steps to reproduce

1. Change `wan` zone input and forward policies from REJECT to DROP.

2. Initiate incoming IPv4 port scans of `wan` interface public IPv4 address using services such as <https://www.grc.com/shieldsup> and <https://dnschecker.org/port-scanner.php>, confirming they correctly get no responses (aka stealthy).

3. Confirm IPv4 DROP behaviour using `tcpdump` for the `wan` interface while repeating the same port scan.

4. Initiate incoming IPv6 port scans of `lan` host's public IPv6 address using service <https://ipv6.chappell-family.com/ipv6tcptest/>, observing that the majority of UDP and TCP ports are marked refused ("RFSD"). A minority of ports *may* be marked stealthy ("UDPSTEALTH" or "STLTH") for those ports blocked by one's ISP[^1].

5. Confirm erroneous IPv6 REJECT behaviour using `tcpdump` for the `wan` interface while repeating the same port scan.

[^1]: My ISP actually does block some ports and I've confirmed with `tcpdump` that there are *no* incoming packets for only those particular ports.

### Actual behaviour

Incoming IPv4 port scan packets of the `wan` interface public IPv4 address get no response packets of any type.

Incoming IPv6 port scan packets of a `lan` host's public IPv6 address are actively refused by the firewall. As expected, they don't get to the `lan` zone. Instead, UDP scans are responded to with ICMPv6 destination unreachable, unreachable port. TCP ports scans get TCP RST packets in response.

As can be seen in my `/etc/config/firewall` configuration (see **Additional info**), I've also chosen to disallow incoming ICMP and ICMPv6 echo requests, so those too should be silently dropped, yet ICMPv6 echo requests get an ICMPv6 destination unreachable, unreachable port response.

See **Additional info** for detailed actual behaviour seen while testing.

### Expected behaviour

As for IPv4, incoming IPv6 packets for any non-open UDP/TCP ports should be silently dropped when DROP policy is specified.

### Additional info

**Configuration and testing**[^2]

<details>
<summary>My `nftables` ruleset:</summary>

```sh
# nft list table inet fw4
table inet fw4 {
        flowtable ft {
                hook ingress priority filter
                devices = { eth0, phy0-ap0, phy0-ap1, phy1-ap0, phy1-ap1 }
                flags offload
[... 648 lines cut ...]

```sh
# tcpdump -i eth0.2 -n -v host 2001:470:971f:99::6 and port 113
tcpdump: listening on eth0.2, link-type EN10MB (Ethernet), snapshot length 262144 bytes
16:53:40.591624 IP6 (flowlabel 0xb75cc, hlim 54, next-header TCP (6) payload length: 40) 2001:470:971f:99::6.47104 > 2001:db8:1:2:3:4:5:6.113: Flags [S], cksum 0x5226 (correct), seq 2738552865, win 64660, options [mss 1220,sackOK,TS val 2364844020 ecr 0,nop,wscale 7], length 0
16:53:40.591854 IP6 (hlim 64, next-header TCP (6) payload length: 20) 2001:db8:1:2:3:4:5:6.113 > 2001:470:971f:99::6.47104: Flags [R.], cksum 0xea83 (correct), seq 0, ack 2738552866, win 0, length 0
```

10. Stop `tcpdump`.

[^2]: Some MAC and IPv4/6 addresses were sanitized.
[^3]: DNS Checker scanner source IPv4 address seems to vary.
[^4]: Chappell scanner source IPv6 address always seems to be the same (`2001:470:971f:99::6`).

</details>

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

GoetzGoerisch, 2025-08-21:
As far as I understood your expected and actual behavior this is intended.
As per RFC certain ICMPv6 message types MUST NOT be filtered.
The OpenWrt default rules are configured to this.

https://www.rfc-editor.org/rfc/rfc4890#section-4.4.1

brada4, 2025-08-21:
https://www.rfc-editor.org/rfc/rfc4443 sec 2.4.f (current "limits" are not workable currently. try https://github.com/openwrt/firewall4/pull/44

dave14305, 2025-08-21:
You need to change the general Forward policy from Reject to Drop also. The zone Forward policy only considers forwards among multiple interfaces in the same zone.

dave14305, 2025-08-22:
```
trace id 0f500d9a inet fw4 forward packet: iif "eth0.2" oif "br-lan" ether saddr 00:00:5E:00:53:b8 ether daddr 00:00:5E:00:53:e0 ip6 saddr 2001:470:971f:99::6 ip6 daddr 2001:db8:1:2:3:4:5:6 ip6 dscp cs0 ip6 ecn not-ect ip6 hoplimit 53 ip6 flowlabel 653951 ip6 nexthdr tcp ip6 length 40 tcp sport 43274 tcp dport 113 tcp flags == syn tcp window 64660
trace id 0f500d9a inet fw4 forward rule iifname "eth0.2" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic" (verdict jump forward_wan)
trace id 0f500d9a inet fw4 forward_wan rule jump drop_to_wan (verdict jump drop_to_wan)
trace id 0f500d9a inet fw4 drop_to_wan verdict continue
trace id 0f500d9a inet fw4 forward_wan verdict continue
trace id 0f500d9a inet fw4 forward rule jump handle_reject (verdict jump handle_reject)
trace id 0f500d9a inet fw4 handle_reject rule meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic" (verdict drop)
```
Some explanations of this IPv6 trace:

- With the default forward policy still set to REJECT, the `handle_reject` jump is added to the `forward` chain.
- With the wan zone forward policy set to DROP, the `drop_to_wan` jump in `forward_wan` is only checking if the outgoing interface matches a wan zone interface, which it does not.
- Since the IPv6 test traffic is destined for `br-lan`, it gets matched in the `handle_reject` chain.
- If you change `firewall.@defaults[0].forward='REJECT'` to `firewall.@defaults[0].forward='DROP'`, the `handle_reject` jump will not be created and the final verdict will be drop, as defined in the `forward` chain hook: `type filter hook forward priority filter; policy drop;`

IPv4 traffic works as expected because the public IP is on the wan interface and considered input traffic. As already mentioned, zone forward traffic is unintuitively limited to traffic forwarded between interfaces within a zone. The wan zone usually only has one interface, so it feels awkward trying to understand the purpose of `forward` in this scenario.

alcoolfire, 2025-08-22:
Apologies for the delay, but I've attached the trace I think @brada4 was asking for, in case it's still useful. It's rather big as it captures the entire <https://ipv6.chappell-family.com/ipv6tcptest/> test. You'll find that the test starts with an ICMPv6 echo-request, then does various UDP scans, followed by a lot of TCP scans.

The patch I used:

```
# diff -u /rom/etc/nftables.d/10-custom-filter-chains.nft /etc/nftables.d/10-custom-filter-chains.nft
--- /rom/etc/nftables.d/10-custom-filter-chains.nft     2025-06-23 13:40:36.000000000 -0700
+++ /etc/nftables.d/10-custom-filter-chains.nft 2025-08-22 00:00:43.000000000 -0700
@@ -1,7 +1,6 @@
 ## The firewall4 input, forward and output chains are registered with
 ## priority `filter` (0).

-
 ## Uncomment the chains below if you want to stage rules *before* the
 ## default firewall input, forward and output chains.

@@ -37,3 +36,25 @@
 #     ct state new log prefix "Firewall4 accepted egress: "
 # }

+## Uncomment the chains below to stage nftrace instrumentation
+## at the beginning of  both base chains filtering packets first
+## Example enables tracing of DNS packets to limit output volume
+## Adjust to Your requirements and observe output using
+## root@OpenWRT:~# nft monitor trace
+
+chain raw_prerouting {
+    type filter hook prerouting priority raw; policy accept;
+    jump trace_chain
+}
+
+chain raw_output {
+    type filter hook output priority raw; policy accept;
+    jump trace_chain
+}
+
+chain trace_chain {
+    ip6 saddr . ip6 daddr { 2001:470:971f:99::6 . ::/0, ::/0 . 2001:470:971f:99::6 } meta l4proto { tcp, udp } th sport . th dport != { 443 . 0-65535, 0-65535 . 443 } meta nftrace set 1
+    ip6 saddr . ip6 daddr { 2001:470:971f:99::6 . ::/0, ::/0 . 2001:470:971f:99::6 } meta l4proto icmpv6 meta nftrace set 1
+    # meta l4proto {tcp,udp} th saddr . th daddr { 113 . 0-65535, 0-65535 . 113 } meta nftrace set 1
+    # meta nftrace set 1
+}
```
I don't know why the router uses its `guest` interface IPv6 address for ICMPv6 responses, but it doesn't have a `wan` address available.

[ipv6-drop-failure,sanitized.log](https://github.com/user-attachments/files/21945780/ipv6-drop-failure.sanitized.log)

alcoolfire, 2025-08-22:
@dave14305 : Thank you, that's a great explanation. I wonder how many other people misread global "defaults" to be the defaults used for zones that do not explicitly override those defaults. Maybe it's not helping that the global default option and the zone option are both called `forward`?

If one had two independent WAN zones and wanted one of them to reject and the other to drop, how would that be achieved?

alcoolfire, 2025-08-26:
@dave14305 : I think it would be helpful if <https://openwrt.org/docs/guide-user/firewall/firewall_configuration#defaults> were to have at least a little conceptual explanation of what the default `input` and `forward` options mean in relation to their zone counterparts, especially considering users who might not even know what `nftables` is. I still don't have a solid mental model for it.

As a bug report, it wasn't a bug, so closing it makes sense. Thanks again.