openwrt/openwrt #19831: Xiami Mi router 3G issue wifi 2.4G 5G openwrt 24.10.2

arlaor, 2025-08-21:
### Describe the bug

Hi
I Try install openwrt 24.10.2 in my router
Model | Xiaomi Mi Router 3G
-- | --
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621

But it doesn't recognize or doesn't have the driver for the 2.4G and 5G Wi-Fi networks, as shown in the attached image.

[image]

In version 24.10.1, both 2.4G and 5G are recognized.

[image]

Has anyone had the same problem? Could you tell me how to fix it?

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-08-22:
There are no radical changes in wifi driver between 24.10.1 and .2
It usually happens if you overdo with KVR settings and then use default sysupgrade image and basic hostapd cannot handle them (same for dnsmasq full)

arlaor, 2025-08-22:
> There are no radical changes in wifi driver between 24.10.1 and .2 It usually happens if you overdo with KVR settings and then use default sysupgrade image and basic hostapd cannot handle them (same for dnsmasq full)

Hi @castiel652 @brada4 
Due to the need to enable the router at home, I didn't log the problem. I only installed the firmware 24.10.2 sysupgrade on the router's website. When I saw that I wasn't getting a Wi-Fi connection, I took screenshots and returned to the 24.10.1 sysupgrade on my router's website, where I got a Wi-Fi connection immediately.
However, something I've been noticing is that although I initially had a 5G connection, when I use it for extended periods of time, such as copying a certain amount of gigabytes through the 5G network, it disconnects. Although the SSID is displayed in the networks to choose from, it doesn't allow any devices to connect until I restart the router and it works normally again, until I use the Wi-Fi network intensively again. This happens to me on both a Xiaomi MiWiFi Mini router with OpenWRT 24.10.2 and a Xiaomi Mi Router 3G with OpenWRT 24.10.1. They have the same behavior.

arlaor, 2025-08-24:
> not sure what you "upgraded" - you run 24.10.1 and attemt to stuff .2 modules

No, I installed OpenWRT 24.10.2 using the ```sysupgrade -n``` command, and I'm attaching the information generated by the system. It's not the result of installing version 24.10.1.

brada4, 2025-08-25:
Please try again like using web interface, currenr installation still kas 24.10.1 kernel.

Djfe, 2025-08-25:
and then please provide `logread` on the first boot after sysupgrading. Hopefully the next sysupgrade fixes your broken installation. weird..

arlaor, 2025-08-26:
Hi @brada4 @Djfe 
This is the log with firmware 
```openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-kernel1.bin```
```openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-rootfs0.bin```

```root@OpenWrt:~# # Flashear kernel en partición kernel (mtd8)
root@OpenWrt:~# mtd write openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squa
shfs-kernel1.bin kernel
# Flashear rootfs en partición ubi (mtd9)
mtd write openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-rootfs0.bin ubi

# Limpiar overlay y configuración
mtd erase rootfs_dataCouldn't open image file: openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-kernel1.bin!
root@OpenWrt:~#
root@OpenWrt:~# # Flashear rootfs en partición ubi (mtd9)
root@OpenWrt:~# mtd write openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squa
shfs-rootfs0.bin ubi
Couldn't open image file: openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-rootfs0.bin!
root@OpenWrt:~#
root@OpenWrt:~# # Limpiar overlay y configuración
root@OpenWrt:~# mtd erase rootfs_data
Could not open mtd device: rootfs_data
Could not open mtd device: rootfs_data
root@OpenWrt:~# cd /tmp
root@OpenWrt:/tmp# # Flashear kernel en partición kernel (mtd8)
root@OpenWrt:/tmp# mtd write openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-s
quashfs-kernel1.bin kernel
 Flashear rootfs en partición ubi (mtd9)
mtd write openwrt-24.10.2Unlocking kernel ...

Writing from openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-kernel1.bin to kernel ...  [ ]-ramips-mt762[e]iaomi_mi-router-3g-squashfs-rootfs0.bin ubi

# Limpiar overl[w]y configuración
mtd erase rootfs_
root@OpenWrt:/tmp#
root@OpenWrt:/tmp# # Flashear rootfs en partición ubi (mtd9)
root@OpenWrt:/tmp# mtd write openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-s
quashfs-rootfs0.bin ubi
Unlocking ubi ...

Writing from openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-rootfs0.bin to ubi ...
root@OpenWrt:/tmp#
root@OpenWrt:/tmp# # Limpiar overlay y configuración
root@OpenWrt:/tmp# mtd erase rootfs_data
Could not open mtd device: rootfs_data
Could not open mtd device: rootfs_data
root@OpenWrt:/tmp# reboot
root@OpenWrt:/tmp# [  379.923146] br-lan: port 2(lan2) entered disabled state
[  379.930127] mt7530-mdio mdio-bus:1f lan1: left allmulticast mode
[  379.936182] mt7530-mdio mdio-bus:1f lan1: left promiscuous mode
[  379.942590] br-lan: port 1(lan1) entered disabled state
[  380.012276] mt7530-mdio mdio-bus:1f lan2: left allmulticast mode
[  380.018393] mtk_soc_eth 1e100000.ethernet eth0: left allmulticast mode
[  380.024919] mt7530-mdio mdio-bus:1f lan2: left promiscuous mode
[  380.031123] br-lan: port 2(lan2) entered disabled state
[  380.082879] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  380.285465] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[  384.461193] reboot: Restarting system

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2018 HackPascal <hackpascal@gmail.com>
Build date 2018-10-14 [git-9463f9c]
Version 1.1 (r1237)

DRAM: 256MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Xiaomi R3G
Clocks: CPU: 880MHz, DDR: 1200MHz, Bus: 220MHz, Ref: 40MHz
Environment variables @ 00060000 on flash bank 0, size 00020000
Flash: ESMT NAND 128MiB 3.3V 8-bit (128MB) on mt7621-nfi.0
[... 610 lines cut ...]
[   34.101403] mt7530-mdio mdio-bus:1f lan1: configuring for phy/gmii link mode
[   34.114930] br-lan: port 1(lan1) entered blocking state
[   34.120402] br-lan: port 1(lan1) entered disabled state
[   34.125840] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
[   34.132327] mtk_soc_eth 1e100000.ethernet eth0: entered allmulticast mode
[   34.142624] mt7530-mdio mdio-bus:1f lan1: entered promiscuous mode
[   34.189331] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   34.199117] br-lan: port 2(lan2) entered blocking state
[   34.204451] br-lan: port 2(lan2) entered disabled state
[   34.210073] mt7530-mdio mdio-bus:1f lan2: entered allmulticast mode
[   34.219550] mt7530-mdio mdio-bus:1f lan2: entered promiscuous mode
[   34.248626] mt7530-mdio mdio-bus:1f wan: configuring for phy/gmii link mode
[   37.733005] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[   37.741125] br-lan: port 2(lan2) entered blocking state
[   37.746344] br-lan: port 2(lan2) entered forwarding state
```
I have also tried to perform the installation with the commands
```sysupgrade -F -n --no-restore-backup -p openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-sysupgrade.bin```
```
also
```firstboot
reboot
sysupgrade -n -F -p openwrt-24.10.2-ramips-mt7621-xiaomi_mi-router-3g-squashfs-sysupgrade.bin```

the same result, not WIFI

DragonBluep, 2025-08-26:
Please follow the wiki installation guide and make sure you have the stock bootloader.

arlaor, 2025-09-24:
I performed the firmware update following the instructions provided in OpenWRT for my router. The Wi-Fi drivers were installed correctly, and I had access to the Wi-Fi network for a certain period of time. After that, access was lost, and I had to reboot the router again to access the Wi-Fi. This became a constant. I read in several forums that OpenWRT's support for my router's Wi-Fi chip wasn't adequate, which is why I was experiencing constant disconnections.
I uninstalled OpenWRT and installed X-WRT, which has a more up-to-date kernel in its firmware and is a bit more stable. For now, I'll stick with this version until OpenWRT improves support for my router, if it does in a new version. Thank you for your cooperation.

sapphiressss, 2025-11-11:
delete tirectory /bin