openwrt/openwrt #19841: ipq40xx/mikrotik snapshot: MikroTik RBcAPGi-5acD2nD (cAP ac) not booting

kmarzic, 2025-08-24:
### Describe the bug

Hi,
MikroTik RBcAPGi-5acD2nD (cAP ac) is not booting with OpenWRT snapshot image after flashing, running 6.12.40 kernel.
Booting using offical snapshot `openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-initramfs-kernel.bin` is going OK.
After flashing official snapshot `openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-squashfs-sysupgrade.bin` it is not possible to access device on 192.168.1.1.

### OpenWrt version

r30830-62d50fb196

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MikroTik cAP ac

### Image kind

Official downloaded image

### Steps to reproduce

Outputs from device after booting `openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-initramfs-kernel.bin`:

[capac.txt](https://github.com/user-attachments/files/21955547/capac.txt)

Steps:

```
$ cd Downloads
$ wget https://downloads.openwrt.org/snapshots/targets/ipq40xx/mikrotik/openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-initramfs-kernel.bin
$ sudo dnsmasq -kd -p 0 -C /dev/null -u nobody --enable-tftp --interface=enp0s31f6   --dhcp-range=192.168.1.50,192.168.1.100,255.255.255.0,1h --dhcp-host=192.168.1.200   --dhcp-boot="openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-initramfs-kernel.bin" --tftp-root="/home/kmarzic/Downloads/"
dnsmasq-dhcp: DHCPDISCOVER(enp0s31f6) 74:4d:28:62:c5:c4
dnsmasq-dhcp: DHCPOFFER(enp0s31f6) 192.168.1.87 74:4d:28:62:c5:c4
dnsmasq-dhcp: DHCPREQUEST(enp0s31f6) 192.168.1.87 74:4d:28:62:c5:c4
dnsmasq-dhcp: DHCPACK(enp0s31f6) 192.168.1.87 74:4d:28:62:c5:c4
dnsmasq-tftp: sent /home/kmarzic/Downloads/capac/openwrt-ipq40xx-mikrotik-mikrotik_cap-ac-initramfs-kernel.bin to 192.168.1.87
$ ssh root@192.168.1.1
BusyBox v1.37.0 (2025-08-23 17:22:53 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r30830-62d50fb196
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------

 OpenWrt recently switched to the "apk" package manager!

 OPKG Command           APK Equivalent      Description
 ------------------------------------------------------------------
 opkg install <pkg>     apk add <pkg>       Install a package
 opkg remove <pkg>      apk del <pkg>       Remove a package
 opkg upgrade           apk upgrade         Upgrade all packages
 opkg files <pkg>       apk info -L <pkg>   List package contents
 opkg list-installed    apk info            List installed packages
 opkg update            apk update          Update package lists
 opkg search <pkg>      apk search <pkg>    Search for packages
 ------------------------------------------------------------------

For more https://openwrt.org/docs/guide-user/additional-software/opkg-to-apk-cheatsheet

root@OpenWrt:~#
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

markmentovai, 2025-10-04:
I’m experiencing this too across several units, with the ipq40xx MikroTik wAP ac (RBwAPG-5HacD2HnD), using `openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade.bin`. The last working image I built was at 1f8391346e1 (2025-07-08). The image at db0b0e14be4c (2025-08-22) is not. 750f2abd8865 (2025-08-17) switched ipq40xx to kernel 6.12, and is in between these two builds.

I’m actually able to boot the newer 6.12 images via bootp/tftp (`openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-initramfs-kernel.bin`), but once I flash a sysupgrade image (`openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade.bin`), the system won’t boot it.

markmentovai, 2025-10-04:
I thought I’d compare a working sysupgrade image (2025-07-08) to a non-working one (2025-08-22), but yafut seems to extract nothing but zeroes from the working image.

```
% …/openwrt/staging_dir/host/bin/yafut -B 64k -C 1040 -EPS -d openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250708.bin -r -i /kernel -o openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250708.bin_kernel
% file openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250708.bin_kernel
openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250708.bin_kernel: data
% hexdump -C openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250708.bin_kernel
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
0036f400  00 00 00 00 00 00 00 00                           |........|
0036f408
```

The non-working one extracts as an ELF image, everything seemingly intact:

```
% …/openwrt/staging_dir/host/bin/yafut -B 64k -C 1040 -EPS -d openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250822.bin -r -i /kernel -o openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250822.bin_kernel
% file openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250822.bin_kernel
openwrt-ipq40xx-mikrotik-mikrotik_wap-ac-squashfs-sysupgrade-20250822.bin_kernel: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
```

Cc @kempniu because I’m not sure what to make of yafut extracting all of those zeroes, when I can clearly see the ELF data in the sysupgrade image, and it is in fact bootable.

Cc @john-tho because I know they’ve done a lot of work with MikroTik sysupgrade images, and they might have thoughts on this.

john-tho, 2025-10-05:
Watchdog is not being pet fast enough…
See https://github.com/openwrt/openwrt/commit/cdf2e21e82d012c0a8b01214e6788008a2d4a506#diff-1395eaa836e1eda8e5d651ed7e9a5d8b4330e297f92ab6a3094c43bfd717f550L6
Due to df59427a1122 ("ARM: qcom: merge remaining subplatforms into sensible Kconfig entry") https://github.com/torvalds/linux/commit/99497df59427a11222571034afebf2f0f7f9a3e2
https://github.com/openwrt/openwrt/blob/18bdeda0113d28170a20e550f45641450089a070/target/linux/ipq40xx/patches-6.12/302-arm-compressed-set-ipq40xx-watchdog-to-allow-boot.patch#L35

Suggestion would be to change this to a new KConfig entry (and introduce it as part of that specific patch), and enable it in ipq40xx Mikrotik subtarget kernel config.

I dug out my 450gx4 (which is a NAND device, but easy to console to start with…), rebased my changes, enabled `DEBUG_LL` in OpenWrt and kernel configs, so I could see `DEBUG_UNCOMPRESSED` output being interrupted mid-uncompress. To make this more confusing, on this device `RouterBOOT backup booter 6.45.9` booted 6.12 fine, `RouterBOOT booter 7.16` did not…