openwrt/openwrt #19858: Add support to uutils' coreutils

mhalano, 2025-08-25:
Uutils is a project that wants to create basic GNU tools written in Rust. One of them is the coreutils itself: https://github.com/uutils/coreutils
Today, it's not possible to compile OpenWrt using uutils because it explicitily requires the GNU version.
The idea would be to find out how we can support compilation using uutils instead of just GNU's coreutils.
Please notice they are not trying to create any compiler, like gcc ou llvm, and nothing like that.
Basically, commands like `install` returns a different string for `--version`, so that's why it breaks the compilation.

brada4, 2025-08-26:
Can it build e.g ubuntu gcc itself?

mhalano, 2025-08-26:
I will do a test and try to compile GCC packages for Ubuntu using uutils' coreutils instead of GNU's coreutils.

brada4, 2025-08-26:
It requires patches from you, until there is a full release with these coreutils.

mhalano, 2025-08-26:
The first problem I found was related to this rejected patch: https://patchwork.ozlabs.org/project/openwrt/patch/6a7864fe2b504093f657442e1849f57324858764.1437346144.git.larry@elder-gods.org/
I applied, and now I'm doing a `make dirclean` compilation. After that I will do a `make distclean` compilation.

mhalano, 2025-08-27:
@brada4 Just to let you know, I used that reject patch that allowed me to bypass the pre-requisite check and use `install` from uutils instead of GNU, and I completed successfully a build after a distclean, so it also compiled the tooling.
I can send you more details about my config file or anything else.
About waiting for a full release, I don't think it's relevant because the tool is already available, it's not exclusivity of Ubuntu, probably is available on Arch, I don't know, but Ubuntu using it will give more visibility to uutils.

brada4, 2025-08-27:
You can make a pr that accepts your experimental toolset alongside normal one.

mhalano, 2025-08-27:
Yeah, but I don't know what the best approach.
Maybe just removing the requirement for `GNU install` would be enough?
GCC itself looks for a BSD-compliant install command, which both GNU and uutils are.