openwrt/openwrt #19870: WiFi interface wlan0 not showed under "ifconfig -a" command

NCSTomWu, 2025-08-26:
### Describe the bug

Same HW and WiFi module, 
using OpenWrt 23.05, wifi interface "wlan0" showed under output of command "ifconfig -a";
using OpenWrt 24.10.1 and OpenWrt 24.10.2, wifi interface "wlan0" not showed under output of command "ifconfig -a", but phy0 related info showed under output of command "iw phy"

wifi module: 
WLE900VX (using ath10k related driver and firmware)
WMX7205 (usnig ath11k related driver and firmware)

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

x86/64

### Device

0

### Image kind

Official downloaded image

### Steps to reproduce

1. Using openwrt 24.10.2 as default os image
2. After booting into os, installing the following IPKs (take WLE900VX as example)
  ath10k-board-qca988x_20241110-r2_x86_64.ipk
  ath10k-firmware-qca988x_20241110-r2_x86_64.ipk
  hostapd-common_2024.09.15~5ace39b0-r2_x86_64.ipk
  iw_6.9-r1_x86_64(1).ipk
  iwinfo_2024.10.20~b94f066e-r1_x86_64.ipk
  kmod-ath10k_6.6.93.6.12.6-r2_x86_64.ipk
  kmod-ath_6.6.93.6.12.6-r2_x86_64.ipk
  kmod-cfg80211_6.6.93.6.12.6-r2_x86_64.ipk
  kmod-crypto-aead_6.6.93-r1_x86_64.ipk
  kmod-crypto-ccm_6.6.93-r1_x86_64.ipk
  kmod-crypto-cmac_6.6.93-r1_x86_64.ipk
  kmod-crypto-ctr_6.6.93-r1_x86_64.ipk
  kmod-crypto-gcm_6.6.93-r1_x86_64.ipk
  kmod-crypto-geniv_6.6.93-r1_x86_64.ipk
  kmod-crypto-gf128_6.6.93-r1_x86_64.ipk
  kmod-crypto-ghash_6.6.93-r1_x86_64.ipk
  kmod-crypto-hmac_6.6.93-r1_x86_64.ipk
  kmod-crypto-manager_6.6.93-r1_x86_64.ipk
  kmod-crypto-null_6.6.93-r1_x86_64.ipk
  kmod-crypto-rng_6.6.93-r1_x86_64.ipk
  kmod-crypto-seqiv_6.6.93-r1_x86_64.ipk
  kmod-crypto-sha3_6.6.93-r1_x86_64.ipk
  kmod-crypto-sha512_6.6.93-r1_x86_64.ipk
  kmod-mac80211_6.6.93.6.12.6-r2_x86_64.ipk
  ucode-mod-nl80211_2025.07.18~3f64c808-r1_x86_64.ipk
  ucode-mod-rtnl_2025.07.18~3f64c808-r1_x86_64.ipk
  wifi-scripts_1.0-r1_all.ipk
  wireless-regdb_2025.07.10-r1_all.ipk
3. Reboot

### Actual behaviour

Interface wlan0 not showed under output of command "ifconfig -a"
Interface wlan0 showed after using iw command: "iw phy phy1 interface add wlan0 type managed"

### Expected behaviour

Interface wlan0 should show under output of command "ifconfig -a"

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-08-27:
Sorry but it is long gone. You will get phy0-ap0 -sta0 -mesh0 -mon0 style netdevs since v22 and no default netdev.

NCSTomWu, 2025-08-29:
> Sorry but it is long gone. You will get phy0-ap0 -sta0 -mesh0 -mon0 style netdevs since v22 and no default netdev.

"What is the 'v22' you mentioned? Is it a version of OpenWRT?
If it is a version of OpenWRT, as I tested, **I do not need to configure the net device manually on OpenWRT 23.05, and the interface wlan0 appears without any manual configuration process, but I need to configure the net device manually on OpenWRT 24.10**.
If it is not a version of OpenWRT, then what is the v22 you mentioned?
And what I want to know is whether this phenomenon is because of an update to the configuration process from OpenWRT 23.05 to 24.10, or some other reason?"