openwrt/openwrt #19885: ASU and Firmware Selector do not work for customized images for OpenWrt-23.05.6

Ro-1-2-3, 2025-08-27:
### Describe the bug

It’s been a week since the release of  OpenWrt-23.05.6, but none of the imagebuilder containers for its targets had yet been created (with the exception of ramips/mt7621 target, which was created manually, as a response to somebody’s open issue for that particular target).

As an example, the mediatek/filogic target container, is missing, so one gets errors when trying to build a 23.05.6 image with a custom package selection, both via router ASU and via Firmware Selector website.

### OpenWrt version

23.05.6

### OpenWrt release

23.05.6

### OpenWrt target/subtarget

All

### Device

All

### Image kind

Official downloaded image

### Steps to reproduce

[image]

### Actual behaviour

See the error message on the above screenshot.

### Expected behaviour

Both ASU and OpenWrt Firmware Selector site are expected to generate customized images of the official release of OpenWrt-23.05.6.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-08-28:
I have triggered the 23.05.6 container builds, some failed but most suceeded so it should work now

Ro-1-2-3, 2025-08-28:
I still see only Snapshots being built and posted in the past several days, and none of 23.05.6:

https://github.com/openwrt/docker/pkgs/container/imagebuilder/versions

robimarko, 2025-08-29:
Works for me

[image]