openwrt/openwrt #19886: Layer 2 connectivity issue on RTL9301-based switch (24×1G RTL8218D + 4×10G SFP+) using OpenWrt (bootm boot)

UntronixLabProject, 2025-08-27:
### Describe the bug

Hello,
I built an OpenWrt image for a switch based on Realtek RTL9301, featuring:

- 24 × 1 Gbps copper ports via 3 × RTL8218D PHYs
- 4 × 10 Gbps SFP+ ports

I am attaching the DTS file.
```markdown
    #include "rtl930x.dtsi"
    #include <dt-bindings/input/input.h>
    #include <dt-bindings/gpio/gpio.h>
    #include <dt-bindings/leds/common.h>
    #include <dt-bindings/thermal/thermal.h>

    / {
        compatible = "realtek,rtl9301-soc", "realtek,rtl9301";
        model = "IQTools Test SE";
        
        aliases {
            i2c0 = &i2c_m1b0;
            i2c1 = &i2c_m1b1;
            i2c2 = &i2c_m1b2;
            i2c3 = &i2c_m1b3; /* будет /dev/i2c-3 */
            i2c4 = &i2c_m1b4;
            i2c5 = &i2c_m1b5;
            i2c6 = &i2c_m1b6;
            serial0 = &uart0;
            led-boot = &led_pwr_sys;
            led-failsafe = &led_pwr_sys;
            led-running = &led_pwr_sys;
            led-upgrade = &led_pwr_sys;
        };
        keys {
            compatible = "gpio-keys";
            
            mode {
                label = "reset";
                gpios = <&gpio0 22 GPIO_ACTIVE_LOW>;
                linux,code = <KEY_RESTART>;
            };
        };
        
        leds {
            compatible = "gpio-leds";
            pinctrl-names = "default";
            pinctrl-0 = <&pinmux_disable_sys_led>;
            
            led_pwr_sys: led-0 {
                label = "green:power";
                color = <LED_COLOR_ID_GREEN>;
                function = LED_FUNCTION_POWER;
                gpios = <&gpio0 0 GPIO_ACTIVE_HIGH>;
            };
        };
        
        led_set: led_set {
            compatible = "realtek,rtl9300-leds";
            active-low;
            
            // LED set 0:
            // Amber: 100M/10M
            // Yellow: 1G
            led_set0 = <0x0a20 0x0b80>;
            
            led_set1 = <0x0b80 0x0a20 0x0a08 0x0a08>;

            led_set2 = <0x0a2a 0x0a0b>;
        };
[... 640 lines cut ...]
            phy17:  ethernet-phy@17  { reg = <17>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 17>; sds=<3>; };
            phy18: ethernet-phy@18 { reg = <18>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 18>; sds=<3>; };
            phy19: ethernet-phy@19 { reg = <19>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 19>; sds=<3>; };
            phy20: ethernet-phy@20 { reg = <20>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 20>; sds=<3>; };
            phy21: ethernet-phy@21 { reg = <21>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 21>; sds=<3>; };
            phy22: ethernet-phy@22 { reg = <22>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 22>; sds=<3>; };
            phy23: ethernet-phy@23 { reg = <23>;  compatible = "ethernet-phy-ieee802.3-c22";  rtl9300,smi-address = <2 23>; sds=<3>; };
            phy24: ethernet-phy@24 { reg = <24>; compatible = "ethernet-phy-ieee802.3-c22"; sfp = <&sfp1>; phy-is-integrated; sds=<4>; };
            phy25: ethernet-phy@25 { reg = <25>; compatible = "ethernet-phy-ieee802.3-c22"; sfp = <&sfp2>; phy-is-integrated; sds=<6>; };
            phy26: ethernet-phy@26 { reg = <26>; compatible = "ethernet-phy-ieee802.3-c22"; sfp = <&sfp3>; phy-is-integrated; sds=<8>; };
            phy27: ethernet-phy@27 { reg = <27>; compatible = "ethernet-phy-ieee802.3-c22"; sfp = <&sfp4>; phy-is-integrated; sds=<9>; };
        };
        
    };
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

plappermaul, 2025-08-29:
There is no "xsgmii" in kernel and "10gbase-r" is totally wrong.

UntronixLabProject, 2025-08-29:
> There is no "xsgmii" in kernel and "10gbase-r" is totally wrong.

I tried usxgmii, doesn't work. About sfp I know that recommended way to specify sfp+ ports is using "1000base-x". But currently i am checking l2 on coppers, not on sfp. But, if we speak about sfp, if i specify "1000base-x" and connect sfp with 10G ethtool show me incorrect speed 1G:)