openwrt/openwrt #19889: could support [Filogic] Banana Bpi r4 lite?

ChaingTsung, 2025-08-28:
Here are docs: https://docs.banana-pi.org/en/BPI-R4_Lite/BananaPi_BPI-R4_Lite
Would you like to add support for this?

castiel652, 2025-08-28:
Are you going to donate the board? :)

brada4, 2025-08-28:
Once page links to actual dts source similarities to similar device dts sources can be assessed. Extracting those from live images then normalizing would be a huge heap of extra effort.

ChaingTsung, 2025-08-29:
### Here are official Source Code & Resources 
- BPI-R4 lite DXF file
https://drive.google.com/file/d/1bEzBUy4ktir53J4RIpUKbVqUf0ZLdm_d/view?usp=sharing
- BPI-R4 lite Schematic
https://drive.google.com/file/d/13RfESOKzXvb0-AZOTVkzU_NrCzcAGu2N/view?usp=sharing
- MT7987A_Wi-Fi_Router_Platform_Datasheet_Public_V1.1
https://drive.google.com/file/d/1Z2QHEvRhfjXCOofliSFezqWhvl2te_nI/view?usp=sharing

System Image
- OpenWRT MTK MP4.2 wifi SDK wifi driver (2 PCIE ，both have 1 Lane）https://drive.google.com/file/d/1PaSYl5P3nP1rep0jmm_nwEuOaTVePGT4/view?usp=sharing
- OpenWRT MTK MP4.2 wifi SDK wifi driver for BE14000 Wifi Card(1 PCIE,2 Lane） https://drive.google.com/file/d/11H7mjv5RAxq_xDv1i7EeGw7LuFIN_APY/view?usp=sharing

> Once page links to actual dts source similarities to similar device dts sources can be assessed. Extracting those from live images then normalizing would be a huge heap of extra effort.

Rondom, 2025-08-30:
Sources are in MTK SDK mtk-openwrt-feeds

justinthegeek, 2025-10-23:
Does this help? https://github.com/BPI-SINOVOIP/BPI-R4Lite-OPENWRT-V24.10.0-Master-Devel

Rondom, 2025-10-28:
#20304

dangowrt, 2025-11-25:
R4 Lite is fully supported now (apart from offloading features)