openwrt/openwrt #19905: Verse 24.10.2 - AX53U WAN does not work. Only downgrading to 22.03.6 seems to work

kermola, 2025-08-30:
### Describe the bug

For some reason, WAN does not seem to connect on 24.10.2, (or any of 23.xx either).
The moment I flash it back to 22.03.6, it works straight away.
I have tried resetting the config, but that doesn't make a difference.

This is the log.

> Sat Aug 30 14:36:48 2025 daemon.notice netifd: Interface 'wan6' has link connectivity
> Sat Aug 30 14:36:48 2025 daemon.notice netifd: Interface 'wan6' is setting up now
> Sat Aug 30 14:36:48 2025 kern.info kernel: [ 1483.175897] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
> Sat Aug 30 14:36:48 2025 daemon.notice netifd: wan (23471): udhcpc: started, v1.36.1
> Sat Aug 30 14:36:48 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
> Sat Aug 30 14:36:48 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
> Sat Aug 30 14:36:48 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
> Sat Aug 30 14:36:48 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
> Sat Aug 30 14:36:48 2025 daemon.notice netifd: wan (23471): udhcpc: broadcasting discover
> Sat Aug 30 14:36:49 2025 kern.info kernel: [ 1484.205776] mtk_soc_eth 1e100000.ethernet wan: Link is Down
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: Network device 'wan' link is down
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: Interface 'wan6' has link connectivity loss
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: wan (23471): udhcpc: received SIGTERM
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: wan (23471): udhcpc: entering released state
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: wan (23471): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: Interface 'wan' is now down
> Sat Aug 30 14:36:49 2025 daemon.notice netifd: Interface 'wan6' is now down
> Sat Aug 30 14:36:50 2025 daemon.warn odhcpd[1756]: No default route present, overriding ra_lifetime to 0!
> Sat Aug 30 14:36:51 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
> Sat Aug 30 14:36:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
> Sat Aug 30 14:36:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
> Sat Aug 30 14:36:51 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: Network device 'wan' link is up
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: Interface 'wan' has link connectivity
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: Interface 'wan6' has link connectivity
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: Interface 'wan6' is setting up now
> Sat Aug 30 14:37:05 2025 kern.info kernel: [ 1500.845809] mtk_soc_eth 1e100000.ethernet wan: Link is Up - 100Mbps/Full - flow control rx/tx
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: wan (23908): udhcpc: started, v1.36.1
> Sat Aug 30 14:37:05 2025 daemon.notice netifd: wan (23908): udhcpc: broadcasting discover
> Sat Aug 30 14:37:06 2025 kern.info kernel: [ 1501.885755] mtk_soc_eth 1e100000.ethernet wan: Link is Down
> Sat Aug 30 14:37:06 2025 daemon.notice netifd: Network device 'wan' link is down
> Sat Aug 30 14:37:06 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
> Sat Aug 30 14:37:06 2025 daemon.notice netifd: Interface 'wan6' has link connectivity loss
> Sat Aug 30 14:37:06 2025 daemon.notice netifd: wan (23908): udhcpc: received SIGTERM
> Sat Aug 30 14:37:06 2025 daemon.notice netifd: wan (23908): udhcpc: entering released state
> Sat Aug 30 14:37:07 2025 daemon.notice netifd: wan (23908): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
> Sat Aug 30 14:37:07 2025 daemon.notice netifd: Interface 'wan' is now down
> Sat Aug 30 14:37:07 2025 daemon.notice netifd: Interface 'wan6' is now down

### OpenWrt version

r20265-f85a79bcb4

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Flash to 24.10.2
Try to connect to the internet

### Actual behaviour

Flash to 24.10.2
WAN does not connect.

### Expected behaviour

Flash to 24.10.2
Try to connect to the internet

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-08-30:
Is 100Mbps connection to ithe nternet intentional?
could you please format log segment as a code block `<>`, in present form it is totally unreadable.

brada4, 2025-08-30:
Also try resetting router configuration, between 22 and 23 wan port is switched to second soc gigabit circuit and complex vlan configs may not be always migrated over.

brada4, 2025-09-03:
I am asking for 5th time - please reset in version 24 or sysupgrade to v24 not keeping configuration. 
Then in luci check that wan port is connected with ip  address or not.

kermola, 2025-09-03:
The WAN Port did not connect, and had no IP address.
However, on a hunch, I changed the WAN MAC address... and everything seems to be working?
Any idea why that would make it work?

Thanks

brada4, 2025-09-03:
Question to your ISP and you - what you did to them to block your mac address.

kermola, 2025-09-03:
But... that MAC address works fine when I use 22.03.6. I cross checked and the MAC address was exactly the same.

As in, the AX53u on openwrt 22.03.6 works fine, WAN DHCP gets ip address etc.
then I sysupgrade to 24.10.2, and WAN doesn't work. But, the moment I change the MAC address... it suddenly works?

I'm confused. But anyway, it seems to be working fine?
Thanks for your assistance.

closed 2025-09-06 by kermola