openwrt/openwrt #19922: Cudy WR1300 v1 bridge broken after 23.05.6

rfc1036, 2025-09-01:
### Describe the bug

After upgrading from 23.05 to 24.10.2 (or else to today's snapshot) networking breaks.
With some work I managed to access the AP with SSH using a link local address and figured out that it failed to add other interfaces (both wired and wireless) to the bridge connected to the CPU interface (eth0).

### OpenWrt version

?

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Cudy WR1300 v1

### Image kind

Official downloaded image

### Steps to reproduce

I tried upgrading from 23.05.*, first to 24.10.2 and after that to today's snapshot.
Downgrading back to 23.05.6 (r24232-539228933c) made everything work again.

### Actual behaviour

Networking is broken because only the CPU interface `eth0` is connected to the `br-lan` bridge.
The other interfaces work fine, but trying to manually add them to `br-lan` with the `brctl` command fails with:

 ```
brctl: bridge test: Invalid argument
```

The opposite is true as well: I created a new bridge and added all the other interfaces to it, but then trying to add `eth0` (after removing it from `br-lan`) would fail with the same message.

This is **not** a user interface issue: I have verified all this from the command line.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-09-01:
eth0 is certainly not a bridge. Is it dsa master interface?

brada4, 2025-09-01:
I dont see eth0 anywhere. What br-lan member list is in /etc/config/network ?