openwrt/openwrt #19937: busybox: multiple vulnerabilities present in 1.36.1 - please upgrade to the latest version

rafalfitt, 2025-09-02:
### Describe the bug

I used the new Firmware analysis in Microsoft Azure and some vulnerabilities were found in busybox.

CVE-2023-42363 - https://bugs.busybox.net/show_bug.cgi?id=15865
CVE-2023-42364 - https://bugs.busybox.net/show_bug.cgi?id=15868
CVE-2023-42365 - https://bugs.busybox.net/show_bug.cgi?id=15871
CVE-2023-42366 - https://bugs.busybox.net/show_bug.cgi?id=15874

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Steps to reproduce

[image]

[image]

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rafalfitt, 2025-09-02:
perhaps you can use these patches:

https://github.com/alpinelinux/aports/blob/8a4bf5971168bf48201c05afda7bee0fbb188e13/main/busybox/0001-awk-fix-use-after-free-CVE-2023-42363.patch
https://github.com/alpinelinux/aports/blob/8a4bf5971168bf48201c05afda7bee0fbb188e13/main/busybox/CVE-2023-42364-CVE-2023-42365.patch
https://github.com/alpinelinux/aports/blob/master/main/busybox/0026-awk.c-fix-CVE-2023-42366-bug-15874.patch

brada4, 2025-09-02:
When your BZ links revive we be able to see if those 3 fixes actually wrap 4 bugs. Obviously you can rebuild bb, just copy patches in patches directory, and bymp minor patchlevel for in-place upgrade.