openwrt/openwrt #19938: RAMIPS: kmalloc Redzone overwritten

rrvirtual, 2025-09-02:
### Describe the bug

I got some redzone errors on ramips ath79 device with OpenWRT 24.

- Snapshot: r28597-0425664679
- Linux version: 6.6.86
- Release: 24.10.1
- target: ramips ath79

During the kernel initialization process, while initializing USB, I got the following error:
```
[    3.774034] =============================================================================
[    3.782521] BUG kmalloc-128 (Not tainted): kmalloc Redzone overwritten
[    3.789270] -----------------------------------------------------------------------------
[    3.789270] 
[    3.799234] 0x81b18812-0x81b1881f @offset=2066. First byte 0x6b instead of 0xcc
[    3.806797] Allocated in usb_get_device_descriptor+0x2c/0x94 age=0 cpu=0 pid=154
[    3.814476]  __kmem_cache_alloc_node+0x80/0x138
[    3.819197]  kmalloc_trace+0x18/0x24
[    3.822906]  usb_get_device_descriptor+0x2c/0x94
[    3.827690]  hub_port_init+0x5a0/0xad0
[    3.831570]  hub_event+0x974/0xeac
[    3.835092]  process_scheduled_works+0x1f0/0x2e0
[    3.839886]  worker_thread+0x2d4/0x320
[    3.843775]  kthread+0xdc/0xe8
[    3.846946]  ret_from_kernel_thread+0x14/0x1c
[    3.851453] Freed in spi_mem_exec_op+0x364/0x3c4 age=0 cpu=0 pid=1
[    3.857851]  __kmem_cache_free+0x80/0xc0
[    3.861928]  spi_mem_exec_op+0x364/0x3c4
[    3.865986]  spi_mem_no_dirmap_read+0x80/0xa0
[    3.870493]  spinand_read_from_cache_op+0x158/0x1a4
[    3.875555]  spinand_read_page+0x80/0xb0
[    3.879623]  spinand_mtd_read+0x430/0x488
[    3.883781]  mtd_read_oob+0xfc/0x1a4
[    3.887498]  mtd_read+0x5c/0x84
[    3.890752]  ubi_io_read+0x16c/0x314
[    3.894478]  ubi_io_read_vid_hdr+0x94/0x1f8
[    3.898816]  ubi_attach+0x440/0x12b0
[    3.902524]  ubi_attach_mtd_dev+0x68c/0xa48
[    3.906861]  ubi_init_attach+0x180/0x2e8
[    3.910921]  do_one_initcall+0x80/0x26c
[    3.914890]  kernel_init_freeable+0x230/0x264
[    3.919423]  kernel_init+0x20/0x10c
[    3.923034] Slab 0x807e0600 objects=10 used=7 fp=0x81b18980 flags=0x800(slab|zone=0)
[    3.931044] Object 0x81b18800 @offset=2048 fp=0x00000000
[    3.931044] 
[    3.938066] Redzone  81b18780: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.947137] Redzone  81b18790: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.956215] Redzone  81b187a0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.965292] Redzone  81b187b0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.974367] Redzone  81b187c0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.983444] Redzone  81b187d0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    3.992520] Redzone  81b187e0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.001597] Redzone  81b187f0: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.010675] Object   81b18800: 12 01 01 02 02 00 00 40 eb 03 04 24 00 01 01 02  .......@...$....
[    4.019750] Object   81b18810: 03 01 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b  ..kkkkkkkkkkkkkk
[    4.028826] Object   81b18820: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.037903] Object   81b18830: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.046979] Object   81b18840: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.056056] Object   81b18850: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.065133] Object   81b18860: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.074208] Object   81b18870: cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc cc  ................
[    4.083286] Redzone  81b18880: cc cc cc cc                                      ....
[    4.091287] Padding  81b188b4: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
[    4.100362] Padding  81b188c4: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
[    4.109440] Padding  81b188d4: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
[    4.118517] Padding  81b188e4: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  ZZZZZZZZZZZZZZZZ
[    4.127592] Padding  81b188f4: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a              ZZZZZZZZZZZZ
[    4.136313] CPU: 0 PID: 154 Comm: kworker/0:4 Not tainted 6.6.86 #0
[    4.142801] Workqueue: usb_hub_wq hub_event
[... 278 lines cut ...]
### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ath79

### Device

ath79

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rrvirtual, 2025-09-02:
The target is Atheros ATH79, ramips, with the Generic subtarget. It's a modified device.

brada4, 2025-09-03:
Also it is up to you and only you to address memory diagnostics produced by enabling wild diag options
```
CONFIG_KERNEL_SLUB_DEBUG=y
CONFIG_KERNEL_SLUB_DEBUG_ON=y
CONFIG_KERNEL_SLABINFO=y
```

Djfe, 2025-09-03:
> > CONFIG_AG71XX_LEGACY=y
> 
> Sorry, you have to contact your BSP supplier, that is OpenWrt 19.07.0 at latest.

what makes you think that? OpenWrt main uses this, too
https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/config-6.12#L2