openwrt/openwrt #19950: DNS stops working once 2.4GHz WiFi is enabled

C0deView, 2025-09-03:
### Describe the bug

Once enable the default 2.4GHz WiFi OpenWrt, can ping 192.168.1.1, but cannot ping www.google.com. After disable the default 2.4GHz WiFi OpenWrt, www.google.com can be accessed in browser again.

### OpenWrt version

24.10.2

### OpenWrt release

r28739-d9340319c6

### OpenWrt target/subtarget

 ipq806x

### Device

Linksys EA8500 WiFi Router

### Image kind

Self-built image

### Steps to reproduce

Build an image with extra packages (listed in Diffconfig) on website https://firmware-selector.openwrt.org/ and install it first. Just enable the default 2.4GHz WiFi OpenWrt on LuCI, then open a browser to access www.google.com.

### Actual behaviour

Cannot access www.google.com.

### Expected behaviour

Can access www.google.com.

### Additional info

For my self-built image, I have lots of additional packages, but I don't change anything else after installation. This bug also can be reproduced with commands only:
ping -c 3 www.google.com
uci set wireless.radio1.disabled='0'
uci set wireless.default_radio1.disabled='0'
uci commit
service network restart
ping -c 3 www.google.com

### Diffconfig

```text
-dnsmasq dnsmasq-full kmod-tcp-bbr ip-full px5g-mbedtls libuhttpd-mbedtls luci-ssl luci-lib-px5g usbutils block-mount kmod-usb-ohci kmod-usb-uhci kmod-usb3 kmod-usb-storage kmod-usb-storage-uas kmod-fs-ext4 kmod-fs-exfat kmod-fs-vfat kmod-fs-ntfs ntfs-3g kmod-nls-cp437 kmod-nls-iso8859-1 e2fsprogs fdisk usbutils luci-app-hd-idle kmod-fs-btrfs btrfs-progs hdparm bonniexx kmod-usb-net kmod-usb-net-rndis usb-modeswitch kmod-usb-net-cdc-ether kmod-usb-acm luci-proto-ncm kmod-usb-net-pegasus r8152-firmware kmod-crypto-sha256 kmod-usb-net-cdc-ncm kmod-usb-net-rtl8152 kmod-libphy kmod-usb-net-asix-ax88179 nano sshfs picocom wget libustream-mbedtls libcurl4 curl ca-certificates coreutils-base64 ca-bundle luci-i18n-base-zh-cn luci-i18n-uhttpd-zh-cn luci-app-uhttpd luci-app-ddns luci-app-upnp vnstat2 luci-app-vnstat2 luci-i18n-vnstat2-zh-cn nlbwmon luci-app-nlbwmon luci-i18n-nlbwmon-zh-cn iftop wireless-tools iwinfo netperf speedtest-netperf openvpn-openssl libopenssl3 libopenssl-conf kmod-cryptodev kmod-crypto-authenc libopenssl-devcrypto iperf3-ssl mailsend libnghttp2-14 travelmate luci-app-travelmate procps-ng-ps watchcat luci-app-watchcat kmod-crypto-user iptables-nft ip6tables-nft kmod-nft-tproxy kmod-nft-netdev kmod-nft-compat libiptext-nft0 xtables-nft kmod-nft-bridge nft-qos luci-app-nft-qos kmod-ipt-conntrack-extra kmod-ipt-ipopt kmod-macvlan mwan3 luci-app-mwan3
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

C0deView, 2025-09-04:
I fixed the wrong information already. Please remove the 'invalid' label.

brada4, 2025-09-04:
Are you able to reproduce the problem with default package set? Notably base-files are absent from your list amd it is surprising your router booted up at all.

C0deView, 2025-09-04:
> Are you able to reproduce the problem with default package set? Notably base-files are absent from your list amd it is surprising your router booted up at all.

As I mentioned, I build my image on the website (https://firmware-selector.openwrt.org/) and put EXTRA packages in Diffconfig. That means the default package set generated by the website automatically for my WiFi router by default is not included in Diffconfig. Because I need a custom build from beginning, a build without those extra packages makes no sense for me. I have never test with the default package set yet ...... But in my test, to enable the default 5GHz WiFi OpenWrt will not cause this problem (I updated the bug description). For this reason, I believe there is a bug and reported. It seems those extra packages are not the root cause of this bug.

brada4, 2025-09-04:
typical mistake is not setting up "network" where wifi should bind, ie. once you start unconnected wifi AP any connections to that fail to see internet.

brada4, 2025-09-04:
Log (logread -e hostap) shoould go like
association-authentication-dhcp

C0deView, 2025-09-07:
> Log (logread -e hostap) shoould go like association-authentication-dhcp

It is not found in the output of the command. I updated my report with details of the output from the command. Please take a look. Thank you again.

brada4, 2025-09-07:
Please attach wireless config file (erase passwords)
I see 11b checkbox on could be problematic esp with modern devices,.

C0deView, 2025-09-16:
> > wireless.radio1.htmode='VHT20'
> > This is not a standard wifi mode. Please select one via luci.

In luci, the default wireless operating frequency mode is 'Legacy', no 'Width' option available. After changed mode to 'N', 'VHT20' can be changed into 'HT20' or 'HT40' in option 'Width'. But after test, I found the change of this option didn't help - the problem is still there.

brada4, 2025-09-16:
Something very fishy about 2ghz vht20 - it looks like you restored non-openwrt backup, like immortalwrt or similar.

brada4, 2025-09-17:
Please reset and setup from zero. VHT20 at 2GHz is not possible to enter via luci , nor start hostapd after it is edited in.