openwrt/openwrt #19952: EEE on BPI-R4

jpsollie, 2025-09-04:
### Describe the bug

after enabling EEE on any of the BPI-R4 lan ports, the systems grows unstable pretty quick:
```
ethtool --show-eee lan3
EEE settings for lan3:
        EEE status: disabled
        Tx LPI: disabled
        Supported EEE link modes:  100baseT/Full
                                   1000baseT/Full
        Advertised EEE link modes:  Not reported
        Link partner advertised EEE link modes:  100baseT/Full
                                                 1000baseT/Full
root@APBureau4:~# ethtool --set-eee lan3 eee on
[172556.382140] mt7530-mmio 15020000.switch lan3: Link is Down
root@APBureau4:~[172556.387819] br-lan: port 3(lan3) entered disabled state
# [172559.749959] mt7530-mmio 15020000.switch lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[172559.749990] br-lan: port 3(lan3) entered blocking state
[172559.763662] br-lan: port 3(lan3) entered forwarding state

root@APBureau4:~# [172640.571309] 
[172640.571309] ==================== FE ====================
[172640.578292] 0x00000000: 810000b0 00000000 01000020 00000000
....
[172641.931295] mtk_soc_eth 15100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 5430 ms
....
```
the complete dmesg (register dump) can be attached, but I'm uncertain whether it has any added value
disabling EEE makes the link work again

### OpenWrt version

r0+30880-44f76177d5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Banana Pi BPI-R4 (2x SFP+)

### Image kind

Self-built image

### Steps to reproduce

enable EEE on any of the DSA ports
enable EEE on the link partner

### Actual behaviour

the system grows unstable, the link drops and both device and link partner show issues with NETDEV watchdog transmit queue timeout

### Expected behaviour

the link keeps up but consumes less power

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_ALL_KMODS=y
[... 1380 lines cut ...]
CONFIG_MOLD=y
# CONFIG_OPENSSL_WITH_ERROR_MESSAGES is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_SRP is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-sha1=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_kmod-nf-log6=y
CONFIG_PACKAGE_kmod-nf-reject6=y
CONFIG_PACKAGE_kmod-nfnetlink=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_libnghttp2=m
CONFIG_PACKAGE_openwrt-keyring=y
# CONFIG_PACKAGE_perf is not set
CONFIG_PACKAGE_rtl8814a-firmware=m
CONFIG_PACKAGE_usign=y
# CONFIG_USE_MOLD is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jpsollie, 2025-09-04:
do you want my feeds.conf file?
```
src-git packages https://git.openwrt.org/feed/packages.git
src-git luci https://git.openwrt.org/project/luci.git
src-git routing https://git.openwrt.org/feed/routing.git
#src-git telephony https://git.openwrt.org/feed/telephony.git
src-git printing https://github.com/jpsollie/openwrt-printing-packages.git
```

brada4, 2025-09-04:
mtk erratum says to disable eee on some parts where it is known to hang. Basically you are documenting issue once again.

brada4, 2025-09-06:
The essence of silicon bug is that switch hangs when it receives low speed PDU, the bug requiring quirk override would be that some component combination would not display adversities.