openwrt/openwrt #19954: Unable to authenticate OpenWRT forum Sign Up

scotthep, 2025-09-04:
I did some work on the Trendnet TEW-810DR and the Linksys EA7300v2 and subsequently upgraded DSL -> Cable.  I lost my original DSL email account and have periodically tried to re-enroll at the forum.  I'm interested in completing Roland Reinl's work on the Linksys MR6350 and getting it into the source tree.  It seems like all the infrastructure is there from the MR8300.  It needs tweaks on the ath10 firmware and the update scripts.  Hopefully the initial install can be done with the TFTP+serial open-the-case which some similiar Linksys IPQ40x use.

Is there an alternative way to authenticate forum sign up?

I'd prefer to work through the issues on the forum in regards to accessible documentation but if it is not possible to work-around the email issue, I'll open a developer issue on the github site.

thess, 2025-09-04:
I could use another hint -- Do you have an old forum account that you want to keep and lost email access? What problem are you having? Have you tried creating a new forum account? If you do, DM me with the old account info and we can go from there.

If you are having some other strange issue - please supply some particulars. What did you try and what failed. Please let us know what messages you got?

scotthep, 2025-09-05:
I had an old account with an email address that is no longer in existance - at least that I'm aware of and can access.
I attempted to make a new forum account but the email never arrives at XXXX@spectrum.net.

I do not have "other" throw-away" email accounts  and would rather not deal with google.

I've tried off and on for roughly 2 years but my Linksys-EA7300v2 went catatonic.  The MR6350 is available refurbish for ~$30USD and it has more cpu/flash/ram.  My cable max's out at 550Mbps so it is a perfect fit.

thess, 2025-09-05:
What is your old account name? Is the new name `s xh  ex  xp  px  xe  xr`? (without x's and spaces)

scotthep, 2025-09-05:
> If you do not get an email from me regarding your account, your email address
> you used to register on the forum is not valid. If you don't give a working
> email, I will have to remove that account and let you register a new one.

Thanks,

A link to one of my posts when I was registered as "shep"
https://forum.openwrt.org/t/solved-new-isp-and-linksys-a-b-dual-boot-wont-boot-openwrt/115358/3

I closed my centrylink account and lost access to the email address
associated with it.

I then tried to register with a spectrum.com email account as shepper.
As you noted, the emails bounce but as you can see from this reply, it
is a real, working email account.

-- 
J. Scott Heppler

Penguin Innovations

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 

NOTICE: This e-mail message and any attachments may
contain legally privileged and confidential information intended
solely for the use of the intended recipients. If you are not an
intended recipient, you are hereby notified that you have
received this message in error and any review, dissemination,
distribution, copying, or other unauthorized use of this email
and any attachment is strictly prohibited. If you have received
this email in error, please notify the sender immediately and
delete the message and any attachments from your system.

closed 2025-09-05 by thess