openwrt/openwrt #19975: Arcadyan AW1000 Modem virtual DHCP interface does not start properly

GithPK, 2025-09-07:
### Describe the bug

Arcadyan AW1000 does not work properly with the last two compilations:
SNAPSHOT (r30963-79d3db7447)
SNAPSHOT (r30972-0ee0846be1)
Modem virtual DHCP interface does not start properly and there is no internet connection available.

Works properly with previous SNAPSHOT r30910-886382ba25

### OpenWrt version

r30972-0ee0846be1

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Arcadyan AW1000

### Image kind

Official downloaded image

### Steps to reproduce

Use attended sysupgrade, upgrade the unit with the mentioned snapshot, then try if the LTE/5G modem connection works after upgrade process and restart.

### Actual behaviour

The virtual modem DHCP interface is not available.

### Expected behaviour

Everything should works fine with no additional actions after restart after the upgrade process.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mariabankso, 2025-09-07:
Can you help me with reproduce?

GithPK, 2025-09-07:
> Can you help me with reproduce?

How can I help you?

mrkiko, 2025-09-08:
Maybe a bootlog with a working and non-working image could help? Or, in general, more info are needed.
I do not have the device, but my guess is - the modem is a Quectel in ECM mode or something?

GithPK, 2025-09-18:
Hi,

Please find below two logs:

Modem does not work...
`
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.12.45 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 14.3.0 r31091-af3ae2ae84) 14.3.0, GNU ld (GNU Binutils) 2.44) #0 SMP Wed Sep 17 21:04:09 2025
[    0.000000] Machine model: Arcadyan AW1000
[    0.000000] OF: reserved mem: 0x0000000040000000..0x0000000040ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x000000004a400000..0x000000004a5fffff (2048 KiB) nomap non-reusable tzapp@4a400000
[    0.000000] OF: reserved mem: 0x000000004a600000..0x000000004a9fffff (4096 KiB) nomap non-reusable bootloader@4a600000
[    0.000000] OF: reserved mem: 0x000000004aa00000..0x000000004aafffff (1024 KiB) nomap non-reusable sbl@4aa00000
[    0.000000] OF: reserved mem: 0x000000004ab00000..0x000000004abfffff (1024 KiB) nomap non-reusable smem@4ab00000
[    0.000000] OF: reserved mem: 0x000000004ac00000..0x000000004affffff (4096 KiB) nomap non-reusable memory@4ac00000
[    0.000000] OF: reserved mem: 0x000000004b000000..0x0000000050efffff (97280 KiB) nomap non-reusable wcnss@4b000000
[    0.000000] OF: reserved mem: 0x0000000050f00000..0x0000000050ffffff (1024 KiB) nomap non-reusable q6_etr_dump@50f00000
[    0.000000] OF: reserved mem: 0x0000000051000000..0x00000000510fffff (1024 KiB) nomap non-reusable m3_dump@51000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 20 pages/cpu s43288 r8192 d30440 u81920
[    0.000000] pcpu-alloc: s43288 r8192 d30440 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 pcie_dev=RG50 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Unknown kernel command line parameters "pcie_dev=RG50", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 262144
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007eb00000-0x000000007ec00000] (1MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000000] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000130] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000144] pid_max: default: 32768 minimum: 301
[    0.005279] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.005293] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.010285] rcu: Hierarchical SRCU implementation.
[    0.010294] rcu: 	Max phase no-delay instances is 1000.
[    0.010593] Timer migration: 1 hierarchy levels; 8 children per group; 1 crossnode level
[    0.010993] smp: Bringing up secondary CPUs ...
[    0.011654] Detected VIPT I-cache on CPU1
[    0.011762] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[... 793 lines cut ...]
[   36.881383] ath11k c000000.wifi phy1-ap0: entered promiscuous mode
[   36.887519] br-lan: port 6(phy1-ap0) entered blocking state
[   36.893660] br-lan: port 6(phy1-ap0) entered forwarding state
[   36.899404] br-lan: port 6(phy1-ap0) entered disabled state
[   37.271332] nss-dp 3a001400.dp3 lan3: PHY Link up speed: 1000
[   37.271394] br-lan: port 3(lan3) entered blocking state
[... cut ...]
[   39.971486] ath11k c000000.wifi phy1-ap1: left promiscuous mode
[   39.977759] br-lan: port 7(phy1-ap1) entered disabled state
[   39.984727] br-lan: port 7(phy1-ap1) entered blocking state
[   39.988859] br-lan: port 7(phy1-ap1) entered disabled state
[   39.994540] ath11k c000000.wifi phy1-ap1: entered allmulticast mode
[   40.000204] ath11k c000000.wifi phy1-ap1: entered promiscuous mode
[   40.006589] br-lan: port 7(phy1-ap1) entered blocking state
[   40.012487] br-lan: port 7(phy1-ap1) entered forwarding state	`

GithPK, 2025-09-19:
@mrkiko is the log suitable for debugging?