openwrt/openwrt #19981: netifd upgrade softbricked Turris Omnia

yarda, 2025-09-07:
### Describe the bug

This was the last what I have seen (I was running `opkg` inside the `screen`):
```
upgrading netifd on root from 2024.12.17~ea01ed41-r1 to
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/arm_cortex-a9_vfpv3-d16/base/netifd_2025.05.23~7901e66c-r1_arm
```
Then all LEDs except the power LED turned off and I had no network wired/wireless, reboot or power cycling didn't help.

Unfortunately, I was unable to enter the failsafe mode, because the router has only reset button which seems to always reset the router no matter when it is pushed. I also never received the UDP packet signalling the the button should be pressed.

I was able to repair the router by connection through the serial port. The router was in the state that it didn't show terminal on the serial port. I had to hit the 'f' key early in the boot process, enter failsafe mode, reinstall the `netifd` (it showed several errors), reboot. Then I got the normal shell on the serial line after boot, I reinstalled the `netifd` package again (now without errors) and I finally got the router working.

As I now have installed the same version that soft bricked my router (netifd-2025.05.23~7901e66c-r1), I am really confused what happened. What is the best practice to avoid such failures next time?

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade netifd from 2024.12.17~ea01ed41-r1 to netifd_2025.05.23~7901e66c-r1.

### Actual behaviour

Router soft bricked.

### Expected behaviour

Package upgraded, no softbrick.

### Additional info

I am going to upgrade from the RC release to the stable release soon, but I think it shouldn't happen even on the RC release.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-09-08:
Never happens if you ever dare to read the documentation.

https://openwrt.org/meta/infobox/upgrade_packages_warning

use owut, luci attendedsysupgrade etc to sysupgrade all at once.

yarda, 2025-09-25:
> If you are accessing the terminal from the network (e.g. ssh, ttyd), you should not modify the network (including network configuration and updating netifd), because this will cause the terminal to be disconnected and the ongoing changes may be killed, resulting in a half-brick. If you only modify the network configuration (/etc/config/network), `reboot` the system for it to take effect, instead of using `/etc/init.d/network reload`

It was in the screen session, so IMHO the disconnection shouldn't be a problem. It must be something else. In the forum I noticed some complaints on the netifd package causing the same problem that happened to me, so I reported it.

brada4, 2025-09-25:
Does the failsafe button function?

yarda, 2025-09-25:
> Does the failsafe button function?

No, but it maybe normal for Omnia. There are two buttons on the Omnia - the reset button and the LED brightness button. I guess both are hard wired. Because no matter when I pushed it, the router always reset (if the reset button is pushed) or the LEDs change brightness (if the brigthness button is pushed). I didn't notice any "special" LED pattern signalling the button should be pushed. I also tried to catch the UDP packet signalling that the button should be pushed, but no packet appeared.