openwrt/openwrt #19996: netifd: wifi uses wrong bridge?

schuettecarsten, 2025-09-09:
### Describe the bug

I have several VLANs and bridges on my device. I can see that my wifi connects to the wrong bridge, but the config looks absolutely correct. Here are some screenshots, followed by config files.

Wireless config says "network mgmt":

[image]

```
config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option network 'mgmt'
```

Interfaces show that the wifi "radio0" is connected to "guest":

[image]

```
config device
        option type 'bridge'
        option name 'br-guest'
        option igmp_snooping '1'
        option ipv6 '1'
        list ports 'br-eth0.200'
        option multicast '1'
        option rpfilter 'strict'
```

Expected connection to "mgmt" is not there:

[image]

```
config device
        option type 'bridge'
        option name 'br-mgmt'
        option igmp_snooping '1'
        option ipv6 '1'
        list ports 'br-eth0.1'
        option multicast '1'
        option rpfilter 'strict'
```

Devices that connect to the Wifi get a DHCP address from "guest" network, which is wrong:

[image]

No idea why netifd (?) selects the wrong network when setting up Wifi.

### OpenWrt version

r30990-04e9d154f2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.5

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

castiel652, 2025-09-09:
> config device
>         option type 'bridge'
>         option name 'br-eth0'
>         list ports 'eth0'
> 
> config bridge-vlan
>         option device 'br-eth0'
>         option vlan '1'
>         list ports 'eth0:u*'
> 
> config device
>         option type 'bridge'
>         option name 'br-mgmt'
>         option igmp_snooping '1'
>         option ipv6 '1'
>         list ports 'br-eth0.1'
>         option multicast '1'
>         option rpfilter 'strict'

Is that a bridge in a bridge?

schuettecarsten, 2025-09-09:
> Is that a bridge in a bridge?

Yes, it was the only way to make the macvlan0 thing work.

schuettecarsten, 2025-09-09:
> Once network has fully started again, interrupt the udebug command (e.g. using `fg` and ctrl+c) and send me the file `/tmp/netifd.pcap`

Interesting. I did serveral reboots and the issue persisted. Now I created a new image from current snapshot which includes `udebug-cli` and not the issue is gone.

[image]

I still run into the issue that one of my two wireguard tunnels do not start correctly, only a reboot helps in this case. I will use udebug to trace this next time and let you know.