openwrt/openwrt #20038: samba4-libs unable to install

maximst, 2025-09-13:
### Describe the bug

Collected errors:
 * verify_pkg_installable: Only have 14196kb available on filesystem /overlay, pkg samba4-libs needs 31660
 * opkg_install_cmd: Cannot install package samba4-libs.
But package size is 7.61Mb

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Team Newifi D2

### Image kind

Official downloaded image

### Steps to reproduce

opkg install samba4-libs

### Actual behaviour

Installing samba4-libs (4.18.8-r1) to root...
Collected errors:
 * verify_pkg_installable: Only have 14196kb available on filesystem /overlay, pkg samba4-libs needs 31660
 * opkg_install_cmd: Cannot install package samba4-libs.

### Expected behaviour

package installed

### Additional info

df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.0M      4.0M         0 100% /rom
tmpfs                   216.4M      1.3M    215.1M   1% /tmp
/dev/mtdblock6           24.7M     10.8M     13.9M  44% /overlay
overlayfs:/overlay       24.7M     10.8M     13.9M  44% /
tmpfs                   512.0K         0    512.0K   0% /dev

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

maximst, 2025-09-13:
[image]

maximst, 2025-09-13:
But if i build image with included samba, there will be enough disk space.

maximst, 2025-09-13:
Doesn't the package unpack to /tmp (RAM) and only then files writes to squashfs? If opkg can't determine how much space files will actually take maybe it's worth this information can be stored in the package itself?

brada4, 2025-09-13:
You can ask such questions in forums. But research what squashfs really is before making more statements.

brada4, 2025-09-18:
You can not guarantee files are compressible.

closed 2025-09-18 by hauke