openwrt/openwrt #20047: NAT Loopback generate wrong nftable rule

xux1217, 2025-09-14:
### Describe the bug

Describe the bug

I have multiple zones with the following IP ranges:
- WorkDevLan: 10.2.0.0/15
- WorkPrdLan: 10.0.0.0/15

I configured a port forward rule from "wan" to "WorkPrdLan" with reflection enabled for other zones. The redirect configuration is as follows:
``` 
config redirect
	option dest 'WorkPrdLan'
	option target 'DNAT'
	option name 'Syncthing-server-22000'
	option family 'ipv4'
	option src 'wan'
	option src_dport '22000'
	option dest_ip '10.1.0.2'
	option dest_port '22000'
	list reflection_zone 'WorkDevLan'
	list reflection_zone 'WorkPrdLan'
```

The DNAT rules work correctly, but the SNAT rules for reflection are incorrect:
```
	chain srcnat {
		type nat hook postrouting priority srcnat; policy accept;
		...(ignore)
		oifname "WorkPrdLan.2" jump srcnat_WorkPrdLan comment "!fw4: Handle WorkPrdLan IPv4/IPv6 srcnat traffic"
		oifname { "WorkDevLan.3" } jump srcnat_WorkDevLan comment "!fw4: Handle WorkDevLan IPv4/IPv6 srcnat traffic"
	}

	chain srcnat_WorkPrdLan {
		ip saddr 10.0.0.0/15 ip daddr 10.1.0.2 tcp dport 22000 snat ip to 10.0.0.1 comment "!fw4: Syncthing-server-22000 (reflection)"
		ip saddr 10.0.0.0/15 ip daddr 10.1.0.2 udp dport 22000 snat ip to 10.0.0.1 comment "!fw4: Syncthing-server-22000 (reflection)"
	}

	chain srcnat_WorkDevLan {
		ip saddr 10.2.0.0/15  ip daddr 10.1.0.2 tcp dport 22000 snat ip to 10.2.0.1 comment "!fw4: Syncthing-server-22000 (reflection)"
		ip saddr 10.2.0.0/15 ip daddr 10.1.0.2 udp dport 22000 snat ip to 10.2.0.1 comment "!fw4: Syncthing-server-22000 (reflection)"
	}
```

The issue is visible in the `srcnat` chain, traffic match the oifname `WorkPrdLan.2` and jump to `srcnat_WorkPrdLan`, but actually, the traffic should jump to the `srcnat_WorkDevLan` chain, because the saddr is in `10.2.0.0/15`, not in `10.0.0.0/15`.

Because the srcnat is not effect, so I can't access the service in WorkPrdLan from WorkDevLan.

**Root Cause Analysis**: The problem appears to be that the rule matching uses oifname (output interface) instead of iifname (input interface) to determine which source NAT chain to use. For reflection traffic, we should match based on the incoming interface rather than the outgoing interface.

### OpenWrt version

r28739-d9340319c6

### OpenWrt release

24.10.2

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

create these zones:
- wan
- lan0
- lan1

create port forward rule:
forward from wan to lan0, and add lan1 to reflection_zone.

check the nat rule about srcnat of lan1.

### Actual behaviour

srcnat not effect, because the saddr is not match in wrong srcnat_WorkPrdLan chain.

### Expected behaviour

the srcnat should effect.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-09-14:
Can you provide parts of rulesets (nft list ruleset)  down to "base hooks" ie ones with chains doing jump/goto to these chains?

brada4, 2025-09-14:
You changed in ucode files or added ruleset include?
You need flat routing and accept-forward rule, not NAT.

xux1217, 2025-09-15:
> You need flat routing and accept-forward rule, not NAT.

I have a public service, that use the "port forward" to public, and I need access the service use the domain and public port, so I need the NAT Loopback function.

----------------------------------------------------------------------
> You changed in ucode files or added ruleset include?
No, and simply analyze this generate nftable rule, you think it is correct? Or the wrong nftable rule just generated at my local machine?

xux1217, 2025-09-27:
> I kind of have a workable solution with:
> 
> --- /rom/usr/share/ucode/fw4.uc 2025-09-19 17:19:38.000000000 -0400
> +++ fw4.uc      2025-09-26 14:36:27.634981545 -0400
> @@ -2971,7 +2971,7 @@
>                                                                                 add_rule(i ? 6 : 4, proto, saddrs, daddrs, rip[i], sport, dport, rport, null, refredir);
> 
>                                                                 refredir.src = null;
> -                                                               refredir.dest = rzone;
> +                                                               refredir.dest = redir.dest;
>                                                                 refredir.target = "snat";
> 
>                                                                 for (let daddrs in subnets_group_by_masking(rip[i]))
> 
> I don't know if it breaks any other scenarios, but the reflection source nat rules should always use the redirect target zone. Only the dnat rules should end up in the unique reflection_zone chains.
> 
> Thoughts?
> 
> Result with 2 zones:
> 
> ```
> chain srcnat_lan {
>     ip saddr 192.168.2.0/24 ip daddr 192.168.1.5 tcp dport 80 snat ip to 192.168.2.1 comment "!fw4: NAT Refl Test (reflection)"
>     ip saddr 192.168.1.0/24 ip daddr 192.168.1.5 tcp dport 80 snat ip to 192.168.1.1 comment "!fw4: NAT Refl Test (reflection)"
> }
> ```

I think your method is right, and modify simple.

Just change the match chain content.

brada4, 2025-09-27:
You need to filter by source interface too.

dave14305, 2025-09-27:
> You need to filter by source interface too.

Why, if we already filter by saddr?

dave14305, 2025-10-07:
Still thinking about this one, and I don't think there's a real problem in the current implementation, except fw4 emits a useless `srcnat_WorkDevLan` chain that isn't needed in the OP's layout.

SNATting for reflection should really only be necessary when the source IP and DNATted destination IP are in the same subnet. When the source IP is in a different subnet, the destination IP reply to the IP outside its subnet and send it to the default gateway (router) where the original DNAT will send it back to the requester with the public IP sending the reply packet.

I feel like something else was causing the breakage for @xux1217, but we probably need more info.

This was my test scenario browsing my printer webUI through a port forward redirect rule (my real public IP replaced by 15.15.15.15):
Network:
```
config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'guest'
        option proto 'static'
        option device 'br-guest'
        option ipaddr '192.168.2.1'
        option netmask '255.255.255.0'
```
Firewall:
```
config zone
        option name 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'
        list network 'lan'

config zone
        option name 'wan'
        option input 'DROP'
        option output 'ACCEPT'
        option forward 'DROP'
        option masq '1'
        option mtu_fix '1'
        list network 'wan'
        list network 'wan6'

config zone
        option name 'guest'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'DROP'
        list network 'guest'

config redirect
        option dest 'lan'
        option target 'DNAT'
        option name 'Printer Forward'
        list proto 'tcp'
        option src 'wan'
        option src_dport '48187'
        option dest_ip '192.168.1.5'
        option dest_port '80'
        list reflection_zone 'guest'
        list reflection_zone 'lan'
```
nft rules:
```
chain dstnat {
		type nat hook prerouting priority dstnat; policy accept;
		iifname "br-lan" jump dstnat_lan comment "!fw4: Handle lan IPv4/IPv6 dstnat traffic"
		iifname "eth1" jump dstnat_wan comment "!fw4: Handle wan IPv4/IPv6 dstnat traffic"
		iifname "br-guest" jump dstnat_guest comment "!fw4: Handle guest IPv4/IPv6 dstnat traffic"
}

chain srcnat {
		type nat hook postrouting priority srcnat; policy accept;
		oifname "br-lan" jump srcnat_lan comment "!fw4: Handle lan IPv4/IPv6 srcnat traffic"
		oifname "eth1" jump srcnat_wan comment "!fw4: Handle wan IPv4/IPv6 srcnat traffic"
		oifname "br-guest" jump srcnat_guest comment "!fw4: Handle guest IPv4/IPv6 srcnat traffic"
}

chain dstnat_lan {
		ip saddr 192.168.1.0/24 ip daddr 15.15.15.15 tcp dport 48187 dnat 192.168.1.5:80 comment "!fw4: Printer Forward (reflection)"
}

chain srcnat_lan {
		ip saddr 192.168.1.0/24 ip daddr 192.168.1.5 tcp dport 80 snat 192.168.1.1 comment "!fw4: Printer Forward (reflection)"
}

chain dstnat_wan {
		meta nfproto ipv4 tcp dport 48187 counter dnat 192.168.1.5:80 comment "!fw4: Printer Forward"
}

chain dstnat_guest {
		ip saddr 192.168.2.0/24 ip daddr 15.15.15.15 tcp dport 48187 dnat 192.168.1.5:80 comment "!fw4: Printer Forward (reflection)"
}

chain srcnat_guest {
		ip saddr 192.168.2.0/24 ip daddr 192.168.1.5 tcp dport 80 snat 192.168.2.1 comment "!fw4: Printer Forward (reflection)"
}
```
tcpdump results 
```
# tcpdump from guest to 15.15.15.15:48187
13:50:55.251350 br-guest In  IP 192.168.2.246.54588 > 15.15.15.15.48187: Flags [P.], seq 0:453, ack 1, win 4096, options [nop,nop,TS val 2710373097 ecr 1648767554], length 453
13:50:55.251371 br-lan Out IP 192.168.2.246.54588 > 192.168.1.5.80: Flags [P.], seq 536447861:536448314, ack 1739497273, win 4096, options [nop,nop,TS val 2710373097 ecr 1648767554], length 453: HTTP: GET /DevMgmt/ProductStatusDyn.xml HTTP/1.1
13:50:55.259481 br-lan In  IP 192.168.1.5.80 > 192.168.2.246.54588: Flags [.], ack 453, win 6900, options [nop,nop,TS val 1648792238 ecr 2710373097], length 0
13:50:55.259500 br-guest Out IP 15.15.15.15.48187 > 192.168.2.246.54588: Flags [.], ack 453, win 6900, options [nop,nop,TS val 1648792238 ecr 2710373097], length 0

# tcpdump from lan to 15.15.15.15:48187
13:56:50.994268 br-lan In  IP 192.168.1.128.62055 > 15.15.15.15.48187: Flags [.], seq 0:1, ack 1, win 252, length 1
13:56:50.994292 br-lan Out IP 192.168.1.1.62055 > 192.168.1.5.80: Flags [.], seq 1849712358:1849712359, ack 3050020795, win 252, length 1: HTTP
13:56:50.994601 br-lan In  IP 192.168.1.5.80 > 192.168.1.1.62055: Flags [.], ack 1, win 8340, length 0
13:56:50.994618 br-lan Out IP 15.15.15.15.48187 > 192.168.1.128.62055: Flags [.], ack 1, win 8340, length 0
```
I don't think it's necessary to emit the srcnat_guest chain at all since it's never reached. Otherwise, I think firewall4 is working as expected.