openwrt/openwrt #20060: Support for DGS-1210-52MP on latest firmware

HDCP, 2025-09-16:
Existing Tech Data: https://openwrt.org/toh/hwdata/d-link/d-link_dgs-1210-52
Wiki Page: https://openwrt.org/toh/d-link/dgs-1210-16_g1

Device Type: DGS-1210-52MP Gigabit Ethernet Switch  
Boot Version: 1.00.010  
Firmware Version: 6.32.B019
Hardware Version: F1

Current firmware provided for DGS-1210-52 will not install on the latest available firmware (6.32.B019) for the DGS-1210-52MP through the D-Link "Firmware Backup & Upgrade" Admin UI screen. The 6.32.B008 firmware is no longer available on the device website for download.

Are we able to add support for this please?

HDCP, 2025-09-16:
> You can find a download link in the very wiki page you linked to.

Not specifically 6.32.B008 for the DGS-1210-52MP as far I can see. I'm happy to test an install on the DGS-1210-52MP if someone is able to provide a compatible firmware.

HDCP, 2025-09-16:
> Yes, 6.20 is the working version, if you have 6.30 or later you have to downgrade. easy as pie.

No, for the DGS-1210-52MP, even on 6.20.B009, this error is received when trying to load the OpenWRT 24.10 firmware .bin file:

_FW upgrade/backup fail, check FW file..._

plappermaul, 2025-09-16:
The word inside  https://ftp.dlink.de/dgs/dgs-1210/archive/driver_software/DGS-1210-52MP_fw_revF3_6-30-B016_all_multi_20211216.zip gives details about normal and encrypted firmware switch and why wee need 6.20.

**52MP** has no official OpenWrt builds and not a single mention in the Wiki. So you seem to be currently the only one to has this device.  Sad to say that it depends on you find out how to get OpenWrt onto it.

HDCP, 2025-09-21:
I created a snapshot build which works with UI installation for the DGS-1210-52MP: [openwrt-realtek-rtl839x-d-link_dgs-1210-52mp-f.zip](https://github.com/user-attachments/files/22453611/openwrt-realtek-rtl839x-d-link_dgs-1210-52mp-f.zip)

Pull Request here for full support into the repository: https://github.com/openwrt/openwrt/pull/20122