openwrt/openwrt #20061: ZTE MF286D no longer acquires an IP address from cellular network

ezplanet, 2025-09-16:
### Describe the bug

Up to OpenWrt 24.10-SNAPSHOT r28814-03fe596699 / LuCI openwrt-24.10 branch 25.206.68186~6da5e65 (last build with no issues) the ZTE MF286D connects to the cellular network and acquires and IP (IPv4) address.
My next build was OpenWrt 24.10-SNAPSHOT r28840-65269fd3fc / LuCI openwrt-24.10 branch 25.245.48024~b8bc15b. From this build onwards the router appears to connect to the cellular network, but fails to acquire an IP address and the interface w

### OpenWrt version

r28840-65269fd3fc

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZTE MF286D

### Image kind

Self-built image

### Steps to reproduce

Configure the router wan as follows:

config interface 'wan'
    option proto 'qmi'
    option device '/dev/cdc-wdm0'
    option apn 'web.omnitel.it'
    option auth 'none'
    option pdptype 'ipv4'
    option dns_metric '10'
    option peerdns '0'
    list dns '1.0.0.1'
    list dns '1.1.1.1'

and start the wan network

### Actual behaviour

The log reports:

2025-09-16T11:46:54+02:00 africo netifd: wan (5288): PIN verification is disabled
2025-09-16T11:46:55+02:00 africo netifd: wan (5288): Waiting for network registration
2025-09-16T11:47:00+02:00 africo netifd: wan (5288): Starting network wan
2025-09-16T11:47:01+02:00 africo netifd: wan (5288): Setting up wwan0
2025-09-16T11:47:01+02:00 africo netifd: Interface 'wan' is now up
2025-09-16T11:47:01+02:00 africo netifd: Network device 'wwan0' link is up
2025-09-16T11:47:01+02:00 africo netifd: Network alias 'wwan0' link is up
2025-09-16T11:47:01+02:00 africo netifd: Interface 'wan_4' is enabled
2025-09-16T11:47:01+02:00 africo netifd: Interface 'wan_4' has link connectivity
2025-09-16T11:47:01+02:00 africo netifd: Interface 'wan_4' is setting up now
2025-09-16T11:47:01+02:00 africo netifd: wan_4 (5370): udhcpc: started, v1.36.1
2025-09-16T11:47:01+02:00 africo firewall: Reloading firewall due to ifup of wan (wwan0)

but ifconfig shows:

wwan0     Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet6 addr: fe80::b01a:74ff:xxxx:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:87 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:22786 (22.2 KiB)

wan_4 is not present and thus the cellular wan interface has no IP address
[... 275 lines cut ...]
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_usbids=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vim-full=y
CONFIG_PACKAGE_vim-runtime=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_whereis=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
# CONFIG_USE_FS_ACL_ATTR is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-09-18:
Looks like the same problem as https://github.com/openwrt/openwrt/issues/20051

ezplanet, 2025-09-19:
I confirm that this is the same as https://github.com/openwrt/openwrt/issues/19945 and it can be closed. 
#19945 now has a solution that I confirmed as working and it is awaiting a PR and approval