openwrt/openwrt #20083: realtek: i2c dts warnings

plappermaul, 2025-09-18:
Compiler complains about the following syntax in rtl930x.dtsi and rtl931x.dtsi.

```
		i2c_mst1: i2c@100c {
			compatible = "realtek,rtl9310-i2c";
			reg = <0x100c 0x18>;
			#address-cells = <1>;
			#size-cells = <0>;
			realtek,scl = <0>;

			status = "disabled";
		};

		i2c_mst2: i2c@1024 {
			compatible = "realtek,rtl9310-i2c";
			reg = <0x1024 0x18>;
			#address-cells = <1>;
			#size-cells = <0>;
			realtek,scl = <1>;

			status = "disabled";
		};
```

If reg attribute contains 2 values we either have 2 address-cells or 1 address-cell and one size-cell.

plappermaul, 2025-09-18:
@jonasjelonek I am lost in upstream discussion. What is the right syntax here?

Sample warning.

```
../dts/rtl930x.dtsi:166.4-23: Warning (reg_format): /switchcore@1b000000/i2c@36c:reg: property has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
../dts/rtl930x.dtsi:175.4-23: Warning (reg_format): /switchcore@1b000000/i2c@388:reg: property has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
```

jonasjelonek, 2025-09-18:
The `#address-cells = <1>` and `#size-cells = <0>` in these I2C nodes should refer to their children's reg property, so this is probably fine? I think the issue is then in the (missing) properties for the switchcore node. 2 and 1 seem to be the default values.

See upstream DTSI for RTL930x [here](https://elixir.bootlin.com/linux/v6.16.4/source/arch/mips/boot/dts/realtek/rtl930x.dtsi), there those are specified correctly and the warning doesn't appear.

So does this warning disappear if you add
```
#address-cells = <1>;
#size-cells = <1>;
```
to the `switchcore@1b...` node in `rtl930x.dtsi`/`rtl931x.dtsi`? Not sure if this causes issues with the other children of switchcore.

plappermaul, 2025-09-19:
Fixed with https://github.com/openwrt/openwrt/pull/20091