openwrt/openwrt #20151: imagebuilder: meditek/filogic:asus_zenwifi-bt8 profile fails to build initramfs-kernel

efahl, 2025-09-24:
Using latest snapshot imagebuilder:
```
Current Target: "mediatek/filogic"
Current Architecture: "aarch64"
Current Revision: "r31146-4eae48d9dc"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd uci uclient-fetch urandom-seed urngd kmod-leds-gpio kmod-gpio-button-hotplug fitblk kmod-crypto-hw-safexcel wpad-basic-mbedtls uboot-envtools procd-ujail dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe apk-mbedtls
```

Steps to reproduce:
```
$ wget https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64.tar.zst
$ tar xvf openwrt-imagebuilder-mediatek-filogic.Linux-x86_64.tar.zst
$ cd openwrt-imagebuilder-mediatek-filogic.Linux-x86_64/

$ make image PROFILE=asus_zenwifi-bt8
...
7445504 bytes (7.4 MB, 7.1 MiB) copied, 0.0280651 s, 265 MB/s
/home/efahlgren/openwrt/image_builder/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64/staging_dir/host/bin/tar: Option --mtime: Treating date '@1758662673' as 2025-09-23 14:24:33
sysupgrade-asus_zenwifi-bt8/
sysupgrade-asus_zenwifi-bt8/CONTROL
sysupgrade-asus_zenwifi-bt8/kernel
sysupgrade-asus_zenwifi-bt8/root
cp: cannot stat '/home/efahlgren/openwrt/image_builder/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64/bin/targets/mediatek/filogic/openwrt-mediatek-filogic-asus_zenwifi-bt8-initramfs-kernel.bin': No such file or directory
make[3]: *** [Makefile:48: /home/efahlgren/openwrt/image_builder/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-asus_zenwifi-bt8-factory.bin] Error 1
make[2]: *** [Makefile:264: build_image] Error 2
make[1]: *** [Makefile:159: _call_image] Error 2
make: *** [Makefile:333: image] Error 2
```

Looking at https://github.com/openwrt/openwrt/blob/9aca8a97d7911230211d98a24e4dc4b0a0173ec2/target/linux/mediatek/image/filogic.mk#L408 I see nothing here has changed in six months.

@blogic @dangowrt Do you see anything amiss there?  I tested another profile, `dlink_aquila-pro-ai-m30-a1`, that has initramfs-kernel, but uses `append-image-stage` instead of `append-image` and it builds fine.

efahl, 2025-09-24:
Additional info:

There's no initramfs-kernel included in the imagebuilder for the BT8 profile.
```
$ find -iname '*initramfs-kernel*'
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-dlink_aquila-pro-ai-m60-a1-initramfs-kernel.bin
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-dlink_aquila-pro-ai-m30-a1-initramfs-kernel.bin
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-xiaomi_redmi-router-ax6000-stock-initramfs-kernel.bin
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-kernel.bin
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-asus_rt-ax52-initramfs-kernel.bin
./staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-xiaomi_mi-router-wr30u-stock-initramfs-kernel.bin
```

efahl, 2025-10-03:
Seeing this also for ath79/generic:elecom,wrc-300ghbk2-i  (https://forum.openwrt.org/t/luci-attended-sysupgrade-support-thread/230552/240).  User was trying 24.10.2, I tested 24.10.0, 24.10-snapshot and main snapshot, which all fail the same way.

```
cp: cannot stat '.../openwrt-24.10.0-ath79-generic-elecom_wrc-300ghbk2-i-initramfs-kernel.bin': No such file or directory
```

@robimarko Could this be related to our recent changes to the imagebuilder makefiles somehow?  Some profiles are built but don't get initramfs-kernel bins in the imagebuilder; I can't find the magic handshake that causes other profiles to properly generate the initramfs-kernel, but not these few.

robimarko, 2025-10-06:
I guess that they generate those images based on staging dir initramfs images, which are included by accident, cause we cannot really generate an initramfs image as there are no kernel sources.

So, yes a check for IB should be added

efahl, 2025-10-19:
@Djfe 

> did you miss the Netgear WAX218?

Weell, not so much missed as "ignored"...  There were a handful of candidates that I found to fail, but was uncomfortable changing because the artifact/file names didn't fit the pattern, so I skipped them.  Now that I've dug into it deeper, I have better understanding and need to fix another half-dozen profiles, including that one.

> how do we prevent new devices from being added that build with an initramfs but forget to add the guard?

Yeah, that's easy. :)

I don't have a good answer, as what I've been doing is completely unsustainable.  I have a script that downloads and sets up the imagebuilder for a version/target/subtarget.  Another script collects all the profiles from that imagebuilder, then does `make image PROFILE=$profile` for each, checking result status, build artifacts, ensures that a `*sysupgrade*` binary was created, looks for extraneous `initramfs` images and so on.  I then manually scan the logs from each run, which takes a looong time; I've run and examined output for 960 profiles (yeah, I did count them) about 4 times now and my brain hurts.

The docker image creation CI does do a "smoke test" on each imagebuilder as it is created, but it only builds the single default profile (i.e., it does `make image` with no profile parameter supplied).  In order to find profiles that don't build and send up a flag, it should probably do what I'm doing above, but that can take a huge amount of wall and CPU time (as a worst-case example, 378 profiles in snapshot ath79/generic right now, ~55 minutes wall time on my beefy 7950x workstation).  I have zero knowledge as to whether this is reasonable in a CI job or not...