openwrt/openwrt #20156: wifi-scripts (ucode): R7800 fails to initialize radios

hnyman, 2025-09-24:
### Describe the bug

@nbd168 
With the newly default ucode wifi scripts, the R7800 ath10k-ct the radio initialisation fails with 8 errors like below. 

```
Wed Sep 24 21:04:23 2025 daemon.notice netifd: radio0 (2323): wifi-scripts: Starting
Wed Sep 24 21:04:23 2025 daemon.notice hostapd: Set new config for phy phy1:
Wed Sep 24 21:04:23 2025 daemon.notice wpa_supplicant[1696]: Set new config for phy phy1
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): command failed: Not supported (-95)
Wed Sep 24 21:04:24 2025 daemon.notice wpa_supplicant[1696]: Set new config for phy phy1
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): Reference error: left-hand side expression is null
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): In device_capabilities(), file /usr/share/ucode/wifi/hostapd.uc, line 448, byte 34:
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):   called from function generate (/usr/share/ucode/wifi/hostapd.uc:465:28)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:542:22)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):   called from function setup (./mac80211.sh:264:20)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):   called from anonymous function (./mac80211.sh:302:14)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):  `    phy_capabilities.ht_capa = band.ht_capa ?? 0;`
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):   Near here --------------------------^
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): wifi-scripts: Configuring 'phy1' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): wifi-scripts: Preparing interface: hn5wpa3 with MAC: dc:ef:09:ef:f3:e6
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio0 (2323): wifi-scripts: Preparing interface: hn5wpa2r with MAC: 78:00:22:EF:F3:E6
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): wifi-scripts: Starting
Wed Sep 24 21:04:24 2025 daemon.notice hostapd: Set new config for phy phy0:
Wed Sep 24 21:04:24 2025 daemon.notice wpa_supplicant[1696]: Set new config for phy phy0
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): command failed: Not supported (-95)
Wed Sep 24 21:04:24 2025 daemon.notice wpa_supplicant[1696]: Set new config for phy phy0
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): Reference error: left-hand side expression is null
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): In device_capabilities(), file /usr/share/ucode/wifi/hostapd.uc, line 448, byte 34:
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):   called from function generate (/usr/share/ucode/wifi/hostapd.uc:465:28)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):   called from function setup (/usr/share/ucode/wifi/hostapd.uc:542:22)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):   called from function setup (./mac80211.sh:264:20)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):   called from anonymous function (./mac80211.sh:302:14)
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):  `    phy_capabilities.ht_capa = band.ht_capa ?? 0;`
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):   Near here --------------------------^
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403):
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): wifi-scripts: Configuring 'phy0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Wed Sep 24 21:04:24 2025 daemon.notice netifd: radio1 (2403): wifi-scripts: Preparing interface: hn2wlan with MAC: dc:ef:09:ef:f3:e7
```

### OpenWrt version

r31159-9aca8a97d7

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Self-built image

### Additional info

wifi config:
```
config wifi-device 'radio0'
[... 490 lines cut ...]
CONFIG_PACKAGE_rrdtool1=y
CONFIG_PACKAGE_sqm-scripts=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_tcpdump-mini=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_tree=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-log=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_unet-cli=y
CONFIG_PACKAGE_unetd=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_UNETD_VXLAN_SUPPORT=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

e9hack, 2025-09-24:
It looks like a option band is mandatory in the wifi-device section.

hnyman, 2025-09-25:
Band is not mandatory in wiki, and the config has worked with the old scripts. Maybe the new ucode scripts are more picky.

Curiously, I seem to have 5 GHz trouble also with MT6000, where I do have the band set in the config. Likely there is some other item that does not get parsed correctly.

nbd168, 2025-09-25:
@hnyman fixed now, thanks for reporting