openwrt/openwrt #20170: IPv6 prefix filter does not have effect on static addresses

kbios, 2025-09-25:
### Describe the bug

If an interface has a configured static IPv6 address, its prefix is included in RAs even when a prefix filter is configured to exclude it

### OpenWrt version

r30990-04e9d154f2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 (stock layout)

### Image kind

Official downloaded image

### Steps to reproduce

* Add a static ipv6 (/64) to an interface
* Make sure Router Advertisements are enabled
* Configure the IPv6 prefix filter to include for example only "local (ULA)" or "wan6"

### Actual behaviour

The prefix corresponding to the static address is still advertised via RAs

### Expected behaviour

If the prefix is not in the filter it should not be advertised

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-09-25:
The `ip6class` prefix filter is only going to filter upstream prefixes from being assigned to the local interfaces. If you manually assign an address to an interface, it will certainly be advertised in the same interface’s RAs. It’s an option used by netifd, not odhcpd (which does the RAs).

dave14305, 2025-09-25:
odhcpd has a `prefix_filter` option, but not exposed in LuCI.
https://github.com/openwrt/odhcpd/blob/b1547922257e484cb18e1719bdfc18f4789eb560/README#L156