openwrt/openwrt #20182: rax3000m compilation issue

zcy85611, 2025-09-26:
### Describe the bug

after recent commit
[028050d](https://github.com/openwrt/openwrt/commit/028050da6978279f337f303eec55bcb45c56423a)
[93b2084](https://github.com/openwrt/openwrt/commit/93b208434dd3153ed0bb204a56beec834e4dde51)
[b8c427f](https://github.com/openwrt/openwrt/commit/b8c427f4867d5f843743d6835821307f29d256ab)
uboot compilation shows the following error

```
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3 >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-ram-ddr3] Error 1 (ignored)
if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3 ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3 && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3/; fi
make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7981-ram-ddr3/compile#0.15#0.15#0.26
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-emmc-ddr3-1866mhz] Error 1 (ignored)
if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz/; fi
make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7981-emmc-ddr3-1866mhz/compile#0.15#0.15#0.25
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz] Error 1 (ignored)
if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz/; fi
make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7981-spim-nand-ddr3-1866mhz/compile#0.15#0.16#0.25
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr4
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr4 >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-ram-ddr4] Error 1 (ignored)
if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr4 ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr4 && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr4/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr4/; fi
make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7981-ram-ddr4/compile#0.16#0.14#0.25
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr4
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr4 >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-emmc-ddr4] Error 1 (ignored)
if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr4 ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr4 && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr4/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr4/; fi
make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
time: package/boot/arm-trusted-firmware-mediatek/mt7981-emmc-ddr4/compile#0.14#0.16#0.25
make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr4
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr4 >/dev/null 2>/dev/null
make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-spim-nand-ddr4] Error 1 (ignored)
[... cut ...]
rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7986-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7986-ram-ddr4 >/dev/null 2>/dev/null
```

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

in compliation

### Device

cmcc_rax3000m

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kousyougi, 2025-09-27:
Same here

> ### Describe the bug
> 
> after recent commit [028050d](https://github.com/openwrt/openwrt/commit/028050da6978279f337f303eec55bcb45c56423a) [93b2084](https://github.com/openwrt/openwrt/commit/93b208434dd3153ed0bb204a56beec834e4dde51) [b8c427f](https://github.com/openwrt/openwrt/commit/b8c427f4867d5f843743d6835821307f29d256ab) uboot compilation shows the following error, is it possible to fix it ?
> 
> ```
> make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3
> rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3 >/dev/null 2>/dev/null
> make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-ram-ddr3] Error 1 (ignored)
> if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3 ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3 && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr3/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr3/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr3/; fi
> make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> time: package/boot/arm-trusted-firmware-mediatek/mt7981-ram-ddr3/compile#0.15#0.15#0.26
> make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz
> rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz >/dev/null 2>/dev/null
> make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-emmc-ddr3-1866mhz] Error 1 (ignored)
> if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-emmc-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-emmc-ddr3-1866mhz/; fi
> make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> time: package/boot/arm-trusted-firmware-mediatek/mt7981-emmc-ddr3-1866mhz/compile#0.15#0.15#0.25
> make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz
> rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz >/dev/null 2>/dev/null
> make[3]: [Makefile:666: install-bin-trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz] Error 1 (ignored)
> if [ -d /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz ]; then install -d -m0755 /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz && cp -fpR /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-spim-nand-ddr3-1866mhz/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz/. /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-spim-nand-ddr3-1866mhz/; fi
> make[3]: Leaving directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> time: package/boot/arm-trusted-firmware-mediatek/mt7981-spim-nand-ddr3-1866mhz/compile#0.15#0.16#0.25
> make[3]: Entering directory '/home/runner/work/Auto/Auto/openwrt/package/boot/arm-trusted-firmware-mediatek'
> rm -rf /home/runner/work/Auto/Auto/openwrt/bin/targets/mediatek/filogic/trusted-firmware-a-mt7981-ram-ddr4
> rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7981-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7981-ram-ddr4 >/dev/null 2>/dev/null
[... cut ...]
> rmdir /home/runner/work/Auto/Auto/openwrt/build_dir/target-aarch64_cortex-a53_musl/arm-trusted-firmware-mediatek-mt7986-ram-ddr4/arm-trusted-firmware-mediatek-2025.07.11~78a0dfd9/.pkgdir/trusted-firmware-a-mt7986-ram-ddr4 >/dev/null 2>/dev/null
> ```
> 
> ### OpenWrt version
> 
> snapshot
> 
> ### OpenWrt release
> 
> snapshot
> 
> ### OpenWrt target/subtarget
[... 14 lines cut ...]
> 
> ### Actual behaviour
> 
> _No response_
> 
> ### Expected behaviour
> 
> _No response_
> 
> ### Additional info
> 
> _No response_
> 
> ### Diffconfig
> 
> ```
> 
> ```
> 
> ### Terms
> 
> * [x]  I am reporting an issue for OpenWrt, not an unsupported fork.

Same here
Help

closed 2025-09-29 by zcy85611

kousyougi, 2025-09-29:
I ran into the same problem. Has this issue been resolved?