openwrt/openwrt #20196: Linksys E8450: 5GHz Wi-Fi fails to initialize after reboot

CharlesMengCA, 2025-09-26:
### Describe the bug

The 5GHz Wi-Fi on my Linksys E8450 router fails to start after booting or rebooting.

### OpenWrt version

r31180-86618df095

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

 Linksys E8450 (UBI)

### Image kind

Self-built image

### Steps to reproduce

1. Flash the firmware.
2. Boot the router.
3. the 5GHz Wi-Fi interface does not start automatically.
4. Check system logs for the following error:
   ```
   daemon.notice netifd: radio1 (2153): wifi-scripts: Bug: PHY is undefined for device
   ```
5. Manually disable and re-enable the 5GHz Wi-Fi interface — it starts successfully.
6. Reboot the router again.
7. The 5GHz Wi-Fi interface is disabled once more and requires manual reactivation.

### Actual behaviour

The 5GHz interface remains disabled after reboot and logs a PHY error.

### Expected behaviour

The 5GHz Wi-Fi interface should initialize and start automatically after boot or reboot.

### Additional info

[SystemLog.txt](https://github.com/user-attachments/files/22568819/SystemLog.txt)

[diffconfig.txt](https://github.com/user-attachments/files/22568837/diffconfig.txt)

### Diffconfig

```text
My diffconfig is kind of long. Refer the attachment above.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

CharlesMengCA, 2025-09-27:
Hi @nbd168,

If you have time, could you help?

Thanks a lot. :)

hauke, 2025-09-27:
Have you set a `phy` attribute in your wireless configuration?

For me it looks like this:
```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option phy 'wl0'
	option band '2g'
	option channel '6'
	option htmode 'HT20'
	option disabled '0'
	option country 'DE'

```

CharlesMengCA, 2025-09-27:
```
root@LINKSYS_E8450:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option phy 'wl0'
        option cell_density '0'
        option htmode 'HT40'
        option band '2g'
        option channel '1'
        option country 'CA'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'QFox'
        option encryption 'none'
        option disabled '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option phy 'wl1'
        option cell_density '0'
        option htmode 'HE80'
        option band '5g'
        option channel '36'
        option country 'CA'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'QFox-5G2'
        option encryption 'sae-mixed'
        option key 'blahblah'
        option ocv '0'

```

CharlesMengCA, 2025-09-30:
**ucode based scripts** still have some bugs.

After reverting https://github.com/openwrt/openwrt/commit/04e9929c47e9fcf09bb32c44eb0a1447fd4962cf

or unsetting WIFI_SCRIPTS_UCODE in config file.

`# CONFIG_WIFI_SCRIPTS_UCODE is not set`

The problem will go away.

CharlesMengCA, 2025-10-07:
Already tried. still not working.

Error during boot/reboot

```
Mon Oct  6 14:55:12 2025 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Oct  6 14:55:12 2025 user.notice ucitrack: Setting up /etc/config/qos reload dependency on /etc/config/firewall
Mon Oct  6 14:55:12 2025 user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
Mon Oct  6 14:55:12 2025 daemon.notice netifd: radio1 (1898): wifi-scripts: Bug: PHY is undefined for device
Mon Oct  6 14:55:12 2025 user.notice ucitrack: Setting up /etc/config/odhcpd reload dependency on /etc/config/dhcp
Mon Oct  6 14:55:12 2025 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Mon Oct  6 14:55:12 2025 daemon.notice netifd: radio0 (1948): wifi-scripts: Bug: PHY is undefined for device
Mon Oct  6 14:55:12 2025 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
Mon Oct  6 14:55:12 2025 daemon.notice netifd: radio1 (2006): wifi-scripts: Bug: PHY is undefined for device
Mon Oct  6 14:55:13 2025 user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Mon Oct  6 14:55:13 2025 daemon.notice netifd: radio0 (2033): wifi-scripts: Bug: PHY is undefined for device

```
Log during Disabled and then Enabled WIFI interface```

Mon Oct  6 15:15:25 2025 daemon.notice wpa_supplicant[1562]: Set MLD config: [ ]
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Set MLD config: [ ]
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Reload all interfaces
Mon Oct  6 15:15:25 2025 daemon.notice netifd: radio1 (3349): wifi-scripts: Starting
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Set new config for phy wl1:
Mon Oct  6 15:15:25 2025 daemon.notice wpa_supplicant[1562]: Set new config for phy wl1
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Set MLD config: [ ]
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Reload all interfaces
Mon Oct  6 15:15:25 2025 daemon.notice wpa_supplicant[1562]: Set MLD config: [ ]
Mon Oct  6 15:15:25 2025 daemon.notice wpa_supplicant[1562]: Set new config for phy wl1
Mon Oct  6 15:15:25 2025 daemon.notice hostapd: Set new config for phy wl1:
Mon Oct  6 15:15:26 2025 daemon.notice netifd: radio1 (3349): wifi-scripts: Configuring 'wl1' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Mon Oct  6 15:15:26 2025 daemon.notice netifd: radio0 (3377): wifi-scripts: Starting
Mon Oct  6 15:15:26 2025 daemon.notice hostapd: Set new config for phy wl0:
Mon Oct  6 15:15:26 2025 daemon.notice wpa_supplicant[1562]: Set new config for phy wl0
Mon Oct  6 15:15:26 2025 daemon.notice wpa_supplicant[1562]: Set new config for phy wl0
Mon Oct  6 15:15:26 2025 daemon.notice hostapd: Set new config for phy wl0:
Mon Oct  6 15:15:26 2025 daemon.notice netifd: radio0 (3377): wifi-scripts: Configuring 'wl0' txantenna: 4294967295, rxantenna: 4294967295 distance: 0
Mon Oct  6 15:15:26 2025 daemon.notice wpa_supplicant[1562]: Start pending MLD interfaces
Mon Oct  6 15:15:33 2025 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Mon Oct  6 15:15:33 2025 daemon.err rmmod: unloading the module failed
Mon Oct  6 15:15:33 2025 daemon.info ksmbd: triggering kill_server
Mon Oct  6 15:15:33 2025 kern.info kernel: [  136.685925] ksmbd: kill command received
Mon Oct  6 15:15:37 2025 daemon.notice ksmbd: Starting Ksmbd userspace service.
Mon Oct  6 15:15:37 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct  6 15:15:37 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mon Oct  6 15:15:37 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct  6 15:15:37 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mon Oct  6 15:15:38 2025 daemon.notice wpa_supplicant[1562]: Set MLD config: [ ]
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Set MLD config: [ ]
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Reload all interfaces
Mon Oct  6 15:15:38 2025 daemon.notice netifd: radio1 (4065): wifi-scripts: Starting
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Set MLD config: [ ]
Mon Oct  6 15:15:38 2025 daemon.notice wpa_supplicant[1562]: Set MLD config: [ ]
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Reload all interfaces
Mon Oct  6 15:15:38 2025 daemon.notice wpa_supplicant[1562]: Set new config for phy wl1
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Set new config for phy wl1: /var/run/hostapd-wl1.conf
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Restart interface for phy wl1
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Configuration file: <inline> (phy wl1) --> new PHY
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Configuration file: Reading configuration file '<inline>'
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: Configuration file: Reading configuration file '<inline>'
Mon Oct  6 15:15:38 2025 kern.info kernel: [  141.779221] br-lan: port 6(wl1-ap0) entered blocking state
Mon Oct  6 15:15:38 2025 kern.info kernel: [  141.784752] br-lan: port 6(wl1-ap0) entered disabled state
Mon Oct  6 15:15:38 2025 kern.info kernel: [  141.790353] mt7915e 0000:01:00.0 wl1-ap0: entered allmulticast mode
Mon Oct  6 15:15:38 2025 kern.info kernel: [  141.796855] mt7915e 0000:01:00.0 wl1-ap0: entered promiscuous mode
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: wl1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Oct  6 15:15:38 2025 daemon.notice hostapd: wl1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
[... cut ...]
Mon Oct  6 15:15:39 2025 daemon.notice hostapd: wl1-ap0: AP-ENABLED
Mon Oct  6 15:15:40 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct  6 15:15:40 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mon Oct  6 15:15:40 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct  6 15:15:40 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Mon Oct  6 15:15:41 2025 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Mon Oct  6 15:15:41 2025 daemon.notice ksmbd: Starting Ksmbd userspace service.

```

rany2, 2025-12-16:
@CharlesMengCA Please try with latest SNAPSHOT with https://github.com/openwrt/openwrt/commit/e4d5e76af059d9c7fe295bbd144a39a77fbb5491 included.