openwrt/openwrt #20200: 24.10.3 broke all wireless (ac/ax/b/g/n) on Cudy WR3000H V1

cociweb, 2025-09-27:
### Describe the bug

After attended sys upgrade, I've lost all of my radios on Cudy WR3000H V1 (MediaTek MT7981)

The network is enabled (by config) but no client is associated.

[image]

[image]

### OpenWrt version

OpenWrt 24.10.3 r28872-daca7c049b / LuCI openwrt-24.10 branch 25.264.71301~3974d8a

### OpenWrt release

24.10.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000H v1

### Image kind

Official downloaded image

### Steps to reproduce

Do an attended sysupgrade from 24.10.2 to 24.10.3 on cudy WR3000H V1

### Actual behaviour

It breaks all of it's radio

### Expected behaviour

Survive an update

### Additional info

Since the device is relatively new, flashed according to the guide, and this was the first attended upgrade from the fully supported openwrt image. 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

cociweb, 2025-09-27:
Sure:
OpenWrt 24.10.3 r28872-daca7c049b / LuCI openwrt-24.10 branch 25.264.71301~3974d8a

cociweb, 2025-09-27:
seems to be similar: https://github.com/openwrt/openwrt/issues/19829

cociweb, 2025-09-27:
Kernel log:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.104 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28872-daca7c049b) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri Sep 19 21:19:38 2025
[    0.000000] Machine model: Cudy WR3000H v1
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x0000000047c80000..0x0000000047d7ffff (1024 KiB) nomap non-reusable wmcpu-reserved@47c80000
[    0.000000] OF: reserved mem: 0x0000000047d80000..0x0000000047dbffff (256 KiB) nomap non-reusable wo-emi@47d80000
[    0.000000] OF: reserved mem: 0x0000000047dc0000..0x0000000047ffffff (2304 KiB) nomap non-reusable wo-data@47dc0000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000004fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
[    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
[    0.000000]   node   0: [mem 0x0000000048000000-0x000000004fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000004fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s35624 r8192 d29912 u73728
[    0.000000] pcpu-alloc: s35624 r8192 d29912 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 0MB
[    0.000000] software IO TLB: area num 2.
[    0.000000] software IO TLB: SWIOTLB bounce buffer size roundup to 0MB
[    0.000000] software IO TLB: mapped [mem 0x000000004fe46000-0x000000004fec6000] (0MB)
[    0.000000] Memory: 239816K/262144K available (9088K kernel code, 1000K rwdata, 1676K rodata, 448K init, 307K bss, 22328K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: 0xffffffc080010080
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000080] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000088] pid_max: default: 32768 minimum: 301
[    0.002946] Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.002953] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.005089] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.005619] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=2.
[    0.005757] rcu: Hierarchical SRCU implementation.
[    0.005760] rcu: 	Max phase no-delay instances is 1000.
[    0.006168] smp: Bringing up secondary CPUs ...
[... 163 lines cut ...]
[   13.821965] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   13.833757] br-lan: port 2(lan2) entered blocking state
[   13.839079] br-lan: port 2(lan2) entered disabled state
[   13.844339] mt7530-mdio mdio-bus:1f lan2: entered allmulticast mode
[   13.852662] mt7530-mdio mdio-bus:1f lan2: entered promiscuous mode
[   13.868693] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
[   13.879992] br-lan: port 3(lan3) entered blocking state
[... cut ...]
[   14.205882] mtk_soc_eth 15100000.ethernet wan: configuring for phy/2500base-x link mode
[   14.246678] br-lan: port 5(wan) entered blocking state
[   14.251823] br-lan: port 5(wan) entered disabled state
[   14.257043] mtk_soc_eth 15100000.ethernet wan: entered allmulticast mode
[   14.263918] mtk_soc_eth 15100000.ethernet wan: entered promiscuous mode
[   18.395917] mtk_soc_eth 15100000.ethernet wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.395961] br-lan: port 5(wan) entered blocking state
[   18.409402] br-lan: port 5(wan) entered forwarding state
```

brada4, 2025-09-27:
bss_transition requires full hostap/wpad, use owut to keep it or tune configsto work with normal packages.

cociweb, 2025-09-27:
it was worked on 24.10.2 for weeks before the update. wpad-basic-mbedtls with its dependencies are installed.

brada4, 2025-09-27:
Impossible. Fix your configuration.

cociweb, 2025-09-28:
okay, it seems, that the attended sysupgrade (from gui) forces to add hostapd-basic-mbedtls while the wpad-basic-mbedtls package is also added to the list. (it seems the compile fails silently and this is resulted that both package is installed (only reportedly). this makes the confusion, and the radio does not able to read the config properly.

After remove the hostapd-basic-mbedtls and reinstall wpad-basic-tls (and recreate the wireless interfaces from scratch wih the same config) solves my issue.

closed 2025-10-02 by cociweb