openwrt/openwrt #20201: fails to initialize luci with kmod-mt7921u built

136185925, 2025-09-27:
### Describe the bug

Raspberry Pi 4B/400/CM4 (64bit)  24.10.3

fails to initialize luci with kmod-mt7921u build

built by firmware selector with uncomment lines

#
 wlan_name="OpenWrt"
 wlan_password="12345678"
#
 root_password=""
 lan_ip_address="192.168.1.1"

### OpenWrt version

24.10.3

### OpenWrt release

24.10.3

### OpenWrt target/subtarget

24.10.3

### Device

Raspberry Pi 4B 8G ram

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-09-27:
What packages you selected?

136185925, 2025-09-27:
> Invalid Version reported. `24.10.3` Is this from a clean repository?

True, 
firmware selection: Raspberry Pi 4B/400/CM4 (64bit)  
24.10.3

Device: Raspberry Pi 4B 8GB

all packages are (default) + kmod-mt7921u

base-files bcm27xx-gpu-fw bcm27xx-utils ca-bundle dnsmasq dropbear e2fsprogs firewall4 fstools kmod-fs-vfat kmod-nft-offload kmod-nls-cp437 kmod-nls-iso8859-1 kmod-sound-arm-bcm2835 kmod-sound-core kmod-usb-hid libc libgcc libustream-mbedtls logd mkf2fs mtd netifd nftables odhcp6c odhcpd-ipv6only opkg partx-utils ppp ppp-mod-pppoe procd-ujail uci uclient-fetch urandom-seed cypress-firmware-43455-sdio brcmfmac-nvram-43455-sdio kmod-brcmfmac wpad-basic-mbedtls kmod-i2c-bcm2835 kmod-spi-bcm2835 kmod-spi-bcm2835-aux kmod-i2c-brcmstb kmod-usb-net-lan78xx kmod-r8169 iwinfo luci kmod-mt7921u

without kmod-mt7921u: I can successfully enter the luci main interface through wifi
with kmod-mt7921u: Unable to display the root login interface

Script to run on first boot (default and uncomment lines)：

# Beware! This script will be in /rom/etc/uci-defaults/ as part of the image.
# Uncomment lines to apply:
#
 wlan_name="OpenWrt"
 wlan_password="12345678"
#
 root_password=""
 lan_ip_address="192.168.1.1"
#
# pppoe_username=""
# pppoe_password=""

# log potential errors
exec >/tmp/setup.log 2>&1

if [ -n "$root_password" ]; then
  (echo "$root_password"; sleep 1; echo "$root_password") | passwd > /dev/null
fi

# Configure LAN
# More options: https://openwrt.org/docs/guide-user/base-system/basic-networking
if [ -n "$lan_ip_address" ]; then
  uci set network.lan.ipaddr="$lan_ip_address"
  uci commit network
fi

# Configure WLAN
# More options: https://openwrt.org/docs/guide-user/network/wifi/basic#wi-fi_interfaces
if [ -n "$wlan_name" -a -n "$wlan_password" -a ${#wlan_password} -ge 8 ]; then
  uci set wireless.@wifi-device[0].disabled='0'
  uci set wireless.@wifi-iface[0].disabled='0'
  uci set wireless.@wifi-iface[0].encryption='psk2'
  uci set wireless.@wifi-iface[0].ssid="$wlan_name"
  uci set wireless.@wifi-iface[0].key="$wlan_password"
  uci commit wireless
fi

# Configure PPPoE
# More options: https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols#protocol_pppoe_ppp_over_ethernet
if [ -n "$pppoe_username" -a "$pppoe_password" ]; then
  uci set network.wan.proto=pppoe
  uci set network.wan.username="$pppoe_username"
  uci set network.wan.password="$pppoe_password"
  uci commit network
fi

echo "All done!"

brada4, 2025-09-27:
Please use formatting button `[<>]` while log segments are selected, it is unparseable.