openwrt/openwrt #20207: Telephony packages failed signature check on 24.10.3 with architecture `aarch64_cortex-a53`

ofry, 2025-09-27:
### Describe the bug

Telephony packages have failed signature check on 24.10.3 when trying to load packages base via `opkg`.

```
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

### OpenWrt version

r28872-daca7c049b

### OpenWrt release

24.10.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Just install OpenWRT 24.10.3 and run `opkg update`

### Actual behaviour

```
Downloading https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/kmods/6.6.104-1-6a9e125268c43e0bae8cecb014c8ab03/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/kmods/6.6.104-1-6a9e125268c43e0bae8cecb014c8ab03/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

### Expected behaviour

Signature check should pass on all packages.

### Additional info

`/etc/opkg/distfeeds.conf`
```
src/gz openwrt_core https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/packages
src/gz openwrt_base https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/base
src/gz openwrt_kmods https://downloads.openwrt.org/releases/24.10.3/targets/mediatek/filogic/kmods/6.6.104-1-6a9e125268c43e0bae8cecb014c8ab03
src/gz openwrt_luci https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/luci
src/gz openwrt_packages https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/packages
src/gz openwrt_routing https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/routing
src/gz openwrt_telephony https://downloads.openwrt.org/releases/24.10.3/packages/aarch64_cortex-a53/telephony

```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ofry, 2025-09-27:
Is it related to https://github.com/openwrt/openwrt/issues/19316 ?

xPXpanD, 2025-09-27:
Does sound like it... Do you also get failures on different files if you retry a few times over, let's say, a span of 30 minutes?

hauke, 2025-10-19:
Just try it again 1 hour later. 

The package feeds are build as a rolling release, when they get updated they are sometimes in an inconsistent state.