openwrt/openwrt #20209: Exploding size of the library libopenssl3 from 3.0.17-r1 to 3.0.17-r3

asmcc, 2025-09-27:
### Describe the bug

The size of libopenssl3 exploded last time from 1.35 MB in 3.0.17-r1 to 3.63 MB in 3.0.17-r3.
Upgrading is impossible, since I have only 2.1 MB free on my device.
Is it really necessary? 

### OpenWrt version

24.10

### OpenWrt release

OpenWrt 24.10.2

### OpenWrt target/subtarget

r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.264.71301~3974d8a

### Device

OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.264.71301~3974d8a

### Image kind

Official downloaded image

### Steps to reproduce

opkg upgrade

### Actual behaviour

Upgrade not possible

### Expected behaviour

Upgrading libopenssl3 on root from 3.0.17-r1 to 3.0.17-r3...
Collected errors:
 * verify_pkg_installable: Only have 2100kb available on filesystem /overlay, pkg libopenssl3 needs 3630
 

### Additional info

I update my packages to actual versions using "opkg upgrade". Therefore this bug regards not directly 24.10 without updated packages. But it is very important for the future, since with such "inflated" library a big part of devices can not be supported.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2025-09-27:
Are you looking at the `Size:` output of `opkg status`?  That is the size of the compressed archive, not the size of the actual contents.

What `opkg` is checking when is the `Installed-Size` field, which is the uncompressed size of the contents.

And you don't have enough space to contain an uncompressed copy of `libopenssl3`, which is why `opkg` is refusing to update.

Depending on your target, your overlay filesystem is compressed, so the actual required size is lower. But `opkg` does not know that your filesystem is compressed, and even if it knew, it cannot know how well it compresses (which also is highly dependent on the type of content).

`jffs2` as used for NOR flash devices does compress, but its compression is worse than what is used for the installable package due to various factors.

So the answer for "how much space does this package need" is somewhere between the 1.35 MB and 3.6 MB, but `opkg` does not know how much it actually is, so it uses the upper bound for checking. Running out of space in your overlay is rather catastrophical, you will need to do a factory reset to get back in a working state and will lose all your configuration (at this point you might as well just upgrade to 24.10.3).

If you feel confident the package will fit in the overlay, you can disable the space check with `--force-space`. Be aware of the risk mentioned above.

asmcc, 2025-09-27:
I found a workaround for my case. As described above in edit to "Additional info" I found the "causer" for it as "Midnight Commander". I do not know, for what mc needs this blowed openssl, but it depends on it. The workaround in my case is:
```root@aaa:~# opkg remove libopenssl3 --force-removal-of-dependent-packages
Removing package mc from root...
Removing package libssh2-1 from root...
Removing package libopenssl3 from root...
root@aaa:~# opkg install mc
Installing mc (4.8.32-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.2/packages/mips_24kc/packages/mc_4.8.32-r1_mips_24kc.ipk
Installing libopenssl3 (3.0.17-r3) to root...
Downloading https://downloads.openwrt.org/releases/24.10.2/packages/mips_24kc/base/libopenssl3_3.0.17-r3_mips_24kc.ipk
Installing libssh2-1 (1.11.0-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.2/packages/mips_24kc/packages/libssh2-1_1.11.0-r1_mips_24kc.ipk
Configuring libopenssl3.
Configuring libssh2-1.
Configuring mc.
```
In this case library is removed including all the packages (mc and libssh2 here). Then you can install it.
@KanjiMonster: Thanks for enlightenment!

jjm2473, 2025-09-28:
Maybe this changes size calculation https://github.com/openwrt/openwrt/commit/49e8f532989c342fff44c1c558f8182585673833