openwrt/openwrt #20218: OpenWRT One: No wifi with version OpenWrt SNAPSHOT r31180-86618df095

wom-bat, 2025-09-27:
### Describe the bug

It seems as though the iwinfo package is missing from the snapshot.  Manualy installing it fixes the problem.

### OpenWrt version

r31180-86618df095

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

OpenWRT One

### Image kind

Official downloaded image

### Steps to reproduce

Flash the firmware; try to bring up the WiFi interfaces.

### Actual behaviour

The interfaces always show as 'disabled' in luci; and do not show up when i type `ip link` in the CLI.
logread shows:
```
daemon.notice netifd: radio0 (2152): ./mac80211.sh: eval: line 644: iwinfo: not found
daemon.notice netifd: radio0 (2152): Could not find PHY for device 'radio0'
```

### Additional info

Installing `iwinfo` from the packages directory fixes the issue; but I'd expect it to be part of the base image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-09-27:
This should be fixed with https://github.com/openwrt/openwrt/pull/20211

efahl, 2025-09-28:
Seem you can fix it in the interim with `apk update && apk add iwinfo`  (https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035/804).

diizzyy, 2025-10-04:
Still broken with recent checkout, https://github.com/openwrt/openwrt/commit/915a57ccd9086db88ee74cfa531ff13f8652a447#comments

ttc0419, 2025-10-16:
@hauke Seems iwinfo still not installed by default, mind to take a look?