openwrt/openwrt #20249: Packets coming from a bridged VXLAN interface are incorrectly truncated

eric-j-ason, 2025-09-30:
### Describe the bug

### Description

Packets coming from a bridged VXLAN interface are incorrectly truncated to **1326 bytes** when they exceed **1398 bytes** in size.

Locally generated packets (e.g., `ping -M do -s 1472`) can reach the full **1500 MTU**, so the truncation only occurs for forwarded traffic through the VXLAN bridge.

I suspect this may be related to the **new GSO fraglist support** introduced in commit 19e977293591febe91fb1340e50dcb6b3d881732 (@nbd168). It looks like stale metadata on the skb may be incorrectly handled when frames come from a VXLAN.

### Setup

```text
[Client PC] (directly connected over Ethernet to eth1, no switch) 
    |
[OpenWrt device] (eth1 / mtk_soc_eth -- bridge -- vxlan -- wg) 
    |
[Remote server] (wg -- vxlan -- bridge)
```

### OpenWrt version

r31013-19e9772935

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-XE3000

### Image kind

Self-built image

### Steps to reproduce

1.  On the client PC, run a ping with DF set and near-MTU size, e.g.:
    ```bash
    ping -M do -s 1472 <server-ip>
    ```
2.  On the OpenWrt device, capture packets on the **eth1** interface:
    ```bash
    tcpdump -i eth1 -n icmp
    ```
3.  Observe that packets are correct and symmetric in size on ingress and egress of `eth1`.    
4.  On the client, notice that received ICMP replies are truncated (1326 bytes instead of the expected size).
5.  Pings with `-s` up to 1370 work as expected.

### Actual behaviour

- `tcpdump` shows correct packet sizes at the OpenWrt NIC (`eth1`).
- The client receives truncated/invalid ICMP responses.

### Expected behaviour

- Packets forwarded through the VXLAN bridge should preserve their size (up to MTU = 1500) and not be truncated.

### Additional info

- Device: **GL.iNet XE3000 (Puli AX)**
- Same behavior on eth0 and eth1
- SoC: **MediaTek MT7981A (Filogic 820)**
- Ethernet interface: **eth0** or **eth1** (driver: **mtk_soc_eth**)
- The issue does not occur if:
    - I build a firmware image with the commit immediately before 19e977293591febe91fb1340e50dcb6b3d881732, or
    - I use a different NIC - for example external USB-dongle

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-09-30:
Can you confirm rx-gro-list culpability setting `ethtool -K vxlanX rx-gro-list off` (repeat on each involved interface) immediatly makes packets not being chopped?

eric-j-ason, 2025-10-01:
> Can you confirm rx-gro-list culpability setting `ethtool -K vxlanX rx-gro-list off` (repeat on each involved interface) immediatly makes packets not being chopped?

Thank you for the suggestion. The issue remained even after that setting, so I suppose that in itself isn't the problem.

Furthermore, it turns out only some copies of the hardware exhibits the issue. That is, the problem arises on one device, but not on another, despite them being seemingly identical, differing only in serial number. This state of affairs remains even after reflashing them both. With the commit just before 19e977293591febe91fb1340e50dcb6b3d881732, everything works on all devices.

brada4, 2025-10-01:
Lets call in @nbd168 ...

Can you try firewall HW offload in a generic NAT scenario? I think previous attempt at mtk optimizing did not go down with that well....

Shine-, 2025-10-02:
There wasn't anything other related to `mtk_eth_soc` that I could find, which wasn't already in 24.10. Give me a hint and I'll try backporting it again. But for me it was also fixed by disabling GSO fraglist anyway.
However, checking out this specific feature was the whole reason why I backported the commit. Since it was broken, the whole purpose of the backport was defeated and I removed it from my builds again :-)

Shine-, 2025-10-02:
That is `680-net-add-TCP-fraglist-GRO-support.patch` in 24.10, actually.

Linaro1985, 2025-10-02:
@Shine- Not all the patches are there; there was a reorganization. I didn't have time to look through them all.

Shine-, 2025-10-02:
Sorry, can't find anything in the commit history mentioning GRO/GSO (outside from kernel bump reorganizations) that was **not** backported. Either way, this is drifting off-topic, the OP is using Snapshot, so we shouldn't get stuck on discussing backports to Stable ;-)

eric-j-ason, 2025-11-17:
> Now that [0ad0955](https://github.com/openwrt/openwrt/commit/0ad09552142a4c155811bf1ad8f59c88592cc3a1) and [05feabf](https://github.com/openwrt/openwrt/commit/05feabfd094aad94f6be4cb2f6cee713a7489847) have been created, things work even without turning features off with ethtool.

It still works, so, from my perspective, it's fine to close this issue now.

brada4, 2025-11-17:
You can tap the button under comment box

eric-j-ason, 2025-11-17:
> You can tap the button under comment box

Indeed. I just wanted to see if those of you who know more about the context around it wanted to keep it open, because of some unresolved aspect outside of my view.

closed 2025-11-29 by eric-j-ason